MichaIng/DietPi #7055: PiVPN (Wireguard) with Pi-Hole with Unbound DoT might leak the DNS query over VPN connection

yuukiAme, 2024-05-03:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

```
root@DietPi:~# cat /boot/dietpi/.version
cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

```
root@DietPi:~# echo $G_DISTRO_NAME $G_RASPBIAN
bookworm 0
```
  
- Kernel version | `uname -a`

```
root@DietPi:~# uname -a
Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```
  
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

```
root@DietPi:~# echo $G_HW_MODEL_NAME
RPi 4 Model B (aarch64)
```
  
- Power supply used | (EG: 5V 1A RAVpower)

```
Raspberry Pi Power supply 5V 3A
```
  
- SD card used | (EG: SanDisk ultra)
  
```
SanDisk Ultra 64G
```

#### Additional Information (if applicable)
- Software title

`PiVPN`
  
- Was the software title installed freshly or updated/migrated?

Freshly installed
  
- Can this issue be replicated on a fresh installation of DietPi?

Not sure.
  
- Bug report ID | `echo $G_HW_UUID`

```
root@DietPi:~# echo $G_HW_UUID
c16aa61b-b23a-4d7c-adcd-91591412a10b
```

#### Steps to reproduce

1. install Dietpi image on RPi

2. apt update && apt upgrade -y

3. dietpi-launcher

4. install Pi-hole, Unbound, activate DoT with the conf from Dietpi Docs

5. install pivpn, wireguard, setup DNS support with Pi-Hole ,add user, scan QR code

6. add conf file to a device (e.g: an Android phone) to connect to pivpn OUTSIDE of the network

7. check for the websites that is KNOWN to be blocked like 18+ sites, see if it returns query on unbound, load the content on the browser within my Local Network, meaning the DoT is working because my DNS query is encrypted and returned to Unbound with no issue.

8. Turn on Wireguard VPN on the device in Step 6
9. check if the known blocked website works and contents are loaded in the browser.

#### Expected behaviour

It should load all the contents of the website in the browser.

#### Actual behaviour

The browser will not load any content and return an error like "Unable to connect". Like an unencrypt query was rejected by the ISP DNS.

#### Extra details

I can use Pi-Hole with Unbound (activated DoT (DNS over TLS)). I can browse normally to any websites I want other than the block list inside Pi-Hole.
My country required ISP to blocks certain websites from their own DNS. So ISP will allow your network to query the website, but it will reject the content of the website.

Help me to troubleshoot if PiVPN is leaking the query because when I access my network over PiVPN, it behaves exactly like when I access the blocked websites WITHOUT Pi-Hole and Unbound DoT which mean my query to unencrypted and intercept by my ISP.

My question is:

How do I check/troubleshoot PiVPN whether it is encrypting my query to Pi-Hole and Unbound on port 5335 or leaving the query unencrypted like port 53?

yuukiAme, 2024-05-03:
> We should clarify a few basic things at the beginning.
> 
> When using a VPN such as Wireguard (PiVPN is only an interface on the server for managing clients), all traffic between the VPN client and VPN server is always encrypted. It doesn't matter what it is. Therefore, all requests, including DNS, between your mobile phone and your server are encrypted.
> 
> Communication between PiHole and Unbound is always unencrypted. However, this does not matter, as this only takes place on the server.
> 
> DNS requests between Unbound and the upstream DoT server, on the other hand, are encrypted again via TLS protocol.
> 
> Normally there should be no difference for DNS requests between a mobile phone via VPN or a client in the local network, as all requests first go to the PiHole and then to Unbound.
> 
> The chain actually looks like this: Client > (VPN) > PiHole > Unbound > Upstream DNS (via DoT)
> 
> Translated with DeepL.com (free version)

I understand all that. Thanks for the clarification. So my current VPN traffic should be encrypted. Only the traffics from Pi-hole and Unbound is unencrypted. 

Then what exactly can I do to test and find out why my query fails to certain websites in my VPN? Yet my traffic in local network works as intended.

Joulinar, 2024-05-03:
Check whether all traffic from the VPN client to the VPN server goes through the tunnel or whether only a split tunnel is active. `AllowedIPs = 0.0.0.0/0` should be set in the Wireguard client configuration.

MichaIng, 2024-05-03:
WireGuard (clients) have a setting `DNS` to enforce a different DNS server to query, which makes sense since a local one might not be reachable when traffic is forced through the tunnel. If your PiVPN is running on the same server which hosts Pi-hole, use the WireGuard-internal VPN server IP with this setting.

Note that this is only a client-side setting, same as whether all traffic is forced through the VPN, requests to the server-side LAN or the server itself only. So it is all about how the clients are configured.

Ah, and don't forget to allow "All origins" in Pi-hole for DNS queries. By default it allows queries only from the LAN/main interface of the server, hence blocks queries form any VPN interface.

yuukiAme, 2024-05-03:
> Check whether all traffic from the VPN client to the VPN server goes through the tunnel or whether only a split tunnel is active. `AllowedIPs = 0.0.0.0/0` should be set in the Wireguard client configuration.

From my Wireguard client conf

```
[Interface]
PrivateKey = iCHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHHcHE=
Address = 10.9.82.2/24
DNS = 10.9.82.1

[Peer]
PublicKey = FkPLcT0Nyt3JJJJJJJJJ+TMJAPwxEaxYOrcNTs=
PresharedKey = P292QilxCCCCCCCCCCCCCCCCCru4pkHzzPqCpK4=
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = example.domain.com:51820

```

Ignore the _PrivateKey_ , _PublicKey_, and _PresharedKey_, I changed to hide the actual value. So I definitely have `AllowedIPs = 0.0.0.0/0, ::/0`

yuukiAme, 2024-05-03:
> WireGuard (clients) have a setting `DNS` to enforce a different DNS server to query, which makes sense since a local one might not be reachable when traffic is forced through the tunnel. If your PiVPN is running on the same server which hosts Pi-hole, use the WireGuard-internal VPN server IP with this setting.
> 
> Note that this is only a client-side setting, same as whether all traffic is forced through the VPN, requests to the server-side LAN or the server itself only. So it is all about how the clients are configured.
> 
> Ah, and don't forget to allow "All origins" in Pi-hole for DNS queries. By default it allows queries only from the LAN/main interface of the server, hence blocks queries form any VPN interface.
> 
> EDIT: Use `AllowedIPs = 0.0.0.0/0, ::/0` at best (if you really want all traffic to be passed through the VPN) to assure IPv6 is as well forced through the tunnel. Else, if IPv6 is enabled on the client, all traffic to IPv6-enabled hosts would bypass the VPN.

It's a full tunnel, which allows access to everything within the network. I can use `http://pi.hole/admin`, even SSH into any VM or server I have in my local network, e.g. Nginx Proxy and Portainer, etc... 

I checked the public IP address and it's the same as the one from the Local Network, not the coffee shop public IP address or LTE Public IP. 
I also checked the dnsleaktest and they all reported that it is using Cloudflare DNS which is my upstream DNS inside of Unbound.

MichaIng, 2024-05-03:
> I also checked the dnsleaktest and they all reported that it is using Cloudflare DNS which is my upstream DNS inside of Unbound.

Hmm, then I do not understand what the issue is, when clients do in fact use that Pi-hole instance as DNS server through the VPN. If those are Linux/UNIX clients, can you check `/etc/resolv.conf`, whether `10.9.82.1` is really the only `namesever` entry?

MichaIng, 2024-05-03:
I mean on the client(s).

yuukiAme, 2024-05-03:
> I mean on the client(s).

RIP. it's an iPhone. 
I also have an Android phone with Termux installed. Give me a few minutes to setup another pivpn client on that Android device.

yuukiAme, 2024-05-03:
The Termux doesn't work with `cat /etc/resolv.conf`. Nevermind. 

Any other methods?

Joulinar, 2024-05-03:
From your mobile phone, you should see the DNS queries inside the PiHole log. Do you? Even for the website in question. If this is the case, DNS resolution is working fine.

MichaIng, 2024-05-03:
Yes, especially check whether you see a query log entry for those cases, where a website is blocked, while you expect it to be accessible thanks to the VPN DNS.

Ah, and also check whether the related browser has DoH enabled, which would bypass the system DNS and VPN.

yuukiAme, 2024-05-03:
> From your mobile phone, you should see the DNS queries inside the PiHole log. Do you? Even for the website in question. If this is the case, DNS resolution is working fine.

@Joulinar Yes, I can access Pi-hole over the VPN, use the `http://pi.hole/admin/, and log in the admin panel, see the query log with my client name called `wireguard.pivpn`. DNS resolution is definitely working. I even check with SSH into the pi (over VPN), use `dig` to check Unbound issues. I even checked the Pi-hole > Tools > Search Adlists, to make sure none of my blocklists blocked the exact domain I want to test.

@MichaIng For sure, the DoH, I always check that one off within the browser, on Android 13 and later, inside its Settings, it has Private DNS. I made sure to turn that off first, not even _Automatic_.

Joulinar, 2024-05-03:
Question was if you see inside PiHole log the DNS request for the website not working. If this is the case, DNS resolution is not your problem.

Joulinar, 2024-05-03:
yes DNS resolution is working, so it's something else. Like the browser escaping the VPN tunnel

yuukiAme, 2024-05-03:
>  Like the browser escaping the VPN tunnel

And exactly like how my browser behaves when my query was unencrypted using 1.1.1.1 at port 53 so my ISP DNS rejects the query. So that's why I'm stuck here. Wondering if PiVPN is "leaking" the query. But I know it shouldn't be.

Joulinar, 2024-05-03:
It's not a DNS issue as you see the query inside PiHole. As well it is forward to Unbound according your PiHole screen shot. It's not the DNS query escaping but somehow the browser not using to VPN to access the website.

yuukiAme, 2024-05-03:
I thought it was a fluke with Firefox. So I installed Vivaldi Browser for Android. I went into Vivaldi Settings > Privacy and security > and **disabled** `use secure DNS` : `Off`

I deliberately clicked on a Vivaldi bookmark added that I know for sure is in the blocklist. It blocked it successfully. But I tried with `pornhub.com` and the content is never delivered like Firefox. 

[image]

[image]

I think you're right, @Joulinar . 

> the browser not using to VPN to access the website.

The browser is clearly using the Pi-hole DNS inside the VPN, but the domains blocked by my country is not working inside it.

MichaIng, 2024-05-03:
Can you check whether the IP address `pornhub.com` is resolved to, is the same when accessing from your PC (within your LAN) and when accessing from the VPN client? If both get the same IP address, and all traffic of the client is passed through the VPN, I have no idea how the client could behave any differently. The browser should not be able to bypass the VPN, at least not when using accessing public/routable IP address. Other than DNS, network traffic/routes are enforced at system level.

And you are sure that Unbound is configured to use DoT? Compare with our docs: https://dietpi.com/docs/software/dns_servers/#__tabbed_2_5

Btw, on your PC `pornhub.com` works since you use DoH there?

yuukiAme, 2024-05-03:
> Can you check whether the IP address pornhub.com is resolved to, is the same when accessing from your PC (within your LAN) and when accessing from the VPN client?

on PC - LAN
Result: Working as intended

on Tablet (Android) - VPN 
Result : Blocked

[image]

It's using the same IP address. The Cache is from the first query from PC.

> And you are sure that Unbound is configured to use DoT? Compare with our docs: https://dietpi.com/docs/software/dns_servers/#__tabbed_2_5

Yes. I was troubleshooting another problem with domain end in `.site` with NLnetLabs/Unbound yesterday on Github. [Issue here](https://github.com/NLnetLabs/unbound/issues/1060#issue-2274998779)

> Btw, on your PC `pornhub.com` works since you use DoH there?

Yup. It works. Same with other 18+ sites and news sites that my countries blocked.

MichaIng, 2024-05-03:
Interesting, the LAN client seems to send a DoH query to Pi-hole, which is forwarded to Unbound, if I understand the "HTTPS" query type correctly?

However, so the A type query is answered as intended, works for the LAN client, is cached as intended, and hence answered from cache to the VPN client, but does not work there.

Confusing, indeed. The Android tablet is currently also located in your LAN, just using the VPN (unnecessarily, for testing), right? So that really only enabling the VPN makes the difference?

yuukiAme, 2024-05-03:
> Interesting, the LAN client seems to send a DoH query to Pi-hole, which is forwarded to Unbound, if I understand the "HTTPS" query type correctly?

I don't have DoH enabled for this network at all. Only Pi-Hole with Unbound DoT. I understand the difference between them. Like how Adguard-DNS / Adguard Home using DNS-over-HTTPS with their SSL cert and privatekey in Encryption Settings.

> The Android tablet is currently also located in your LAN, just using the VPN (unnecessarily, for testing), right? So that really only enabling the VPN makes the difference?

I have 3 different ways of connecting to the internet. 

1. Network #1 have Pi-Hole with Unbound DoT and PiVPN 
2. Network #2 for my family uses, have no DNS server, or anything. Just ISP modem/router combo. So ISP can see my unencrypted query easily and reject it.
3. 4G/LTE from my phones. 

So I use the Tablet connected to Network #2 and turn on Wireguard app to connect to PiVPN and it connected to Network #1 and I open `pornhub.com` in my Firefox browser and it's blocked.

I tried the exact same thing with 4G/LTE network as above. Same result.

Which led me to the Title of the issue "Leak DNS query over VPN connection". I know it isn't true. 
That's why I'm asking for help to find faults in my understanding with PiVPN and disprove the title.

MichaIng, 2024-05-03:
PiVPN means a (regular) WireGuard server in your case. Might be important when you check for answers elsewhere. PiVPN is only the installer, and the CLI to create client configs, otherwise it has no effect on the VPN.

With `AllowedIPs = 0.0.0.0/0, ::/0`, leaking the DNS query or anything else should not be possible. This creates a route which forces every routable request through the VPN. Only quests to local/LAN range IP addresses can then bypass the VPN. A killswitch can be used to prevent that as well. So I think, whatever happens, it happens at the VPN server/LAN side. We see the DNS query at Pi-hole, so that does not seem to be the issue. Probably traffic from the `pornhub.com` IP is blocked, not the DNS query, but that should apply for any other request from LAN the same way.

Maybe best to use `tcpdump` at the server, so see which requests are are coming and going from the VPN client. Also, when you temporarily switch to using Pi-hole as DNS resolver for the VPN/Pi-hole server itself (`/etc/resolv.conf`), does accessing `pornhub.com` via `curl` work?

yuukiAme, 2024-05-04:
> when you temporarily switch to using Pi-hole as DNS resolver for the VPN/Pi-hole server itself (/etc/resolv.conf), does accessing pornhub.com via curl work?

output

```
root@DietPi:~# nslookup pornhub.com
Server:         1.1.1.1
Address:        1.1.1.1#53

Non-authoritative answer:
Name:   pornhub.com
Address: 66.254.114.41
```

follow up output with `curl`

```
root@DietPi:~# curl pornhub.com
<!DOCTYPE html>
<html>
<header><title>404 Page</title></header>
<body>
<font size="6"><b>We're sorry but the page you're looking for could not be found!!!!!</b></font>
</body>
</html>
```

which is expected because of the transparent DNS query.

I will do the `tcpdump` later because this will take a while to check.

yuukiAme, 2024-05-04:
Wait. 
Should I change the **Static DNS**  in `Dietpi-config` from `Cloudflare DNS` to  to `Pi-Hole with unbound DoT itself`instead?

Joulinar, 2024-05-04:
usually, DNS setting in DietPi itself has no relation to PiHole. And best practice is not to change to a local DNS server like PiHole or AGH.

yuukiAme, 2024-05-04:
Background: I brought my Macbook Air to the coffee shop. It took a while to setup the Wireguard because of the newer Wireguard cannot be installed from App Store. I had to deal with brew, Wireguard-tools and, wireguard-go

Macbook output `/etc/resolv.conf`

```

mac@192 ~ % cat /etc/resolv.conf 
#
# macOS Notice
#
# This file is not consulted for DNS hostname resolution, address
# resolution, or the DNS query routing mechanism used by most
# processes on this system.
#
# To view the DNS configuration used by this system, use:
#   scutil --dns
#
# SEE ALSO
#   dns-sd(1), scutil(8)
#
# This file is automatically generated.
#
nameserver 10.9.82.1

```

Macbook output on `curl pornhub.com`

```
mac@192 ~ % curl pornhub.com
<html><body><h1>408 Request Time-out</h1>
Your browser didn't send a complete request in time.
</body></html>
```

Macbook requests after using `curl pornhub.com` being caputure with `tcpdump` on the Server (Pi-hole)

MichaIng, 2024-05-04:
> which is expected because of the transparent DNS query.

I am not sure about that: The DNS query and the actual connection to the resolved host are two different things. Usually, sites are blocked by ISP by spoofing the DNS response with none/invalid. The IP you get from the transparent DNS response however is valid. It seems that the response from the pornhub host itself is intercepted. `curl pornhub.com` should return a 301 to HTTPS. Can you test this:
```sh
curl https://www.pornhub.com/
```

Your `tcpdump` also shows the response from `reflectededge.reflected.net.http`, which is indeed the DNS pornhub uses, i.e. the same happens when I do a successful request here.

HTTPS responses cannot be spoofed without the client at least warning about a wrong certificate. So if above HTTPS call fails the same way for you, it is pornhub themselves, or their CDN, who deny to send you data, not the ISP.

yuukiAme, 2024-05-04:
@MichaIng 

output of `curl -I https://www.pornhub.com/`

```
sh-3.2# curl -I https://www.pornhub.com
curl: (35) LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to www.pornhub.com:443
```

yuukiAme, 2024-05-04:
At this point, I am considering switching to another DNS, like Adguard Home and using my own SSL from my own domain.

It doesn't tunnel or change my public IP address, but at the very least, my traffic isn't blocked.

MichaIng, 2024-05-04:
The DNS is irrelevant. The client does get the correct IP. AdGuard Home will behave exactly the same.

What do you mean with your "own SSL" from your "own domain"? Your client either has the IP of your home/location where the VPN server is located, or the mobile/public WiFi IP.

The strange thing is why visiting pornhub from the PC at the same location works, when using DoH, while the "actual connection" (not DNS) from the VPN server system and VPN clients (using DoT) is responded with fake error documents, even different ones for both.
```console
sh-3.2# curl -I https://www.pornhub.com
curl: (35) LibreSSL SSL_connect: SSL_ERROR_SYSCALL in connection to www.pornhub.com:443
```
Hmm, this indeed looks like the connection is roughly intercepted and a MITM answers which does not accept HTTPS connections at all. But probably it is also a different issue: https://github.com/libressl/portable/issues/369
Though I guess it is similar to this one: https://serverfault.com/questions/955667/https-connection-tls-hangs-and-eventually-fails-ssl-error-syscall

Can you test this:
```sh
openssl s_client -connect pornhub.com:443
```

yuukiAme, 2024-05-04:
> What do you mean with your "own SSL" from your "own domain"? Your client either has the IP of your home/location where the VPN server is located, or the mobile/public WiFi IP.

The AdGuard Home can be used like `Adguard-dns.io` from AdGuard which is free up to 300k query per month. AdGuard Home is a self-host solution.

Screenshot of the Encryption settings for Private DNS can be reached over the internet with HTTPS

[image]

I can reach AdGuard Home DNS from outside (internet) after I setup the SSL cert and private key. Then setup the Nginx Proxy Manager and point the proxy host to AdGuard Home (Local address - https://192.168.1.2/). So I can use the Adguard Home in the local network and outside the network with a URL like this 

> https://adguard.domain.com/dns-query/<-clientID->

after I put that into the **Private DNS** in the Android Settings or use the Adguard.apk.

---

output of `curl -I https://dietpi.com/` on Macbook over VPN

```
sh-3.2# curl -I https://dietpi.com
HTTP/2 200 
date: Sat, 04 May 2024 13:04:52 GMT
content-type: text/html; charset=utf-8
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 0
x-robots-tag: all
x-permitted-cross-domain-policies: none
referrer-policy: no-referrer-when-downgrade
content-security-policy: upgrade-insecure-requests; frame-ancestors 'self'; base-uri 'self' https://dietpi.com/matomo/index.php https://dietpi.com/grafana/; default-src 'none'; object-src 'none'; style-src 'self' 'unsafe-inline'; font-src 'self' data:; img-src * data: blob:; script-src 'self' 'unsafe-inline' 'unsafe-eval' https://cdnjs.cloudflare.com; worker-src 'self' blob:; frame-src 'self'; manifest-src 'self'; connect-src 'self' https://api.github.com
permissions-policy: accelerometer=(), autoplay=(), browsing-topics=(), camera=(), display-capture=(), encrypted-media=(), fullscreen=(), geolocation=(), gyroscope=(), interest-cohort=(), magnetometer=(), microphone=(), midi=(), payment=(), usb=(), screen-wake-lock=()
cache-control: public, max-age=86399
strict-transport-security: max-age=31536000; includeSubDomains; preload
last-modified: Thu, 02 May 2024 20:56:15 GMT
vary: Accept-Encoding
content-language: en
cf-cache-status: DYNAMIC
server: cloudflare
cf-ray: 87e8b1805e9b8508-HKG
alt-svc: h3=":443"; ma=86400
```

output of `openssl s_client -connect pornhub.com:443`

```
sh-3.2# openssl s_client -connect pornhub.com:443
CONNECTED(00000005)
depth=2 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Global Root G3
verify return:1
depth=1 C = US, O = DigiCert Inc, CN = DigiCert Global G3 TLS ECC SHA384 2020 CA1
verify return:1
depth=0 C = CY, L = Dali, O = AYLO Freesites Ltd, CN = *.pornhub.com
verify return:1
---
Certificate chain
 0 s:/C=CY/L=Dali/O=AYLO Freesites Ltd/CN=*.pornhub.com
   i:/C=US/O=DigiCert Inc/CN=DigiCert Global G3 TLS ECC SHA384 2020 CA1
 1 s:/C=US/O=DigiCert Inc/CN=DigiCert Global G3 TLS ECC SHA384 2020 CA1
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert Global Root G3
 2 s:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert Global Root CA
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert Global Root CA
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIFYDCCBOagAwIBAgIQD837D4uUBWkl8BJGQLk/eTAKBggqhkjOPQQDAzBZMQsw
CQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMTMwMQYDVQQDEypEaWdp
Q2VydCBHbG9iYWwgRzMgVExTIEVDQyBTSEEzODQgMjAyMCBDQTEwHhcNMjQwMTE1
MDAwMDAwWhcNMjUwMjE0MjM1OTU5WjBRMQswCQYDVQQGEwJDWTENMAsGA1UEBxME
RGFsaTEbMBkGA1UEChMSQVlMTyBGcmVlc2l0ZXMgTHRkMRYwFAYDVQQDDA0qLnBv
cm5odWIuY29tMFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEQ6ks4Tmx27FAmv5Q
[... 47 lines cut ...]
    0010 - ea 51 7d 45 3f c8 15 82-60 10 10 04 4c 81 ce 3d   .Q}E?...`...L..=
    0020 - 4d 16 dd c5 00 6b 41 d4-f9 23 df 82 9a 77 2d 60   M....kA..#...w-`
    0030 - 76 6c b9 83 4b f8 22 d0-d6 8d f7 36 3e 72 bb 7d   vl..K."....6>r.}
    0040 - bc 50 75 5d df 17 5d 58-22 06 f9 4a 94 26 0f 35   .Pu]..]X"..J.&.5
    0050 - a5 1d 0e 7e 30 7f 29 5d-10 b5 f6 a3 5e 07 68 9e   ...~0.)]....^.h.
    0060 - f6 48 93 6f 93 3c d0 51-2d 4c e3 6f 1d c7 4c a6   .H.o.<.Q-L.o..L.
    0070 - 7c 0d 94 63 94 2d 3d 87-5e 65 e8 c3 30 af df 70   |..c.-=.^e..0..p
    0080 - ba 98 11 37 7e f0 f5 f6-4a e2 a7 e3 7b c1 1f 37   ...7~...J...{..7
    0090 - 42 3b c3 70 dc d6 3d d2-e1 23 04 ba 5e 2f 95 b7   B;.p..=..#..^/..

    Start Time: 1714827957
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
---
HTTP/1.1 408 Request Time-out
Content-length: 110
Cache-Control: no-cache
Connection: close
Content-Type: text/html

<html><body><h1>408 Request Time-out</h1>
Your browser didn't send a complete request in time.
</body></html>
closed
```

yuukiAme, 2024-05-04:
@MichaIng 

I just checked in on my AdGuard Home Upstream DNS. I found this. The default DNS already changed to Pi-Hole for my entire network which is why AdGuard Home was using Pi-Hole as one of the upstream DNS.

screenshot below

[image]

It seems that the AdGuard Home was using the Pi-hole **DNS without Unbound** 

> 192.168.16.2:53

That's my Pi-Hole local IP Address. Shouldn't it be `192.168.16.2:5335` ?

Joulinar, 2024-05-04:
Nope, Pihole will ask Unbound not the client

yuukiAme, 2024-05-04:
Okay. I will trust you on that. I only here to help troubleshooting the problem. I haven't gotten a clue if it's the update or browser or anything here.

We can rule out it's not DNS or PiVPN or Wireguard.

MichaIng, 2024-05-04:
> The AdGuard Home can be used like `Adguard-dns.io` from AdGuard which is free up to 300k query per month. AdGuard Home is a self-host solution.

Yes, exactly like Pi-hole. But what you mean it probably that AGH supports downstream DoT and DoH, i.e. browsers can query it via DoH. Since you connect to your Pi-hole via VPN, it _should_ not make a difference. But since somehow the VPN made the difference, worth to give it a try. But please do not run your AGH as open resolver. I am not sure whether DoH/DoT-only servers can be misused for [DNS amplification attacks](https://blog.apnic.net/2022/05/13/open-dns-resolvers-from-bad-to-worse/), but to rule out being abused for any bad things, or even just get overloaded from public access, configure AGH to accept connections only form your known clients. Since their IP changes with location, I am actually not sure right now how to even achieve that. It is not possible, try first with VPN only (not forwarding DoH port), and see whether DoH over VPN works.

The pornhub certificate is the original one. Actually, the timeout response is no fake. When I run that `openssl` command at let it just wait, I get the same after a while. Probably `curl` on Apple devices behaves a little different, or indeed the connection to the pornhub CDN is flaky/slow for whatever reasons.

yuukiAme, 2024-05-04:
> But please do not run your AGH as open resolver.

I don't think I'm running AGH as open resolver. It has long string of ClientID for each of my devices. Unless I'm misunderstanding what you are talking about "open resolver".

> I am not sure whether DoH/DoT-only servers can be misused for [DNS amplification attacks](https://blog.apnic.net/2022/05/13/open-dns-resolvers-from-bad-to-worse/), but to rule out being abused for [...]

I will have to do more research on that. Thanks for the tip. Ooh. Scary stuffs.

> The pornhub certificate is the original one. Actually, the timeout response is no fake. When I run that openssl command at let it just wait, I get the same after a while. Probably curl on Apple devices behaves a little different, or indeed the connection to the pornhub CDN is flaky/slow for whatever reasons. Or the VPN for whatever reasons slows down the connection. But that would affect not only pornhub then ...

Are we on the "right" track? I'm still not sure what we are nailing down here yet. But I am willing to keep troubleshooting. Just let me know.

MichaIng, 2024-05-04:
> I don't think I'm running AGH as open resolver. It has long string of ClientID for each of my devices. Unless I'm misunderstanding what you are talking about "open resolver".

But port 443 is still forwarded, so anyone can theoretically connect directly to your AGH, isn't it? Bots earlier or later scan and find all open ports.

> Are we on the "right" track? I'm still not sure what we are nailing down here yet. But I am willing to keep troubleshooting. Just let me know.

I am not sure either and wouldn't have any explanation. But if I understood your earlier posts correctly, using DoH with any upstream DoH resolver from within your LAN without VPN, it works, but once enabling WireGuard to use Pi-hole>Unbound>DoT (and disable DoH in browser), it does not work. So either the VPN or DoH seem to make the difference.

yuukiAme, 2024-05-04:
the port 443 is port forwarded to reverse proxy with Nginx. I also use Cloudflare domain. So Cloudflare themselves reverse proxy my domain behind their servers.

> But if I understood your earlier posts correctly, using DoH with any upstream DoH resolver from within your LAN without VPN, it works

That's correct.

>  but once enabling WireGuard to use Pi-hole>Unbound>DoT (and disable DoH in browser), it does not work. So either the VPN or DoH seem to make the difference.

That's also correct. 

> We know that you get the correct IP, so it is not the DNS response, but just in theory, when you have an ISP router which tracks DNS traffic within your LAN ... that would be so shady,

My router is a TP-Link Archer one. There is no modem. I use XZ000-G3 as the GPON converter. Between the GPON converter and TP-Link Router, they handle the PPPoE authentication for my internet access. Within PPPoE, they uses their own ISP DNS there,

MichaIng, 2024-05-04:
> the port 443 is port forwarded to reverse proxy with Nginx. I also use Cloudflare domain. So Cloudflare themselves reverse proxy my domain behind their servers.

A proxy only hides the IP for clients who know/access the hostname. But it does not help against bots scanning for open ports across plain IP ranges. At a very minimum, you must disable plain port 53 DNS service in AGH, if even possible. But as said, probably open DoT and DoH resolvers can be misused just the same or similar way. Assure that before running your AGH like that. These kind of attacks cause real damage to others!

> My router is a TP-Link Archer one. There is no modem. I use XZ000-G3 as the GPON converter. Between the GPON converter and TP-Link Router, they handle the PPPoE authentication for my internet access. Within PPPoE, they uses their own ISP DNS there,

As of my edit above, the router should not even see any plain DNS traffic from VPN clients, as everything which reaches and leaves this very VPN/Pi-hole/Unbound server is/should be encrypted. However, since we do not have any other explanation so far, to test DoH with AGH, you could just do that within your LAN and keep the port closed. On the VPN client, you can add the public hostname of the VPN server to `/etc/hosts` with its LAN IP (or macOS/iOS has an own mechanism to add host entries). Or did you already test it while the port was open?

yuukiAme, 2024-05-05:
I didn’t port forwarding port 53 at all. How do I check if I open port 53 in the local network of AGH?

and the test for `/etc/hosts/`

MichaIng, 2024-05-06:
> How do I check if I open port 53 in the local network of AGH?

By using it as DNS with some client, e.g. However, if port 53 is not forwarded, it should be fine.

> and the test for `/etc/hosts/`

It is a file, not a directory. On macOS it is indeed used, hence could be edited like this form a terminal:
```sh
sudo nano /etc/hosts
```
Then you could add a line like that:
```sh
192.168.1.2 adguard.domain.com
```
However, I just recognised that it does not work like that, since the AdGuard proxy just redirects requests in the `/dns-query/<clientID>` path. And your AdGuard host does not react on this path. So I am currently not sure how to test this with closed port 443 bypassing the proxy.

Btw, does it work when using the WireGuard VPN as well as DoH with the AdGuard proxy?

yuukiAme, 2024-05-07:
@MichaIng 

> It is a file, not a directory. On macOS it is indeed used, hence could be edited like this form a terminal:
> sudo nano /etc/hosts
> Then you could add a line like that:
> 192.168.1.2 adguard.domain.com
> However, I just recognised that it does not work like that, since the AdGuard proxy just redirects requests in the /dns-query/<clientID> path. And your AdGuard host does not react on this path. So I am currently not sure how to test this with closed port 443 bypassing the proxy.

I guess I will skip this test for now. Until you come up with method to test.

> Btw, does it work when using the WireGuard VPN as well as DoH with the AdGuard proxy?

I will make a new persistent ClientID just for the Wireguard VPN under [Interface] - `DNS = 10.9.82.1` to `DNS = adguard.domain.com/dns-query/<ClientID>`

I did the test on Macbook and changed its Wireguard conf `DNS = 10.9.82.1` to `DNS = https://adguard.domain.com/dns-query/<ClientID>`

The outcome was

```
** Error : The parameters were not valid.
```
when I ran `sudo wg-quick up wg0`.

so I changed its Wireguard conf `DNS = 10.9.82.1` to `DNS = adguard.domain.com/dns-query/<ClientID>`

this time the output was normal, but the query log in AGH did not show the ClientID in the log at all. None showed up.

So the DNS wireguard is using `192.168.97.1` which is the coffee shop default DNS.

I also redid the test again after clearing the dns cache on the Macbook with `sudo dscacheutil -flushcache; sudo killall -HUP mDNSResponder`

What else should I test?

MichaIng, 2024-05-07:
```
** Error : The parameters were not valid.
```
This is expected. A full URL as DNS nameserver is valid only for DoH. WireGuard accepts a hostname only, not `https://` URLs. Please remove this setting completely and just use DoH in your browser with this URL.

yuukiAme, 2024-05-07:
> Please remove this setting completely and try to just use DoH in your browser with this URL

Okay. Here are what I did.

I changed the `wg0.conf` back to the `DNS : 10.9.82.1`. Turned on Wireguard. Changed the DNS in Firefox to `Max Protection : adguard.domain.com/dns-query/<ClientID>`.

I tested with 3 domains. 
- google | working as intended
- pornhub | never loaded
- medium | PR_connection_reset

All query above are found in AGH Query log. All of the query was `Processed`.

MichaIng, 2024-05-07:
Please remove (or comment) the "DNS" setting completely, just to be sure/rule this out. It is optional.

The client is currently in the same LAN as the server? Then see whether disabling the VPN (with removed "DNS" setting) fixes pornhub access, enabling the VPN breaks it again. If so, we can basically rule out the DNS resolution and the way/protocol this is done with, and all that matters is whether the pornhub access itself (to the already resolved IP) is done from/through the VPN server system or not. No idea how this can make a difference, since in both cases, traffic is routed through the router and NAT, so the ISP and remote host should not be able to differentiate between requests from different clients behind the same NAT. pornhub is IPv4-only, and the HTTP protocol version is defined by client and remote server, so the VPN has no effect on this.

A `traceroute` might be interesting from the client. From macOS terminal:
```sh
traceroute www.pornhub.com
traceroute 66.254.114.41
```
Once with once without VPN enabled. So we see whether from the VPN server on, the route is exactly the same or not, and again that the preceding DNS resolution has no effect on this in either case.

yuukiAme, 2024-05-07:
> Please remove (or comment) the "DNS" setting completely, just to be sure/rule this out. It is optional.

Done.

> The client is currently in the same LAN as the server? 

The client here is Macbook. I tried using my network with Pi-Hole and Unbound. It fails to use `www.pornhub.com` now. Which is strange. So I used another daily driver PC - O11. It works with Vivaldi browser and Firefox browser. so I went back and tried Safari browser on Macbook with `www.pornhub.com` again but it fails with 408 Request Time-out

Summary table - on Pi-Hole and Unbound DoT network

|#|Device|Browser|Status|
|---|---|---|---|
|1|PC|Firefox, Vivaldi|Working as intended|
|2|Macbook|Firefox|Secure Connection Failed - PR_END_OF_FILE_ERROR|
|3|Macbook|Safari|408 Request Time-out|

> Then see whether disabling the VPN (with removed "DNS" setting) fixes pornhub access, enabling the VPN breaks it again. If so, we can basically rule out the DNS resolution and the way/protocol this [...]

I removed the entry `DNS = 10.9.82.1` completed. Saved the config. Turned off `DNS over HTTPS : Off` in Firefox on Macbook. Turned on Wireguard VPN. 

I will address the traceroute at the later comment. Because I know the traceroute will reveal a lot of public IP address that might compromise my real location. So it will takes me some time to redact certain IP Addresses.

yuukiAme, 2024-05-07:
the result is too long so I will put it upload a .txt file instead.
I already redacted IP address that could reveal my location.

Text file below

[traceroute.txt](https://github.com/MichaIng/DietPi/files/15236686/traceroute.txt)

Summary table

|#|device|test|note|status|
|---|---|---|---|---|
|1|macbook|traceroute www.pornhub.com| with no VPN on coffee internet - default ISP DNS|Failed|
|2|macbook|traceroute 66.254.114.41|same as above|Failed|
|3|macbook|traceroute www.pornhub.com| with VPN on - no DNS entry in the config|Failed|
|4|macbook|traceroute www.pornhub.com| with VPN on - with 10.9.82.1 added back to DNS entry|Failed|
|5|macbook|traceroute 66.254.114.41 |same as above|Failed|
|6|PC|traceroute www.pornhub.com| on PC - Pi-hole with Unbound DoT - no VPN|Success - Trace complete|

MichaIng, 2024-05-07:
First of all, of course, when you disable the DNS setting (i.e. not use Pi-hole>Unbound>DoT) in the WireGuard config, you need to enable DoH in browsers. The idea was to test whether, when DoH is used, toggling the VPN alone toggles the issue. Since Pi-hole does not support DoH, you can test this with any upstream DoH provider, or test again with AdGuard Home.

Of course `pi.hole` is only available, when using Pi-hole for DNS resolution, hence this only works when enforcing Pi-hole as DNS via VPN config or without VPN via LAN/DHCP server, and DoH must be disabled. However, this is irrelevant for the pornhub issue.

When you remove the "DNS" setting and stop the VPN, it is possible that you need to disconnect and reconnect to your LAN network, to get the DNS server from the DHCP server. I am not sure whether it has the original DNS server still stored somewhere internally to recover it on VPN stop. However, this should be a one-time action, since the VPN won't touch `/etc/resolv.conf` anymore on startup now. This would explain why accessing any hostname, including Google, failed.

Regarding the `traceroute`, interesting would be which route exactly it tried. It would be interesting to see the routes for the two test cases of my first sentence: VPN enabled + DoH, VPN disabled + DoH, both when connected to the home LAN
I assume that, while DoH is used, enabling the VPN breaks pornhub access, disabling the VPN fixes it. And the route (after home router) is the only difference I can think about, while I have no idea how using the VPN from within your home LAN can have an effect on it.

yuukiAme, 2024-05-07:
> when you disable the DNS setting (i.e. not use Pi-hole>Unbound>DoT) in the WireGuard config, you need to enable DoH in browsers. The idea was to test whether, when DoH is used, toggling the VPN alone toggles the issue. 

Well, I redid the test like you suggested above, 
- deleted DNS entry in conf file and saved the file
- change Firefox Settings > DNS over HTTPS > Max protection > custom > my own AGH with ClientID

outcome: 
- Browser cannot use the internet (i.e. www.google.com)
- Browser can access local network like the coffee shop's internet default Gateway (`192.168.1.1`)
- AGH query log has no entry from Macbook ClientID at all.
- using `nslookup www.google.com` in Terminal returned a google public IP address (`141.251.220.14`).

changed Max protection > Cloudflare (default)

- Browser cannot use the internet (i.e. www.google.com)
- Browser can access local network like the coffee shop's internet default Gateway (`192.168.1.1`)

Same outcome. 

> Since Pi-hole does not support DoH, you can test this with any upstream DoH provider, or test again with AdGuard Home.

It seems like without the DNS entry in the config file, the VPN will use the default DNS by the router which has the ISP DNS and block my query immediately if it's banned in my country.

I have nothing else to add related to matters above. 

But one interesting tidbits, my country blocked `https://store.steampowered.com` this entire time since April 2024 - now, but I never noticed it because of Pi-hole, Unbound DoT, they unblock everything for me. I'm so glad to be using DietPi to assist me with all this. 

I didn't believe it until I tried `nslookup https://store.steampowered.com` with default DNS provided by my ISP and it returned `127.0.0.1` . But with PiVPN on Macbook, `nslookup https://store.steampowered.com` returned the Steam Public IP.

I guess my ISP is cracking down harder on `www.pornhub.com` traffics than `https://store.steampowered.com`. But I still have no idea why my PC (O11) is working as intended but my other devices cannot access `www.pornhub.com`.

MichaIng, 2024-05-07:
> It seems like without the DNS entry in the config file, the VPN will use the default DNS by the router which has the ISP DNS and block my query immediately if it's banned in my country.

Hmm, yes, the "system" uses the default DNS from the ISP or whatever you configured in your LAN. However, with DoH enabled, the browser overrides this. Hence if you cannot access Google with whichever DoH you configured, then your system seems to have no Internet access at all. Does pinging or `curl`ing the plain Google IP `141.251.220.14` work?

What I cannot derive from your info is whether you had WireGuard enabled or not in which case. I guess all tests were with VPN enabled, while with VPN disabled (and DoH) it works? If Internet access itself is broken once the VPN is enabled, then probably IP forwarding is missing at the VPN server? Can you check this on the VPN server system:
```sh
sysctl net.ipv4.conf.wg0.forwarding net.ipv4.conf.eth0.forwarding
iptables -L FORWARD
iptables -L POSTROUTING -t nat
```

However, this did work before (all but pornhub), didn't it? Probably best to test from home LAN, to take out differences of the coffee shop Internet for now.

yuukiAme, 2024-05-07:
> Hmm, yes, the "system" uses the default DNS from the ISP or whatever you configured in your LAN. However, with DoH enabled, the browser overrides this/does not use the system DNS. Hence if you cannot access Google with whichever DoH you configured, then your system seems to have no Internet access at all. Does pinging or curling the plain Google IP 141.251.220.14 work?

Yeah. I couldn't figure it out either. So I rebooted the Macbook just to be sure nothing was misconfigured. 

Steps I did:
1. removed the DNS entry again from the conf file and saved the file 
2. change Firefox Settings > DNS over HTTPS > Max protection > Cloudflare

outcome:
- I have internet again.
- google.com works
- nslookup google works
- curl 142.250.207.78 works > https://www.google.com
- local gateway works in the browser

it might have been a bad cache or improper setting during the whole up/down from the wg0 interface.

> If Internet access itself is broken once the VPN is enabled, then probably IP forwarding is missing at the VPN server? Can you check this on the VPN server system:

```
root@DietPi:~# sysctl net.ipv4.conf.wg0.forwarding net.ipv4.conf.eth0.forwarding
iptables -L FORWARD
iptables -L POSTROUTING -t nat
net.ipv4.conf.wg0.forwarding = 1
net.ipv4.conf.eth0.forwarding = 1
Chain FORWARD (policy ACCEPT)
target     prot opt source               destination
Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination
MASQUERADE  all  --  10.9.82.0/24         anywhere             /* wireguard-nat-rule */
```

I don't see the `wg` you're referring to in the output. But I can confirm that `10.9.82.0/24` is used by my PiVPN WG

I'm back at home. I will use my secondary WAN (no Pi-hole, unbound DoT, AGH) as an outside network to VPN if you want. Or should I use the 1st network (with Pi-hole, etc) to test the VPN and `www.pornhub.com` again?

yuukiAme, 2024-05-07:
Extra comment related to this test above

> Steps I did:

> removed the DNS entry again from the conf file and saved the file
change Firefox Settings > DNS over HTTPS > Max protection > Cloudflare
outcome:

> I have internet again.
> google.com works
> nslookup google works
> curl 142.250.207.78 works > https://www.google.com/
> local gateway works in the browser

All this over PiVPN Wireguard without the DNS entry in the conf file.

`nslookup www.pornhub.com` 
Address: 127.0.0.1 
which is blocked by default DNS (expected)

`nslookup store.steampowered.com`
Address: 127.0.0.1 

`nslookup xhamster.com`
Address: 127.0.0.1 

Firefox browser - enabled Max Protection > Cloudflare

|website|status|
|---|---|
|pornhub.com|timed out|
|xhamster.com|working as intended|
|store.steampowered.com|working as intended|

So my ISP really did something to `pornhub.com`

MichaIng, 2024-05-07:
Okay, so now everything but pornhub works with VPN enabled and DoH. And when you disable the VPN now, can pornhub be accessed via browser?

What I want to find out is whether the VPN alone, taking DNS out of the game, has an effect on whether pornhub can be accessed or not.

`wg` is a command from WireGuard to see which clients was or is connected.

yuukiAme, 2024-05-07:
>  And when you disable the VPN now, can pornhub be accessed via browser?

Steps I did:

`sudo wg-quick down wg0` - so VPN is down.

`nslookup google.com` to confirm it's the DNS switched back to Default DNS by ISP. It did and returned `142.250.66.142`

Good. I have internet.

Back to Firefox with DNS over HTTPS > Cloudflare

outcome - it timed out and never loaded on Firefox browser.

Just to be sure, I checked with `xhamster.com` and `store.steampowered.com`. They both working as intended in the browser.

as for the `wg` command, I turned on Wireguard again

ran the command, output below

```
sh-3.2# wg
interface: utun2
  public key: +REDACTED_LONG_STRING=
  private key: (hidden)
  listening port: 65446

peer: REDACTED_LONG_STRING=
  preshared key: (hidden)
  endpoint: <PUBLIC IP>:51820
  allowed ips: 0.0.0.0/0, ::/0
  latest handshake: 12 seconds ago
  transfer: 40.92 KiB received, 20.14 KiB sent

```

yuukiAme, 2024-05-07:
> so pornhub is not accessible on the Mac?

True. But if I use Proton VPN, it will work. 

> Was it every accessible there?

Not sure what this comment means.

> we should probably compare Mac and PC, as the PC is the only device where pornhub was ever accessible?

I can confirm that my Xiaomi (Android) and Samsung Tablet (Android) did open `www.pornhub.com` in Firefox browser on **BOTH** LAN and VPN connection. Work just like my PC (Windows).

I only have a laptop that is a Mac and it uses similar commands to Linux and I can move it around to different locations so that's why I chose it to test out VPN vs no VPN.

MichaIng, 2024-05-07:
> True. But if I use Proton VPN, it will work.

Makes sense, as your ISP then does not see any traffic from your home to pornhub (IP).

> Not sure what this comment means.

I mean whether pornhub was ever accessible on the Mac (when not using ProtonVPN or any other public provider).

> I only have a laptop that is a Mac and it uses similar commands to Linux and I can move it around to different locations so that's why I chose it to test out VPN vs no VPN.

Good, so we know the (home) VPN is not the reason for the issue, but it also does not prevent it. And encrypted DNS does not resolve it either, since it is the HTTP(S) connection itself which fails. And for now, based on earlier `curl` and `openssl` output, pornhub themselves get and answer the request, but the TLS handshake it is somehow not successful.

EDIT: Okay, some more devices are affected, interesting that one Android works, the other one not. Also the DietPi server is affected, getting a 404 response with `curl`, instead of a redirect. However, so we need to find out what the difference between requests from those devices is.

Can you run the traceroute on Windows like this:
```sh
tracert www.pornhub.com
```
And on the Mac and/or DietPi like this:
```sh
apt install mtr
mtr www.pornhub.com
```
and paste the results? I want to see whether the routes differ at any point. You can mask IPs/hostnames from your LAN, of course.

`mtr` instead of `traceroute` on Linux, since in my case the latter has issues finding the route to pornhub, while `mtr` and `tracert` (on Windows) succeed, and `curl` delivers the fill pornhub HTML document successfully.

yuukiAme, 2024-05-07:
> I mean whether pornhub was ever accessible on the Mac (when not using ProtonVPN or any other public provider).

Never. I hardly ever use this laptop. It's an older Macbook. It has like 4 GB of RAM and Intel duo cores. Super slow. So I never used it to access `pornhub.com` before yesterday.

> traceroute on Windows 

```
C:\Users\PC>tracert www.pornhub.com

Tracing route to pornhub.com [66.254.114.41]
over a maximum of 30 hops:

  1    <1 ms    <1 ms    <1 ms  tplinkwifi.net [192.168.16.1]
  2    21 ms    19 ms    19 ms  O11 [2x.xx.xxx.xxx]
  3     2 ms     1 ms     1 ms  10.255.39.245
  4    21 ms    21 ms    21 ms  O11 [2x.xx.xxx.xxx]
  5    20 ms    24 ms    22 ms  O11 [2x.xx.xxx.xxx]
  6    21 ms    20 ms    20 ms  O11 [2x.xx.xxx.xxx]
  7    46 ms    44 ms    44 ms  O11 [2x.xx.xxx.xxx]
  8    40 ms    40 ms    75 ms  125.xxx.xxx.xxx.isp.domain.com  [125.xxx.xxx.xxx]
  9    39 ms    39 ms    39 ms  213.248.97.50
 10     *        *        *     Request timed out.
 11    40 ms    40 ms    40 ms  62.115.137.243
 12    76 ms    76 ms    76 ms  hnk-b4-link.ip.twelve99.net [62.115.112.223]
 13   245 ms   245 ms   245 ms  tky-b3-link.ip.twelve99.net [62.115.134.187]
 14   242 ms   240 ms   241 ms  lax-b23-link.ip.twelve99.net [62.115.137.108]
 15   241 ms   240 ms   240 ms  lax-b22-link.ip.twelve99.net [62.115.143.39]
 16   245 ms   245 ms   245 ms  haproxy-ic-328269.ip.twelve99-cust.net [213.248.75.179]
 17   230 ms   230 ms   229 ms  cust-reflected-svc11502.ip.reflected.net [64.210.157.119]
 18   245 ms   245 ms   244 ms  reflectededge.reflected.net [66.254.114.41]

Trace complete.
```

> mtr www.pornhub.com

Wow. that's a cool application/software. I like it a lot. Thanks for introducing me to `mtr`.

output

```
DietPi (192.168.16.2) -> www.pornhub.com (66.254.114.41)                                       2024-05-08T02:38:53+0700
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                                               Packets               Pings
 Host                                                                        Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. 192.168.16.1                                                              0.0%     5    0.4   0.6   0.4   0.6   0.1
 2. localhost                                                                 0.0%     5    2.0   1.8   1.4   2.0   0.2
 3. 10.255.39.245                                                             0.0%     5    1.7   1.9   1.6   2.6   0.4
 4. localhost                                                                20.0%     5   20.1  21.5  20.1  24.6   2.1
 5. localhost                                                                 0.0%     5   22.0  24.1  20.7  27.5   3.0
 6. localhost                                                                 0.0%     4   20.3  20.5  20.3  20.9   0.3
 7. localhost                                                                 0.0%     4   44.4  44.8  44.4  45.0   0.3
 8. 125.xxx.xxx.xxx.isp.domain.com                                           0.0%     4   40.1  47.8  40.1  70.0  14.8
 9. (waiting for reply)
10. (waiting for reply)
11. 62.115.137.243                                                            0.0%     4   40.3  40.3  40.0  40.5   0.2
12. hnk-b4-link.ip.twelve99.net                                               0.0%     4   76.6  76.9  76.5  77.2   0.4
13. tky-b3-link.ip.twelve99.net                                               0.0%     4  245.7 245.7 245.6 245.8   0.1
14. lax-b23-link.ip.twelve99.net                                              0.0%     4  241.1 241.1 241.0 241.2   0.1
15. lax-b22-link.ip.twelve99.net                                              0.0%     4  245.6 245.4 245.2 245.6   0.2
16. haproxy-ic-328269.ip.twelve99-cust.net                                    0.0%     4  245.8 245.7 245.4 246.1   0.3
17. cust-reflected-svc11502.ip.reflected.net                                  0.0%     4  229.9 232.6 229.8 240.3   5.2
18. reflectededge.reflected.net
```

They both seem to work.

yuukiAme, 2024-05-07:
After `brew install mtr` and dealing with `mtr command not found` - I had to create ~/.zshrc to deal with it. I'm good

output for Macbook - without VPN on Pi-Hole network

```
mac.192 (192.168.16.32) -> www.pornhub.com (66.254.114.41)       2024-05-08T02:50:42
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                      Packets               Pings
 Host                                               Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. tplinkwifi.net                                   0.0%    16    2.2   1.6   1.2   2.3   0.3
 2. localhost                                        0.0%    16    3.2  13.9   2.5  62.3  16.9
 3. 10.255.39.245                                    0.0%    16    3.3   3.2   2.6   4.9   0.5
 4. localhost                                        0.0%    16   21.7  22.4  21.3  23.6   0.7
 5. localhost                                        0.0%    15   28.3  26.1  22.3  28.6   1.9
 6. localhost                                        0.0%    15   21.8  21.8  21.3  23.1   0.5
 7. localhost                                        0.0%    15   47.0  47.1  45.8  52.3   1.6
 8. 125.xxx.xxx.xxx.isp.domain.com                  0.0%    15   45.6  46.6  45.1  56.4   2.7
 9. 213.248.97.50                                   57.1%    15   44.9  45.1  44.9  45.2   0.1
10. (waiting for reply)
11. 62.115.137.243                                   0.0%    15   45.2  42.0  41.0  45.3   1.4
12. hnk-b4-link.ip.twelve99.net                      0.0%    15   77.5  78.5  77.4  84.4   1.8
13. tky-b3-link.ip.twelve99.net                      6.7%    15  246.4 243.5 242.4 247.4   1.7
14. lax-b23-link.ip.twelve99.net                     6.7%    15  246.2 246.4 245.9 247.2   0.4
15. lax-b22-link.ip.twelve99.net                     0.0%    15  242.9 242.2 241.5 243.0   0.4
16. haproxy-ic-328269.ip.twelve99-cust.net           0.0%    15  247.7 250.3 246.1 276.6   8.0
17. cust-reflected-svc11502.ip.reflected.net         0.0%    15  226.3 228.3 226.3 234.9   3.1
18. reflectededge.reflected.net                      0.0%    15  246.0 243.0 241.8 246.3   1.6

```

output for Macbook - with VPN on Pi-Hole network

```
mac.192 (10.9.82.3) -> www.pornhub.com (66.254.114.41)              2024-05-08T02:51:57
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                         Packets               Pings
 Host                                                  Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. pi.hole                                             0.0%    12    2.0   2.5   2.0   3.2   0.4
 2. tplinkwifi.net                                      0.0%    12    3.1   2.7   2.3   3.5   0.3
 3. localhost                                           0.0%    12    3.8   5.5   3.6  15.7   3.4
 4. 10.255.39.245                                       0.0%    12    4.2   4.2   3.7   4.9   0.4
 5. localhost                                           0.0%    12   24.3  23.6  22.1  24.6   0.7
 6. localhost                                           0.0%    12   29.1  27.0  22.7  29.9   1.9
 7. localhost                                           0.0%    12   23.1  23.0  22.4  23.7   0.4
 8. localhost                                           0.0%    11   46.8  47.5  46.1  48.9   0.9
 9. 125.xxx.xxx.xxx.isp.domain.com                     0.0%    11   42.2  42.9  41.7  43.9   0.7
10. 213.248.97.50                                      90.0%    11   46.0  46.0  46.0  46.0   0.0
11. (waiting for reply)
12. 62.115.137.243                                      0.0%    11   49.8  47.8  46.6  52.1   1.7
13. hnk-b4-link.ip.twelve99.net                         0.0%    11   78.7  80.6  78.5  90.7   3.8
14. tky-b3-link.ip.twelve99.net                         0.0%    11  257.1 249.0 247.6 257.1   2.7
15. lax-b23-link.ip.twelve99.net                        0.0%    11  245.6 243.6 242.9 245.6   0.7
16. lax-b22-link.ip.twelve99.net                        0.0%    11  243.1 243.4 242.9 243.9   0.3
17. haproxy-ic-328269.ip.twelve99-cust.net              0.0%    11  281.5 251.3 247.6 281.5  10.1
18. cust-reflected-svc11502.ip.reflected.net            0.0%    11  228.8 228.1 227.2 229.4   0.6
19. reflectededge.reflected.net                         0.0%    11  247.3 247.2 247.0 247.7   0.2

```

yuukiAme, 2024-05-07:
Extra tests from Macbook

Test 3: mtr number 3 on network number 2 (not Pi-hole) no VPN - default DNS

```
192.168.1.106 (127.0.0.1) -> www.pornhub.com (127.0.0.1)                          2024-05-08T03:00:37
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                                  Packets               Pings
 Host                                                           Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. localhost                                                    0.0%    15    0.3   0.2   0.1   0.5   0.1

```

Test 4: mtr number 4 network number 2 with vpn no dns entry in conf file 

```
192.168.1.106 (127.0.0.1) -> www.pornhub.com (127.0.0.1)                            2024-05-08T03:01:31
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                                    Packets               Pings
 Host                                                             Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. localhost                                                      0.0%    11    0.1   0.1   0.1   0.2   0.0

```

Test 5 : mtr number 5 network  number 2 with vpn with dns entry `10.9.82.1`

```
mac.local (10.9.82.3) -> www.pornhub.com (66.254.114.41)               2024-05-08T03:02:46
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                                                    Packets               Pings
 Host                                                             Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. pi.hole                                                        0.0%    11    3.4   3.8   3.4   4.7   0.5
 2. tplinkwifi.net                                                 0.0%    11    4.2   4.2   3.5   5.1   0.4
 3. localhost                                                      0.0%    11    4.8  12.2   4.4  42.8  11.6
 4. 10.255.39.245                                                  0.0%    11    5.3   5.5   5.0   7.2   0.6
 5. localhost                                                      0.0%    11   24.0  24.6  23.6  25.9   0.9
 6. localhost                                                      0.0%    11   26.6  27.6  23.5  32.3   2.6
 7. localhost                                                      0.0%    11   23.9  24.2  23.5  26.0   0.7
 8. localhost                                                      0.0%    10   44.6  44.9  44.0  45.7   0.5
 9. 125.xxx.xxx.xxx.isp.domain.com                                 0.0%    10   49.4  51.0  47.6  71.9   7.4
10. 213.248.97.50                                                 88.9%    10   42.9  42.9  42.9  42.9   0.0
11. (waiting for reply)
12. 62.115.137.243                                                 0.0%    10   44.8  45.5  43.6  49.4   2.3
13. hnk-b4-link.ip.twelve99.net                                    0.0%    10   82.2  81.0  80.0  85.0   1.5
14. tky-b3-link.ip.twelve99.net                                    0.0%    10  249.0 249.2 248.5 250.0   0.5
15. lax-b23-link.ip.twelve99.net                                   0.0%    10  248.7 248.7 248.5 249.2   0.2
16. lax-b22-link.ip.twelve99.net                                   0.0%    10  248.8 249.0 248.4 250.2   0.5
17. haproxy-ic-328269.ip.twelve99-cust.net                         0.0%    10  244.5 244.9 244.3 246.0   0.5
18. cust-reflected-svc11502.ip.reflected.net                       0.0%    10  229.5 234.4 228.8 262.5  10.7
19. reflectededge.reflected.net                                    0.0%    10  248.0 248.3 247.9 249.1   0.4

```

MichaIng, 2024-05-07:
Ah sorry, right, on Mac of course the `apt install` was nonsense. However, not bad to have the result from the DietPi/Linux system as well.

Yeah, `mtr` seems to be better than the classic `traceroute`, not only as it continues to monitor all nodes to get averages, but also since at least in this particular case, from the systems I can test this, `traceroute` was simply not able to find the route to pornhub.

ISP DNS blocks *.pornhub.com, hence all ends at localhost, makes sense. With any other DNS, the VPN adds the VPN/Pi-hole server as first hop, otherwise has no effect. And also the routes on all devices without VPN are identical. So that is not the difference. Probably it is not even the ISP which blocks this in a way, but the TLS libraries used on those systems somehow have an issue in your particular case. But I am pretty much out of ideas how to further debug.

Just one last test, on the DietPi system, which seem to have been affected as well. Can you test again with `curl` as well as `wget`? Both use different TLS libraries. If not the case, use Pi-hole for DNS resolving, as we know it is otherwise blocked by ISP DNS:
```sh
cp -a /etc/resolv.conf{,.bak}
echo 'nameserver 127.0.0.1' > /etc/resolv.conf
curl -L https://www.pornhub.com
wget -O- https://www.pornhub.com
mv /etc/resolv.conf{.bak,}
```
In case of success, both return the full HTML document, which is huge, so do not wonder, and no need to paste that 😄. If it ends with `</html>`, it was a success, that is all we need to know.

Not sure whether someone on our forum has further ideas how to debug, or in general how requests can be (made) fail in such a way. Otherwise, probably people on some StackExchange site, like https://superuser.com/, have an idea. In case, it makes sense to leave the (home) VPN out of the game. It was a reasonable idea to test whether it helps, but we showed that it is irrelevant in regards of the issue. But that using an encrypted public DNS is required in any case, as the ISP DNS blocks it, is of course important, and that using a public VPN provider fixes the issue on the affected systems, is relevant as well. So it is somehow these particular systems or clients sending the HTTP(S) request through this particular ISP route, having issues, while other systems/clients, sending the same request through the same route, succeed.

An interesting case, so in case you're doing to further debug, I am interested about the outcome. You can ping me on superuser as well (same username).

[15 more replies]

closed 2024-05-11 by yuukiAme