MichaIng/DietPi #7060: Open Beta v9.4 | Please help testing and hardening our upcoming release

MichaIng, 2024-05-06:
RC version | v9.4.0
----------- | -------
v9.4.0 PR | #7051
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2024-05-11**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av9.4
____
### Beta v9.4.0
_(2024-05-06)_

#### New images
- ROCK 4C Plus | Since our Radxa ROCK 4 image has become incompatible with the ROCK 4C Plus, we provide a new image for this particular variant. Many thanks to @lfiorini and @arpegius5555 for reporting this issue: https://github.com/MichaIng/DietPi/issues/6593, https://github.com/MichaIng/DietPi/issues/6935
- Orange Pi 3 LTS | Added support for this Allwinner H6 based SBC.
- Radxa ZERO 3 | Added support for this Rockchip RK3566 based SBC, the 3E variant with Ethernet as well as the 3W variant with onboard WiFi: #7024
- Orange Pi Zero 2W | Support for this small form factor SBC with Allwinner H618 SoC was added: #6827

#### Enhancements
- General | Our scripts do now internally enforce the default `umask` 0022. Many config and install options rely on this, hence it can cause issues when e.g. 0027 (deny read access for "other" users) has been applied on the parent shell. This affects only the (sub)shell of our scripts, while the current and default `umask` of parent and of shells/consoles remains untouched.
- NanoPi R5C | New images, or when flashing the new bootloader binary via `dietpi-config` > Advanced Options > Update MMC bootloader, support M.2 WiFi modules.
- Orange Pi 3B | Added an option for updating the SPI bootloader via `dietpi-config` > Advanced Options > Update SPI bootloader.
- DietPi-Software | Snapcast: 64-bit ARM and Debian Bookworm + Trixie will have the now available packages from Snapcast installed, instead of those from the Debian repository.

#### Bug fixes
- NanoPi NEO3 | Resolved an issue where the Ethernet adapter was not available after reboots. Many thanks to @murraythegoz for reporting this issue: https://github.com/MichaIng/DietPi/issues/6987
- DietPi-VPN | Resolved an issue where installing the IPVanish config failed, due to a changed URL, and startup failed on Bookworm systems since their configs contain a deprecated option. Many thanks to @DreamPhreak and @Michael-Robson for reporting this issue: https://github.com/MichaIng/DietPi/issues/7043
- DietPi-Imager | Resolved an issue where the imager could have failed on GPT partitioned images, if either the source image size did not leave space for the GPT backup partition table, or the first usable LBA/sector was above 34. The GPT backup partition table is now created only at the end of the image generation, not anymore additionally at the start, and the required size it takes is correctly obtained. Many thanks to @SelfhostedPro and @disablewong for reporting this issue: https://github.com/MichaIng/DietPi/issues/7024, https://dietpi.com/forum/t/18035
- DietPi-Software | Snapcast: Resolved an issue where version 0.27.0 was installed, because since version 0.28.0, Snapcast is provided with client and server packages wrapped into one archive, instead having individual downloads for each package.
- DietPi-Software | OctoPrint: Resolved an issue where the installation failed on RISC-V and ARMv6/7 Bullseye systems, due to changed dependencies.
- DietPi-Software | X.Org X Server: Resolved an issue where an important config file might have been missing on RPi 5 and potentially Amlogic S905 SBCs, if the config directory `/etc/X11/xorg.conf.d` did not exist and was not created as part of the related APT packages. Many thanks to @rmscode for reporting this issue: https://dietpi.com/forum/t/19963

matttttttttttttt, 2024-05-06:
Thank you for opening the beta to v9.4 :-) 

When trying to apply this update on a RPI5 in run into the following issues:

MichaIng, 2024-05-07:
Many thanks for reporting. A typo in the pre-patch. Fixed with #7061. Please try again.

matttttttttttttt, 2024-05-07:
Thank you Michael for the quick fix.

How can I get the new version? When applying the update I still have the same issue. Should I do something special to make it repopulate again?

- Martin

MichaIng, 2024-05-07:
Just rerun the update. Sometimes it takes a minute or two for the `raw.githubusercontent.com` to be updated after a commit, but it should be the case now. In case check like this, whether the file still contains the faulty package name:
```sh
curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/beta/.update/pre-patches' | grep 'tiny-initamfs'
```
If there is no output, it is good, else give it another few minutes.

matttttttttttttt, 2024-05-07:
It worked now with just dietpi-update! Thank you so much, Michael :)

oxoocoffee, 2024-05-07:
I am running v9.3.0.  `dietpi-update`  is not finding the bata. on rPI4 and 5. Do I have to download the entire fresh image?

Joulinar, 2024-05-07:
you need to switch to BETA branch as describe above on first post.

arpegius5555, 2024-05-08:
Starting to test on a Rock4C+ now. 
Changed to DEV_GITBRANCH=beta

- Booting now. I can see the blue LED coming up and also HDMI working out of initial boot, with no manual intervention
- After going through initial setup, I did dietpi-config > network adapter > WiFi is available!
- I am able to get an IP on wlan0

I was also able to install pihole and pivpn as a test. Anything else you need me to check on?

DreamPhreak, 2024-05-09:
Tested dietpi-vpn/ipvanish setup and confirming the fixes worked seamlessly. Thank you! 

[image]

MichaIng, 2024-05-09:
Many thanks for testing 👍.

MichaIng, 2024-05-11:
#### Beta v9.4.1 has been merged: #7062

MichaIng, 2024-05-12:
#### Beta v9.4.2 has been merged: #7068

MichaIng, 2024-05-12:
#### DietPi v9.4 has been released: #7063
Many thanks to all testers ❤️ !

closed 2024-05-12 by MichaIng