MichaIng/DietPi #7072: Bare minimum backup and Better explanation of the backup process

hllhll, 2024-05-14:
It's awesome that DietPi have internal SW mechanism, in my mind, it means, also, that I won't need to backup the entire software data, configuration, binaries and resource, just backup what is needed for the individual software/ service. 
As DietPi makes use of docker for services redundant, I thought this would also address that (I'm a newbie DietPi user who came from rpios+containers).

Using docker with a proper configuration, I can just include docker-compose file and mount the config, log and other userdata locally and just back them up, w/o needing to backup the base image. on a new device I just docker compose up and everything pulled etc.

I thought I could use DietPi also for a similar behavior. 
When looking into what DietPi backs up, I couldn't find any detailed information about what exactly is backed up.
Tried the filter feature, then I get this
```
# DietPi-Backup include/exclude filter

# Prefix "-" exclude items, "+" include items which would match a wildcard exclude rule.
# Suffix "/" match directories only, no files or symlinks.
# Wildcard "*" matches any item name or part of it, but "/dir/*" does not match the dir itself.
# Since the list is processed from top to bottom and the first match defines the applied rule:
# - Includes need to be defined before their wildcard exclude rule.
# - Excludes need to be defined before their wildcard include rule.
# Symlinks are handled as such and never processed recursively.
# Excluded directories are not processed recursively, so contained items cannot be included.
# Included directories are not processed recursively, so contained items cannot be excluded.
# Hence, to include items within an excluded directory:
# - Do not exclude the directory itself, but contained items via wildcard.
# - Define includes first, to override the wildcard exclude rule.
# - See the below default rules, how we exclude all items below /mnt/
#   but include the dietpi_userdata directory, if it is no symlink.
# To prevent loops, the backup target dir, log and config are excluded internally.

+ /mnt/dietpi_userdata/
- /mnt/*
- /media/
```
English is indeed not my native language, but I'm proficient in it, I read this 5-8 times in different days, and I still can't understand anything that is written above...

Okay I thought this would just back up user data, maybe some other minimum stuff?
I tried a backup and I can see that it backs up files all over the place \~200k of them (while I probably edited only\~ 20 or less, running a few home servers on a rpi; ~50k of these files are docker filesystem files which is okay I guess I can just ignore them as I'm using bind mount for my configuration files found in dietpi_userdata)

So what I'd like to see:
1. Enhanced explanation about how the update process works, what is backed up out of the box
2. Better explanation to how to configure the filter file
3. "Minimum backup" feature; As DietPi-Softwrae integration is one-by one, is dietpi aware of the individual confutation options for each software? of not, can this be included as part of onboarding new software?
    1. There are some examples that I can see that DietPi-Software is aware HomeAssistant: Aware of the "configuration" folder; docker - Aware of docker-data.  So for HA for instance, there is just a need to backup the configuration folder, that's it, no other files are actually required.

MichaIng, 2024-05-14:
> 1. Enhanced explanation about how the update process works, what is backed up out of the box

Does the "Help!" text not make it sufficiently clear, that it is a (full) system backup, and not (just) user data? Otherwise, please suggest how to rephrase this.

> 2. Better explanation to how to configure the filter file

I really tried my best, but the `rsync` filter rules are hard to explain. Please read the man page, see whether you understand it better, and do suggestions about how to better explain the most relevant info in a few sentences: https://manpages.debian.org/rsync#FILTER_RULES
At the top right of that page, you can also switch the language, but it does not seem to have many translations.

Adding this link to our text makes sense, at least.

> 3. "Minimum backup" feature

As said, `dietpi-backup` is meant to restore a previous snapshot of the base system, in case something is broken. Naturally, this cannot be "minimal" or not minimal, but you simply need everything, except end user software/media data, maybe. These are btw excluded, if you moved `/mnt/dietpi_userdata` to an external drive, _as, like our text explains, `/mnt/dietpi_userdata/` does not match a symlink, only a directory_. A transparent toggle for this probably makes sense.

To backup individual directories, software data/configs etc, have a look at `dietpi-sync`.

hllhll, 2024-05-14:
TY
> Does the "Help!" text not make it sufficiently clear, that it is a (full) system backup, and not (just) user data? Otherwise, please suggest how to rephrase this.

I'm not all that keen on devops, I guess for me the word "full" would have helped.
So dietpi-backup is not tailored for dietpi (as I would have thought), this is more a generic tool, in this case.

> I really tried my best, but the `rsync` filter rules are hard to explain. Please read the man page, see whether you understand it better, and do suggestions about how to better explain the most [...]

Maybe it would help to add one row that by default everything is included?
> As said, `dietpi-backup` is meant to restore a previous snapshot of the base system, in case something is broken. Naturally, this cannot be "minimal" or not minimal, but you simply need everything, [...]

Don't you think that given that dietpi software onboarding is individual per-software it makes sense to optionally include some app-specific backup process (either through sync or backup or otherwise)? I'm looking over some of the dietpi-install scripts and there is occasionally specific patching for service configuration files by default, that means that when integrating this information does somewhat exists in the mind of the integrator, it's petty that it goes to waste, isn't it?
In this case, Dietpi doesn't make docker containers/using bind mount redundant because this is a feature supplied with docker when tailoring a specific image (which often exists, some examples: [snapcast](https://github.com/sweisgerber/docker-snapcast), [homeassistant](https://www.home-assistant.io/installation/linux#platform-installation) having their configuration files bound to a specific user-supplied path (that may be /mnt/dietpi_userdata/<app> for instance in our case) )

MichaIng, 2024-05-14:
> So dietpi-backup is not tailored for dietpi

It is just not what you though it is. It is a very minimal light (system!) backup solution, which creates a system backup that protects you against setup or update failures and mistakes, similar to how system recovery on Windows, time machine on macOS, or similar embedded (system) backup solutions work. In that way. `dietpi-sync` on the other hand seems to be exactly what you want, isn't it? Though it allows to sync one directory only.

> Don't you think that given that dietpi software onboarding is individual per-software it makes sense to optionally include some app-specific backup process (either through sync or backup or otherwise)?

Sure, that would be awesome to have, but huge work to implement and maintain all relevant config/data directories for every of the ~200 (okay may ~100 where you can reasonably backup something) software options. Maybe a new tool, like `dietpi-software-backup` or so, or a sub-menu for `dietpi-backup`. Just to give an impression about how this could look like, here is a script I write long ago to backup ownCloud and Nextcloud, so that it can be migrated form one to another DietPi system: https://github.com/MichaIng/DietPi/blob/master/.meta/dietpi-cloud-migration

Surely this can be unified for many software titles, like an array containing install dir, data/config dir, and in case (MariaDB/PostgreSQL) database name.