MichaIng/DietPi #7073: dietpi-software: snapweb snapcast server invalid web root

hllhll, 2024-05-14:
#### Required Information
- DietPi version - 9.3.0
- Distro version - bookworm
- Kernel version - Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model - RPi 4 Model B (aarch64)

#### Additional Information (if applicable)
- DietPi-Software: snapcast server
- Software installed freshly, The issue most likely reproducible with a fresh install (as I did)

#### Steps to reproduce
Install snapcast server using dietpi-software
navigate to http://\<host\>:1780

#### Expected behaviour
[snapweb](https://github.com/badaix/snapcast?tab=readme-ov-file#webapp) should load

#### Actual behaviour
Web page reports an error:
```
The resource '/' was not found.
```

#### Extra details
Looks like the path to the www document root should contain a file named `index.html`; Default path configured during installation is  `/usr/share/snapserver/snapweb` which doesn't exist. index file seem to be in `/usr/share/snapserver/index.html`

I Will probably fix it on my own and perform a PR, eventually :)

MichaIng, 2024-05-14:
Hmm, snapweb is not included in this package anymore:
```console
root@VM-Bookworm:~# l /usr/share/snapserver/
total 8.0K
-rw-r--r-- 1 root root 1.7K Dec 22  2021 index.html
drwxr-xr-x 2 root root 4.0K May 14 17:58 plug-ins
```
Using `/usr/share/snapserver` just shows a generic default page, stating
> The Snapserver installation should include a copy of [Snapweb](https://github.com/badaix/snapweb), located in /usr/share/snapserver/snapweb/

which is obviously not true. Not sure whether this is intended or a packaging bug in the latest version.

hllhll, 2024-05-14:
Under my setup, which package would the dietpi-software installer use? 
I've looked into [snapcast_0.28.0_amd64-debian-bookworm.zip](https://github.com/badaix/snapcast/releases/download/v0.28.0/snapcast_0.28.0_amd64-debian-bookworm.zip) and it's there in usr/share/snapserver/snapweb)

MichaIng, 2024-05-14:
Ah lol, my testing VM was at DietPi v9.3 as well. Please update to latest version and reinstall it:
```sh
dietpi-software reinstall 191
```
The latest Snapcast release added packages which support for Debian Bookworm. Before that, we installed the packages from Debian themselves on Bookworm, which indeed do not include snapweb (anymore).

hllhll, 2024-05-14:
Due to some wired permission issues, I hade to uninstall and reinstall in addition to that, but now it works, Thanks!

closed 2024-05-14 by hllhll

MichaIng, 2024-05-14:
Since it is still relevant for RISC-V, I just fixed it, installing snapweb in those cases where we install the Debian package: https://github.com/MichaIng/DietPi/commit/2b34d77

Now I understand why it was disabled OOTB in Debian's `snapsever.conf`. It was removed from the Debian package, since it relies (or relied) on a JavaScript library, for which the source code is missing. Libflac.js to be precise. Checking back at current sources, this is installed as Node dependency: https://github.com/badaix/snapweb/blob/develop/package.json
And the source code is perfectly available: https://www.npmjs.com/package/libflacjs/v/5.0.0 => https://github.com/mmig/libflac.js

Indeed this changed in the meantime: https://github.com/badaix/snapweb/pull/46
So Debian could re-add it to their package.

MichaIng, 2024-05-14:
I also found the weird permissions issue. It is since the Debian package uses the `_snapserver` user, and the badaix package uses the `snapserver` user. Weird unnecessary difference, IMO. When upgrading from the Debian to badaix package or the other way round, the other respective user will remain on the system, also after purging the package.

Both addressed here: https://github.com/MichaIng/DietPi/commit/0091219