MichaIng/DietPi #7078: dietpi-letsencrypt

fagron, 2024-05-16:
#### Details:
- Date           | Thu May 16 17:18:25 CEST 2024
- DietPi version | v9.4.2 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://dietpi.com/apt bookworm InRelease
Hit:2 https://dietpi.com/apt all InRelease
Hit:3 https://archive.raspberrypi.com/debian bookworm InRelease
Hit:4 https://deb.debian.org/debian bookworm InRelease
Hit:5 https://deb.debian.org/debian bookworm-updates InRelease
Get:6 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Hit:7 https://deb.debian.org/debian bookworm-backports InRelease
Reading package lists...
E: Release file for https://deb.debian.org/debian-security/dists/bookworm-security/InRelease is not valid yet (invalid for another 1h 24min 41s). Updates for this repository will not be applied.
```

MichaIng, 2024-05-16:
Your system time seems a little behind. Please force a sync and retry:
```sh
/boot/dietpi/func/run_ntpd 1
```

fagron, 2024-05-16:
Thank you, it has worked.

MichaIng, 2024-05-16:
Great. I'll mark this as closed, feel free to reopen when required.