MichaIng/DietPi #7080: Can't update Jellyfin

gioxx, 2024-05-17:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

I'm trying to banally update Jellyfin, since it's being reported as available when logging in to the server, but there's something I'm missing, it won't update and I'm getting it every time.
I then tried to follow the upgrade instructions suggested on the DietPi website (https://dietpi.com/docs/software/media/#__tabbed_27_4) but I still cannot come out victorious.

```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
```

Linux Jellyfin 5.10.0-29-amd64 # 1 SMP Debian 5.10.216-1 (2024-05-03) x86_64 GNU/Linux

#### Steps to reproduce

```
apt update
apt install jellyfin jellyfin-ffmpeg5
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://dietpi.com/apt bullseye InRelease
Hit:6 https://repo.jellyfin.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
jellyfin-ffmpeg5 is already the newest version (5.1.4-3-bullseye).
jellyfin-ffmpeg5 set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 jellyfin-ffmpeg6 : Conflicts: jellyfin-ffmpeg5 but 5.1.4-3-bullseye is to be installed
E: Unable to correct problems, you have held broken packages.
```

Is anyone able to tell me where I am so stupidly failing? 🥲 
Thanks!

MichaIng, 2024-05-18:
The error message says everything, isn't it 🙂?
> jellyfin-ffmpeg6 : Conflicts: jellyfin-ffmpeg5 but 5.1.4-3-bullseye is to be installed

You do not need to install `jellyfin-ffmpeg*` explicitly, the `jellyfin` meta package depends on the correct version of it:
```sh
apt-mark auto jellyfin-ffmpeg5
apt install jellyfin
```
This should offer you to uninstall `jellyfin-ffmpeg5` while having `jellyfin-ffmpeg6` as new dependency installed.

gioxx, 2024-05-20:
Ciao @MichaIng.
As mentioned above, I was sure it was a very dumb problem to solve but -- banally -- I had tried to run apt upgrade since I was presented with an upgrade to run when I logged into DietPi, and then I couldn't complete the operation because of ffmpeg5 as you noted. 

I have now uninstalled ffmpeg5 and correctly updated to version 6 and the latest Jellyfin package.

Thank you very much!

closed 2024-05-20 by gioxx

MichaIng, 2024-05-22:
Right, `apt upgrade` will keep back upgrades, if those would require the removal of another package, like `jellyfin-ffmpeg5` here. So an `apt full-upgrade` would be required. `apt install jellyfin` should however offer you the same dependency switch. I wonder why Jellyfin guys do not simply name it `jellyfin-ffmpeg5`, and have the FFmpeg version part of the package version string (seems to be the case already), to avoid this hassle.