MichaIng/DietPi #7085: Errors were encountered while processing:  raspi-firmware E: Sub-process /usr/bin/dpkg returned an error code (1)

systembetrieb, 2024-05-20:
#### Details:
- Date           | Mon May 20 02:16:41 CEST 2024
- DietPi version | v8.25.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux VPN-GW-MWR 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  cpio tiny-initramfs tiny-initramfs-core
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up raspi-firmware (1:1.20240424-1) ...
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Processing triggers for tiny-initramfs (0.1-5) ...
Errors were encountered while processing:
 raspi-firmware
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2024-05-20:
Did you install `raspi-firmware` manually? It should not be pulled anymore as dependency 🤔. Does this lead to any other removal?
```sh
apt autopurge raspi-firmware
```

systembetrieb, 2024-05-20:
OK 👍  ... check now ...

systembetrieb, 2024-05-20:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v9.4.2
[ INFO ] DietPi-Update | Latest version  : v9.4.2
[  OK  ] DietPi-Survey | Purging survey data
[  OK  ] DietPi-Update | sync
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : webmin
[  OK  ] DietPi-Services | restart : docker
[  OK  ] DietPi-Services | restart : cron

💯  :-*
MTHX

MichaIng, 2024-05-20:
Background: `raspi-firmware` is part of the new RPi kernel/firmware package stack, which requires a different boot partition mount and is hence incompatible with our (old) RPi images. Unintentionally, `rpi-eeprom` did pull it as dependency at first, but that was lifted. Hence I am wondering why this package was installed on your system in the first place.

About migrating to the new kernel/firmware package stack: #6676

MichaIng, 2025-06-02:
This issue is solved, right? Else, feel free to reopen.

closed 2025-06-02 by MichaIng

MichaIng, 2025-11-08:
Please do not manually migrate from old to new firmware stack, these steps are incomplete, and will cause issues. We have a migration option in `dietpi-config` advanced options for that, and DietPi updates trigger it automatically on RPi Bookworm systems.

But out of interest, I wonder why `raspi-firmware` was installed in the first place? It does not exist on Bullseye, and editing sources lists for a Bookworm upgrade should not install it automatically either. So you tried to install that package manually?