MichaIng/DietPi #7086: Error caused by update to v9.4.2 and by Dietpi-Software install attempt

SystemTheory, 2024-05-22:
#### Details:
- Date           | Wed May 22 10:01:30 EDT 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model A+ (aarch64) (ID=3)
- Kernel version | `Linux dpi3A 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Upgrade from v9.2.1 to v9.4.2 fails!
2. Dietpi-Software install fails!
#### Expected behaviour:

- ...
#### Actual behaviour:

- Getting the rred error message both cases.
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Get:5 https://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:6 https://deb.debian.org/debian bookworm-updates/main arm64 Packages.diff/Index [10.6 kB]
Get:7 https://deb.debian.org/debian bookworm-updates/main arm64 Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:8 https://deb.debian.org/debian-security bookworm-security/main arm64 Packages [153 kB]
Get:9 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages.diff/Index [4995 B]
Get:10 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Ign:10 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index
Get:11 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages.diff/Index [28.6 kB]
Get:12 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:13 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages T-2024-05-17-2022.39-F-2024-04-16-1405.22.pdiff [2238 B]
Get:14 https://deb.debian.org/debian bookworm-backports/main arm64 Packages [206 kB]
Get:15 https://archive.raspberrypi.com/debian bookworm/main arm64 Packages [413 kB]
Reading package lists...
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly
```

SystemTheory, 2024-05-22:
$ apt list --upgradable
Listing... Done
bsdutils/stable-security 1:2.38.1-5+deb12u1 arm64 [upgradable from: 1:2.38.1-5+b1]
dpkg/stable 1.22.6~bpo12+rpt2 arm64 [upgradable from: 1.21.22]
fdisk/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
libblkid1/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
libc-bin/stable 2.36-9+rpt2+deb12u7 arm64 [upgradable from: 2.36-9+rpt2+deb12u4]
libc-l10n/stable 2.36-9+rpt2+deb12u7 all [upgradable from: 2.36-9+rpt2+deb12u4]
libc6/stable 2.36-9+rpt2+deb12u7 arm64 [upgradable from: 2.36-9+rpt2+deb12u4]
libfdisk1/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
libmount1/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
libsmartcols1/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
libuuid1/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
locales/stable 2.36-9+rpt2+deb12u7 all [upgradable from: 2.36-9+rpt2+deb12u4]
mount/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
util-linux-extra/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]
util-linux/stable-security 2.38.1-5+deb12u1 arm64 [upgradable from: 2.38.1-5+b1]

$ apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Hit:5 https://archive.raspberrypi.com/debian bookworm InRelease
Get:6 https://deb.debian.org/debian bookworm-updates/main arm64 Packages.diff/Index [10.6 kB]
Get:7 https://deb.debian.org/debian bookworm-updates/main arm64 Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:8 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages.diff/Index [4995 B]
Get:9 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:9 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Ign:9 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index
Get:11 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:12 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages T-2024-05-17-2022.39-F-2024-04-16-1405.22.pdiff [2238 B]
Get:13 https://deb.debian.org/debian bookworm-backports/main arm64 Packages [206 kB]
Reading package lists... Done
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly
E: Method rred has died unexpectedly!
E: Sub-process rred received a segmentation fault.
E: Method /usr/lib/apt/methods/rred did not start correctly

$ apt upgrade

The upgrade command succeeds. There are no more upgradable packages. But then  running $ apt update produces the same error for rred died unexpectedly.

SystemTheory, 2024-05-22:
Sorry I hit the closed button by accident the interface is confusing.

SystemTheory, 2024-05-22:
Thank you. Now I am getting some key errors in the repositories?

root@dpi3A /root
$ cd /tmp

root@dpi3A /tmp
$ wget https://deb.debian.org/debian/pool/main/a/apt/apt_2.6.1_arm64.deb
--2024-05-22 12:21:18--  https://deb.debian.org/debian/pool/main/a/apt/apt_2.6.1_arm64.deb
Resolving deb.debian.org (deb.debian.org)... 151.101.118.132, 2a04:4e42:1c::644
Connecting to deb.debian.org (deb.debian.org)|151.101.118.132|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1310288 (1.2M) [application/vnd.debian.binary-package]
Saving to: ‘apt_2.6.1_arm64.deb’

apt_2.6.1_arm64.deb       100%[====================================>]   1.25M  2.79MB/s    in 0.4s

2024-05-22 12:21:19 (2.79 MB/s) - ‘apt_2.6.1_arm64.deb’ saved [1310288/1310288]

root@dpi3A /tmp
$ dpkg -i apt_2.6.1_arm64.deb
(Reading database ... 20516 files and directories currently installed.)
Preparing to unpack apt_2.6.1_arm64.deb ...
Unpacking apt (2.6.1) over (2.6.1) ...
Setting up apt (2.6.1) ...
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.36-9+rpt2+deb12u7) ...

root@dpi3A /tmp
$ rm apt_2.6.1_arm64.deb
rm: remove regular file 'apt_2.6.1_arm64.deb'? y
removed 'apt_2.6.1_arm64.deb'

root@dpi3A /tmp
$ cd

root@dpi3A /root
$ apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Hit:5 https://archive.raspberrypi.com/debian bookworm InRelease
Err:1 https://deb.debian.org/debian bookworm InRelease
  Unknown error executing apt-key
Err:2 https://deb.debian.org/debian bookworm-updates InRelease
  Unknown error executing apt-key
Err:3 https://deb.debian.org/debian-security bookworm-security InRelease
  Unknown error executing apt-key
Err:4 https://deb.debian.org/debian bookworm-backports InRelease
  Unknown error executing apt-key
Err:5 https://archive.raspberrypi.com/debian bookworm InRelease
  Unknown error executing apt-key
Fetched 112 kB in 4s (27.3 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.debian.org/debian bookworm InRelease: Unknown error executing apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.debian.org/debian bookworm-updates InRelease: Unknown error executing apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.debian.org/debian-security bookworm-security InRelease: Unknown error executing apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://deb.debian.org/debian bookworm-backports InRelease: Unknown error executing apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://archive.raspberrypi.com/debian bookworm InRelease: Unknown error executing apt-key
W: Failed to fetch https://deb.debian.org/debian/dists/bookworm/InRelease  Unknown error executing apt-key
W: Failed to fetch https://deb.debian.org/debian/dists/bookworm-updates/InRelease  Unknown error executing apt-key
W: Failed to fetch https://deb.debian.org/debian-security/dists/bookworm-security/InRelease  Unknown error executing apt-key
W: Failed to fetch https://deb.debian.org/debian/dists/bookworm-backports/InRelease  Unknown error executing apt-key
W: Failed to fetch https://archive.raspberrypi.com/debian/dists/bookworm/InRelease  Unknown error executing apt-key
W: Some index files failed to download. They have been ignored, or old ones used instead.

MichaIng, 2024-05-22:
Hmm, the SD card might be damaged (reached its EOL). `apt-key` is from the very same package, so unless some shared library is damaged, replacing the file obviously did not help or even cause a new issue.

Also check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

SystemTheory, 2024-05-22:
```dmesg -l 0,1,2,3```
[    0.883401] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.

MichaIng, 2024-05-22:
No filesystem or I/O errors, that is good. What does executing `apt-key` manually return?
```sh
apt-key
```

SystemTheory, 2024-05-22:
```apt-key```
Usage: apt-key [--keyring file] [command] [arguments]

Manage apt's list of trusted keys

  apt-key add <file>          - add the key contained in <file> ('-' for stdin)
  apt-key del <keyid>         - remove the key <keyid>
  apt-key export <keyid>      - output the key <keyid>
  apt-key exportall           - output all trusted keys
  apt-key update              - update keys using the keyring package
  apt-key net-update          - update keys using the network
  apt-key list                - list keys
  apt-key finger              - list fingerprints
  apt-key adv                 - pass advanced options to gpg (download key)

If no specific keyring file is given the command applies to all keyring files.

SystemTheory, 2024-05-22:
I am looking into these sources of information for a fix. Otherwise I am considering a fresh dietpi install on a new SD card:

https://stackoverflow.com/questions/68992799/warning-apt-key-is-deprecated-manage-keyring-files-in-trusted-gpg-d-instead

https://www.linuxuprising.com/2021/01/apt-key-is-deprecated-how-to-add.html

https://github.com/MX-Linux/checkaptgpg/blob/master/scripts/checkaptgpg

I am reluctant to use the MX Linux script as a possible "fix" because I don't have the chops to understand it. I don't really understand the other sources but they seem to be good advice.

MichaIng, 2024-05-22:
The manual usage of `apt-key` is indeed deprecated, but APT uses it internally to check repository signatures, and this, of course, does not fail if everything is in order.

So the CLI command generally works. Not sure how to get more information out of it, there is no verbose more or similar 🤔. Let's try to reinstall some more packages, which are used by APT for signature verification. Nasty that APT itself cannot be used for that, while it is broken:
```sh
cd /tmp
wget -O package.deb https://deb.debian.org/debian/pool/main/a/apt/libapt-pkg6.0_2.6.1_arm64.deb
dpkg -i package.deb
apt update
wget -O package.deb https://deb.debian.org/debian/pool/main/g/gnupg2/gpgv_2.2.40-1.1_arm64.deb
dpkg -i package.deb
apt update
wget -O package.deb https://deb.debian.org/debian/pool/main/libg/libgcrypt20/libgcrypt20_1.10.1-3_arm64.deb
dpkg -i package.deb
apt update
```
Also, let's check for fsck logs and force a filesystem check+repair on reboot:
```sh
journalctl -t systemd-fsck
> /forcefsck
reboot
```
After reboot, check again logs:
```sh
journalctl -t systemd-fsck
```

SystemTheory, 2024-05-22:
## Thank you very much! Problem seems to be fixed after reboot:

root@dpi3A /root
$ journalctl -t systemd-fsck
May 22 18:48:17 dpi3A systemd-fsck[137]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
May 22 18:48:18 dpi3A systemd-fsck[159]: /dev/mmcblk0p2: 23741/950272 files (0.7% non-contiguous), 652818/3770112 blocks
May 22 18:48:20 dpi3A systemd-fsck[200]: Please pass 'fsck.mode=force' on the kernel command line rather than creating /forcefsck on the root file system.
May 22 18:48:21 dpi3A systemd-fsck[246]: fsck.fat 4.2 (2021-01-31)
May 22 18:48:21 dpi3A systemd-fsck[246]: /dev/mmcblk0p1: 377 files, 32783/130039 clusters

root@dpi3A /root
$ ls

root@dpi3A /root
$ apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Hit:5 https://archive.raspberrypi.com/debian bookworm InRelease
Get:6 https://deb.debian.org/debian bookworm-updates/main arm64 Packages.diff/Index [10.6 kB]
Get:7 https://deb.debian.org/debian bookworm-updates/main arm64 Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:7 https://deb.debian.org/debian bookworm-updates/main arm64 Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:8 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages.diff/Index [4995 B]
Get:9 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages.diff/Index [28.6 kB]
Get:10 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Ign:10 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index
Get:11 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:11 https://deb.debian.org/debian bookworm-backports/non-free arm64 Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:12 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages T-2024-05-17-2022.39-F-2024-04-16-1405.22.pdiff [2238 B]
Get:12 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages T-2024-05-17-2022.39-F-2024-04-16-1405.22.pdiff [2238 B]
Get:13 https://deb.debian.org/debian bookworm-backports/main arm64 Packages [206 kB]
Fetched 318 kB in 4s (83.6 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

closed 2024-05-22 by SystemTheory

MichaIng, 2024-05-23:
Okay great. Seems one of the reinstalls solved it 🙂.