MichaIng/DietPi #7089: Odroid C1 | USB ports are not functional

MichaIng, 2024-05-26:
It has been enabled in Armbian for `edge` branch: https://github.com/armbian/build/pull/6559/files
But it has no hot-plug detection, keyboard input misses keystrokes, boot takes longer an additional kernel errors show up: https://github.com/MichaIng/DietPi/issues/6332#issuecomment-2125957003

Enabling it in `current` branch seems to worsen those symptoms: https://github.com/MichaIng/DietPi/issues/6332#issuecomment-2127837112

A udev event loop seems to cause issues, probably the USB hub or device is repetitively disconnected and reconnected kernel-wise.

For now, we leave it disabled, until a fix for trouble-free USB usage has been found.

juanlufont, 2024-05-26:
I will keep an eye on this issue and I will test new images when possible :+1:

MichaIng, 2024-06-12:
@th2j
Let's continue the topic here.

First of all, please try `edge` kernel. The [PR at Armbian](https://github.com/armbian/build/pull/6559) at least states, that it fixes the USB hub, but the same change did not work with `current` kernel branch:
```sh
apt install linux-{image,dtb}-edge-meson
```
I'll do a rebuild of the `current` kernel based on latest mainline Linux and Armbian patches + the config change from this PR. And also a rebuild of `edge`, just in case something has changed in mainline Linux in the meantime.

th2j, 2024-06-12:
ok usb works now, thanks

this doesn't seem to be a problem, but i have new warnings from boot :

root@DietPi:~# journalctl -b -p warning
juin 12 15:28:10 DietPi dhclient[326]: send_packet: No buffer space available
juin 12 15:28:10 DietPi dhclient[326]: dhclient.c:2600: Failed to send 300 byte long packet over eth0 interface

MichaIng, 2024-06-12:
Okay, please let me know whether there are any issues with your USB DAC, since @juanlufont reported missing keystrokes when using a USB keyboard.

Regarding your boot warnings: It is from the DHCP client. As long as you do have network/an IP address, it seems to have succeeded later, so you can ignore them. You can check the whole log of network bringup:
```sh
journalctl -u ifup@eth0
```
_________
Nearly all Armbian kernel patches for the Odroid C1, are about fixing HDMI: https://github.com/armbian/build/tree/main/patch/kernel/archive/meson-6.6

However, there have been quite a lot of commits in mainline Linux in the meantime: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/gpu/drm/meson/meson_drv.c?h=linux-6.6.y

Armbian patches are not regularly revisited, especially for SBCs which have no Armbian maintainer (like Odroid C1), are often many years old, just ported from one Linux version to the next. I could hence imagine that most or all of these patches are obsolete, and probably conflicting. Would not have been the first time. If one of your guys is open to test a few things, I would generate new kernel builds with all these patches removed, to see whether this does not break HDMI output, and probably even fixes USB support on Linux 6.6.

th2j, 2024-06-12:
i can test it
but i am not an expert

MichaIng, 2024-06-12:
Do you have a way to edit the content of this drive from another system, just in case it breaks something? Since you have the edge kernel installed, that one will remain on the system, but some symlinks in `/boot` need to be altered to point back to the edge kernel/dtb/initramfs, if something happens.

However, here the "current" kernel build with Armbian patches removed, hence unmodified mainline Linux 6.6.y:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-meson.deb
dpkg -i linux-{dtb,image}-current-meson.deb
reboot
```

th2j, 2024-06-12:
it doesn't boot,
i will reinstall from scratch, i don't know how to manually switch to the old kernel

MichaIng, 2024-06-13:
> it doesn't boot

That is weird. You mean it does not boot at all, or is there just no HDMI output? Because, as said, most of the Armbian patches are around enabling HDMI, which was at least broken in mainline Linux for some years.

> i don't know how to manually switch to the old kernel

As said, there are 3 symlinks in `/boot`: `Image` (pointing to `vmlinuz-*`), `dtb` (pointing to `dtb-*`) and `uInitrd` (pointing to `uInitrd-*`), which should currently all point to the 6.6.x version. You need to recreated those symlinks to point to the 6.8.x version, to "revert" to the functional edge kernel.

MichaIng, 2024-06-13:
I also generated newer edge kernel builds. So @juanlufont, if you want to test whether these enhance USB support in your case:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-edge-meson.deb
dpkg -i linux-{dtb,image}-edge-meson.deb
reboot
```
I want to apply the reboot fix to all existing C1 users, lifting the hold on kernel packages, but of course we cannot do so, when this breaks USB instead. So would be good to know whether the edge kernel solves both, in which case I would apply that one instead. Since we control package releases ourselves now, shipping edge kernel builds is not such an issue anymore. Previously, using the Armbian repo, it was always a higher risk that any edge kernel upgrade breaks things, because it is pretty untested.

juanlufont, 2024-06-13:
@MichaIng I will test it this weekend

th2j, 2024-06-13:
> That is weird. You mean it does not boot at all, or is there just no HDMI output? Because, as said, most of the Armbian patches are around enabling HDMI, which was at least broken in mainline Linux [...]

It does't boot at all,
I reinstall from scratch. The usb wasn't working so i updated to edge and it works now. Reboot is also ok

In /boot, i don't see the symlink : 

root@DietPi:~# ls -l /boot
total 33980
-rwxr-xr-x 1 root root     3274 12 mai   23:14 boot.ini
-rwxr-xr-x 1 root root   205117 25 mai   17:28 config-6.8.11-edge-meson
drwxr-xr-x 4 root root     4096 13 juin  13:21 dietpi
-rwxr-xr-x 1 root root    18092 12 mai   22:00 dietpi-LICENSE.txt
-rwxr-xr-x 1 root root    16059 12 mai   22:00 dietpi-README.md
-rwxr-xr-x 1 root root    17865 13 juin  14:36 dietpi.txt
-rwxr-xr-x 1 root root     3950 12 mai   23:12 dietpi-wifi.txt
drwxr-xr-x 2 root root     2048 13 juin  13:58 dtb
-rwxr-xr-x 1 root root  6395385 13 juin  13:01 initrd.img-6.6.32-current-meson
-rwxr-xr-x 1 root root  6398149 13 juin  13:59 initrd.img-6.8.11-edge-meson
-rwxr-xr-x 1 root root  4422658 25 mai   17:28 System.map-6.8.11-edge-meson
-rwxr-xr-x 1 root root 10900032 25 mai   17:28 uImage
-rwxr-xr-x 1 root root  6398213 13 juin  13:59 uInitrd

MichaIng, 2024-06-13:
> It does't boot at all

Crazy that mainline Linux support is then completely broken. Would be interesting to see UART/serial console output. Aside of HDMI support, two Armbian patches revert each an upstream commit, regarding PWM and clock speeds. Maybe those are strictly required. However, let's not waste too much time with this.

> In /boot, i don't see the symlink

Oh right, it has a dedicated boot FAT partition, hence no symlinks. In this case, we move/rename the files/dirs, so reverting is not possible 🤔. I think we should copy them instead, to allow switching back and forth between installed kernel versions. The Odroid C1 is however literally the only left SBC where this is the case/still needed 😄.

Okay, focusing on edge kernel, then the only left thing would be to have some more tests results about USB stability, i.e. whether e.g. the new build (Linux 6.9 already) fixes the missing keystrokes issue.

MichaIng, 2024-07-03:
I just pushed new kernel builds to our APT server, including "current" ones based on latest upstream with `CONFIG_USB_ONBOARD_HUB=y`. Could someone give those a last try, so see whether by any chance this works now as well with "current" Linux 6.6?

If not, especially @juanlufont could you give the "edge" ones another try, to see whether those provide stable USB support for you now as well? I'd in case create fresh images with edge kernel for now, and migrate all existing Odroid C1 system to that one, to avoid a regression when upgrading from the old Linux 6.1 kernel, where USB was still working.

maxdd, 2024-08-04:
I'm on 

`Linux DietPi 6.6.36-current-meson #1 SMP Thu Jun 27 11:49:15 UTC 2024 armv7l GNU/Linux`

I can test if you tell me what i need to do. I'm need of using WiFi via USB Adapter

MichaIng, 2024-08-04:
Hmm, so both kernel versions do not work for you? Seems to be different for everyone 😕.

maxdd, 2024-08-04:
i might have a power supply issue on the USB, i will investigate more

MichaIng, 2024-08-04:
Okay, for everyone who has issues with USB on current and edge kernel, I just pushed the legacy one (Linux 6.1):
```sh
apt update
apt install linux-{dtb,image}-legacy-meson
```
If this is the only one where USB works flawlessly, then we may migrate all Odroid C1 system to that one, or at least offer it on DietPi update.

th2j, 2024-08-04:
usb works for me on Linux DietPi 6.9.7-edge-meson

closed 2024-10-18 by MichaIng

juanlufont, 2024-11-15:
@MichaIng, sorry for the long delay (life happens)

I tested the image available on the dietpi.com site and it worked fine, both video and USB

Thank you so much for the improvements for the Odroid C1 :-)

MichaIng, 2024-11-15:
I am glad to hear that, thanks for the feedback.

MichaIng, 2025-06-23:
Hey guys, in the meantime, the "current" branch is Linux 6.12 and the "edge" branch has been removed for Odroid C1 from Armbian build system.

Hence I aim to migrate systems back to current branch with next DietPi release, hence upgrading to Linux 6.12. Luckily we are at the start of the development cycle, so I would be happy if someone could test it, to be sure there are no issues with Linux 6.12:
```sh
packages=()
dpkg-query -s 'linux-headers-edge-meson' &> /dev/null && packages+=('linux-headers-current-meson')
dpkg-query -s 'linux-libc-dev-edge-meson' &> /dev/null && packages+=('linux-libc-dev-current-meson')
sudo apt install linux-{image,dtb}-current-meson "${packages[@]}" linux-u-boot-odroidc1-current
```
If all is fine:
```sh
G_AGP linux-{image,dtb,headers,libc-dev}-edge-meson linux-u-boot-odroidc1-edge
```