MichaIng/DietPi #7090: Dietpi on Rockchip RK3066

actraiser30, 2024-05-26:
Hello

I have a similar sbc like the Marsboard pro

With the Rockchip RK3066

If I try to Install dietpi manual is this chipset supported???

Thanks

MichaIng, 2024-06-03:
DietPi is mostly userland composition, so the question is more whether you have a kernel and bootloader for this SBC/SoC. We use the Armbian build system for our kernel and bootloader packages in most cases. I checked the 32-bit Rockchip kernel and it does have device trees for 4 RK3066 SBCs: 
[image]

AFAIK, Armbian does not support any of those, but mainline Linux obviously does: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm/boot/dts/rockchip
If your's is among them, you only need a bootloader source. Mainline U-Boot supports the Marsboard, at least: https://github.com/u-boot/u-boot/blob/master/configs/marsboard_defconfig

If you have a booting image and want to run `dietpi-installer` on it, take care that the kernel remains installed, if it is an APT package. Our installer removes unknown APT packages quite radically.

... thinking about it, I just have a great idea how to prevent that for kernel packages explicitly. Hang on 😄.

actraiser30, 2024-06-16:
Thanks a lot

actraiser30, 2024-06-16:
Mine is the mk808,Thanks a lot

Now I have to figure out how to load a Linux image in it, if I can find any :/

Is there a DietPi "generic" image to include boot files like these ones for Rockchip rk3066
To work in unsupported arm boards like mine?

Are there plans to do one in the Future?

Thanks

MichaIng, 2024-06-16:
In the ARM world, something like a generic image which boots on every SBC/chip does not exist. It always requires individual drivers and the device tree (not only Linux, but the U-Boot image as well). As there is an mk808 device tree in the Linux image, that one should theoretically work. And luckily, mainline U-Boot has a config for this board as well: https://github.com/u-boot/u-boot/blob/master/configs/mk808_defconfig

actraiser30, 2024-06-16:
Yes,but I don't know how to do all this to work.and in what image how and where I put the uboot/config etc.
By generic i mean the older devices only that dietpi have not an image or the image is discontinued.
"generic for old boards" for 3 or 4 boards like cubieboard 1,2,mk802,mk808 etc.

Thanks