MichaIng/DietPi #7091: LUKS on Pine64 Quartz64 Model A not working

gxsw, 2024-05-27:
First of all thank you so much to the DietPi community, really great project!

### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable
- Distro version | bookworm
- Kernel version | 
Linux DietPi 6.8.9 #1 SMP PREEMPT Sat May 11 20:41:03 UTC 2024 aarch64 GNU/Linux

- SBC model | Quartz64 (aarch64) Model A

#### Additional Information (if applicable)
I tried to mount a luks encypted drive, which is working on other systems.

#### Steps to reproduce
cryptsetup luksOpen /dev/sda1 tmpluks
Enter passphrase for /dev/sda1: 
device-mapper: reload ioctl on tmplu (253:0) failed: Invalid argument

#### Expected behaviour
Luks to mount successfully

#### Actual behaviour
[](device-mapper: reload ioctl on tmplu (253:0) failed: Invalid argument)

#### Extra details
Do I need to activate dm-crypt anywhere after installing cryptsetup?

MichaIng, 2024-06-01:
I guess some kernel module is missing. I'll check back which one, add it to our kernel config and do a new kernel package built ASAP.

MichaIng, 2024-06-09:
Change done: https://github.com/MichaIng/DietPi/commit/a4b1248 (as well for Star64 and VisionFive 2 on our dedicated Linux branch)
Kernel build running: https://github.com/MichaIng/DietPi/actions/runs/9437946257

Would be great if you could give it a try, after the build has finished:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/firmware-quartz64a.deb
dpkg -i firmware-quartz64a.deb
reboot
```

gxsw, 2024-06-10:
Thank you so much! it's working!

closed 2024-06-10 by MichaIng