MichaIng/DietPi #7093: Portainer / Docker Odroid C4 fails

fynnstr, 2024-06-01:
#### Details:
- Date           | Sat Jun  1 10:33:42 KST 2024
- Bug report     | 2f782359-1703-4b59-a323-b57f1dfdda30
- DietPi version | v9.4.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Odroid C4/HC4 (aarch64) (ID=16)
- Kernel version | `Linux dietpi 6.6.32-current-meson64 #1 SMP PREEMPT Sat May 25 14:22:56 UTC 2024 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `apt-get -y --allow-change-held-packages install docker-ce`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Clean install of dietpi on Odroid C4
2. Install portainer via dietpi-software

#### Expected behaviour:
- installation successful
- 
#### Actual behaviour:

- installation fails
- 
#### Extra details:

- clean install of dietpi
- other installs via dietpi-software are working
- 
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
docker-ce is already the newest version (5:26.1.3-1~debian.12~bookworm).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 docker-ce : Depends: containerd.io (>= 1.6.24) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

fynnstr, 2024-06-01:
Upated my local/regional settings and got it working.

closed 2024-06-01 by fynnstr

MichaIng, 2024-06-01:
In this case `apt --fix-broken install` is offering a fix. This error indicates that there was an aborted or failed APT installation or upgrade previously. In such cases, APT does not fix it right away, since this can also imply the removal of packages. So when it suggests to run this command, try it. If it does not remove packages but just installs or reconfigures (finish configuration) of some, you can safely confirm. It it would remove packages not related to the current install attempt, it makes sense to have a closer look into the dependency conflict.