MichaIng/DietPi #7095: Mediawiki: Image upload with default installation not possible

Ernstian, 2024-06-02:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'
- Distro version | bookworm 0
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)

#### Additional Information (if applicable)
- Software title | Mediaiwki
- Was the software title installed freshly or updated/migrated? Fresh

#### Steps to reproduce
1. Install the default mediawiki
2. Upload a file via /wiki/index.php?title=Special:Upload

#### Expected behaviour
File should be uploaded

#### Actual behaviour
`Could not open lock file for "mwstore://local-backend/local-public/4/4c/whereverfile.png".`

#### Extra details
`sudo chown -cR www-data:www-data /var/www/wiki/images` fixes the problem. Owner was "root".

MichaIng, 2024-06-02:
Hmm, this is weird, since the whole directory structure should be owned by `www-data`. These are the last commands we run:
```sh
G_EXEC chown -R www-data:www-data mediawiki-*
[[ -d '/var/www/wiki' ]] && G_EXEC rm -R /var/www/wiki

# Move new instance in place
G_EXEC mv mediawiki-* /var/www/wiki
```
So the whole downloaded archive is `chown`ed to `www-data`, before it is moved in place.

But I see another problem now: We could remove all uploaded files on reinstall. Is `images` the only directory used for this, or are there other directories used for custom uploads? We should of course preserve those, like we do with the config file.

MichaIng, 2024-06-02:
On a fresh installation my end:
```sh
root@VM-Bookworm:~# ls -dl /var/www/wiki/images/
drwxr-xr-x 2 www-data www-data 4096 Jun  2 20:01 /var/www/wiki/images/
```
I.e. I cannot replicate it.

Ernstian, 2024-06-02:
The error happened on my end, as I forgot that I did an update of mediawiki didn't change the ownership of the new files. Sorry for the confusion! I changed all ownership to www-data now.

Reading https://www.mediawiki.org/wiki/Special:MyLanguage/Manual:Backing_up_a_wiki , wiki/images and wiki/LocalSettings.php are the files that need to be backed up (in addition to the database)

MichaIng, 2024-06-02:
Thanks. I figured this as well and implemented it already: https://github.com/MichaIng/DietPi/commit/5216429
Only `.htaccess` and `README` are taken from the new version, all other content is preserved.

Hence after next DietPi release, you can update MediaWiki via:
```sh
dietpi-software reinstall 210
```

closed 2024-06-02 by MichaIng