MichaIng/DietPi #7101: dietpi-update error while upgrading to latest DietPi version

Pipodi, 2024-06-07:
#### Details:
- Date           | Fri Jun  7 13:36:29 CEST 2024
- Bug report     | bbeac364-546e-4963-b19a-2c3a3e57b127
- DietPi version | v9.3.0 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
I just ran `sudo dietpi-update`.
#### Expected behaviour:
dietpi-update should complete successfully
#### Actual behaviour:
apt-get upgrade gets an error while reading list file
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  containerd.io docker-ce docker-ce-cli docker-compose-plugin tailscale
debconf: delaying package configuration, since apt-utils is not installed
5 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/86.7 MB of archives.
After this operation, 305 kB of additional disk space will be used.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'php-common' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

MichaIng, 2024-06-07:
> `files list file for package 'php-common' is missing final newline`

`/var/lib/dpkg/info/php-common.list` is damaged. Can you check and post its content?
```sh
cat /var/lib/dpkg/info/php-common.list
```

Pipodi, 2024-06-07:
> `/var/lib/dpkg/info/php-common.list` is damaged. Can you check and post its content?

Yeah, I think so.

```
dietpi@DietPi:~$ cat /var/lib/dpkg/info/php-common.list
/./e`c
/et#/croN.d
▒etc/cro?.$/▒)p▒od)b
/▒)b/Syqtamoly"/▒Y▒~emd+syStA,
/lybosysTmmd/systemophps%srionslea~.sdrfice
/lYb?wywte▒`.{qste▒/rlpsess▒onclean.|koeouqr
/ug▒^ij
▒usr&lkb-▒hp
/uqr/l▒b7php▒plp-"po-soaket-he▒Q▒r
/usr
    lib/pipp|▒)helrmr
▒el15bib/pht/php-mAM▒|{gv)pt
,usrli▒/ujp/Sessioncdea>/5sv/#bin▒/usr/qbin-pip▒nmoD▒/twr/sbyn/p▒pquury/u▒r/rHMrG
/usr+▒haRu/doc
/usr+Slare/dog/0hp-}eO.
sommgn/c angelog.
/qSr?#hjsaodmc/`▒0-cnimoN)c
                           p▒right
/usb/sha{e'lhnuian?usr/2jr▒/l)ntIaN/Wvr▒▒de/USr/shaqeliNtial?overriTer/pht-commwn
                                                                                 -t`r
/var/lib
/6ar-▒ib/php
/var/l)b/phpmGlulew
Var/lIg▒hp/rass▒on/us2/sbin&phpnism/$
```

I immagine it was corrupted somehow.

MichaIng, 2024-06-07:
> I immagine it was corrupted somehow.

Completely 😄. Let's repair it:
```sh
cd /tmp
apt download php-common
dpkg-deb -R php-common*.deb php-common
find php-common -mindepth 1 ! \( -path 'php-common/DEBIAN*' -prune \) | sed 's/^php-common//' > /var/lib/dpkg/info/php-common.list
rm -R php-common*
```

Pipodi, 2024-06-07:
Thank you so much. File restored and the update was successful. Closing the issue.

closed 2024-06-07 by Pipodi