MichaIng/DietPi #7102: Docker pull error on Pine64 Quartz64 Model A

gxsw, 2024-06-10:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bookworm
- Kernel version | Linux nextcloudpi 6.9.3 #1 SMP PREEMPT Sun Jun  9 16:47:50 UTC 2024 aarch64 GNU/Linu
- SBC model | Quartz64 (aarch64)
- Power supply used | na
- SD card used | eMMC module

#### Steps to reproduce

1. install docker
2. docker pull caddy:alpine (OR pihole/pihole:latest) 
3. output: failed to register layer: lsetxattr security.capability /usr/bin/caddy: operation not supported

#### Expected behaviour
docker pull should be successful

#### Actual behaviour
Several standard docker images working on other sbc show error, when pulled
Error message: failed to register layer: lsetxattr security.capability /usr/bin/caddy: operation not supported

#### Extra details
related resources
https://github.com/AdguardTeam/AdGuardHome/issues/6816
https://forum.openwrt.org/t/docker-pull-fails-failed-to-register-layer-operation-not-supported/138253/63?page=2
=> it might be related to Kernel setting   CONFIG_KERNEL_EXT4_FS_SECURITY and KERNEL_EXT4_FS_POSIX_ACL
- ...

gxsw, 2024-06-10:
I just moved dietpi_userdata containing docker image folder from the system ext4 mount to a btrfs mount => it's working there

MichaIng, 2024-06-10:
Thanks for reporting. `CONFIG_EXT4_FS_SECURITY` was missing, indeed.

Added with: https://github.com/MichaIng/DietPi/commit/a026453
Kernel package builds are running. I'll push them to our APT server, once done.

MichaIng, 2024-06-11:
I pushed the new packages to our APT server, please check whether this solves your issue:
```sh
apt update
apt upgrade
reboot
```

closed 2024-06-13 by MichaIng