MichaIng/DietPi #7103: /boot not mounted by PARTUUID

mw75, 2024-06-10:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.5.1
- Distro version | bookworm 0
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 5A
- SD card used | none, booting from 2G Kingston USB 

#### Additional Information (if applicable)
- Software title | bare System
- Was the software title installed freshly or updated/migrated?
  - Updated from the latest 9.4.? to 9.5.1
- Can this issue be replicated on a fresh installation of DietPi?
  - Yes, repeated two times with different USB flash devices

#### Steps to reproduce
1. dd if=DietPi_RPi-ARMv8-Bookworm.img of=/dev/<my usb blockdevice>
2. mount first partition
3. edit config.txt to change resolution to VGA
4. edit dietpi-config.txt to enable wifi, disable ethernet, set country code and keyboard to "de"
5. boot on rpi4 4gb
6. fulfill the init wizard with 9.5.1 update and install vim and git with dietpi-software
7. shutdown
8. power cycle
9. got stuck in ...ifupdown_online... and ...dev-disk-by...
10. enter recovery shell by root password
11. mount shows /boot is missing, root is fine
12. `cat /etc/fstab` and compare /boot with `blkid` - all fine
13. replace ```PARTUUID=f0cf2e93-01``` with ```/dev/sda1```
14. works again

#### Expected behaviour
mounting by PARTUUID shold by possible

#### Actual behaviour
mount needs partition device file

#### Extra details
- rebooted between 12 and 13
- fsck'ed both partitions on an other pc
- have NOT tried to get the problem back after successful boot

MichaIng, 2024-06-10:
Did you check whether the actual PARTUUID of the boot partition matches the one in `/etc/fstab`?
You can do so from another Linux system, e.g. via
```sh
lsblk -no PARTUUID /dev/sda1
```
There you can also test to mount it with this PARTUUID, like so
```sh
mount PARTUUID=9f102456-01 /mnt/mountpoint
```
Replace the values accordingly.

mw75, 2024-06-11:
Please see action item 12 : "cat /etc/fstab and compare /boot with blkid - all fine"
blkid also shows the partuuid and they matched up perfectly. That is the reason for the first "Extra detail" - I've meant it should have worked....

MichaIng, 2024-06-11:
I am not sure how it can have this effect, but since it was one change in DietPi v9.5 and you mentioned it is stuck at `ifupdown-sonething`, can you try this, before reboot:
```sh
sed -i '/^[^#].*network-pre.target/s/^/#/' /etc/systemd/system/ifupdown-pre.service.d/dietpi.conf
systemctl daemon-reload
reboot
```
Respectively remove/comment the two `network-pre.target` lines in `/etc/systemd/system/ifupdown-pre.service.d/dietpi.conf`?