MichaIng/DietPi #7107: Wrong RAM calculation in dietpi banner

jwgn, 2024-06-14:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `bookworm`
- Kernel version | `Linux futro-s920-thinclient 6.1.0-21-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.90-1 (2024-05-03) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)` (Futro S920)
- Power supply used | (Original Power Supply)
- SD card used | None

#### Additional Information (if applicable)
- Software title | None
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `d75a4926-90a3-4fb7-9515-f6d1ab0a6b34`

#### Steps to reproduce

1. Activate `17  RAM usage` in `dietpi-banner`
2. Close connection
3. Login via SSH

#### Expected behaviour

- `- RAM usage : 505 of 15876 MiB (3%)`

#### Actual behaviour

-  `- RAM usage : 15428 of 505 MiB (3058%)`

#### Extra details

- It's correct after opening and closing dietpi-banner, but incorrect right after ssh login.

MichaIng, 2024-06-14:
> It's correct after opening and closing dietpi-banner, but incorrect right after ssh login.

Interesting, probably you locale switches the arguments. Can you show the output of:
```sh
free -b
```

jwgn, 2024-06-14:
```
root@futro-s920-thinclient:~# free -b
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:16646668288   509284352 16182902784   100954112   324689920 16137383936
Swap:              0           0           0
```

Locale was set to `de_DE.UTF-8`.

I switched to `en_US.UTF-8`, now it is displayed correctly, even after a logout and new ssh connection:

`- RAM usage : 496 of 15876 MiB (3%)`

But the `free -b` output is identical:

```
root@futro-s920-thinclient:~# free -b
               total        used        free      shared  buff/cache   available
Mem:     16646668288   520585216 16171954176   100954112   324333568 16126083072
Swap:              0           0           0
```

MichaIng, 2024-06-14:
> `Speicher:16646668288`

Oh, this is the issue, there is no space after the colon, so that the 1st and the 2nd field merge into one. We could fix that one instead, but it is easier/faster to just apply `C.UTF-8` locale, like we do on the interactive execution. This also avoids other possible parsing issues.

MichaIng, 2024-06-14:
Fixed with: https://github.com/MichaIng/DietPi/commit/bea468b
Please give it a try:
```sh
wget 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/dietpi/func/dietpi-banner' -O /boot/dietpi/func/dietpi-banner
chmod +x /boot/dietpi/func/dietpi-banner
dietpi-banner 1
```

jwgn, 2024-06-14:
Looks good. Switched back to German locale, RAM calculation is correct after ssh login.

closed 2024-07-02 by MichaIng