MichaIng/DietPi #7109: DietPi_NanoPiR6C-ARMv8-Bookworm Image not working at all

labmaster, 2024-06-15:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
NanoPiR6C-ARMv8-Bookwor
Phihong 5V 5A 
tried different SD Cards (SanDisk Ultra 16GB, PNY 16GB,  Greenliant EX93 Industrial Highend SLC 16GB)

#### Additional Information (if applicable)
Issue can be replicated 
https://dietpi.com/forum/t/nanopi-r6c-image-not-working-at-all-edit/20471

#### Steps to reproduce
1. Download image from  https://dietpi.com/downloads/images/DietPi_NanoPiR6C-ARMv8-Bookworm.img.xz
2. Use a flasher tool to write it to a SD Card (tried Win32DiskImager and BalenaEtcher)
3. Insert SD card to NanoPi R6C and power it up

#### Expected behaviour
Boot up DietPi from SD-Card

#### Actual behaviour
On Powerup  SYS LED is constant on, nothing else happens.
Serial Debug console shows an error preventig system from go any further

#### Extra details
`
DDR 9fffbe1e78 cym 24/02/04-10:09:20,fwver: v1.16
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=16 CS1 Row=16 CS=2 Die BW=16 Size=2048MB
Manufacturer ID:0x1
CH0 RX Vref:27.1%, TX Vref:21.8%,20.8%
CH1 RX Vref:25.8%, TX Vref:19.8%,20.8%
CH2 RX Vref:26.7%, TX Vref:21.8%,21.8%
CH3 RX Vref:25.4%, TX Vref:21.8%,21.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09 (Jun 12 2024 - 21:32:04)
unknown raw ID 0 0 0
unrecognized JEDEC id bytes: 00, 00, 00
Trying to boot from MMC2
spl: partition error
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(7efcd01a0f...) + OK
## Checking uboot 0x00200000 ... sha256(2f31ed4655...) + OK
## Checking fdt 0x0031ec90 ... sha256(c5d44f02b4...) + OK
## Checking atf-2 0x000f0000 ... sha256(da90adf3a4...) + OK
## Checking atf-3 0xff100000 ... sha256(1163474a5b...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 371.864 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-589-g3389cfdda:derrick.huang
NOTICE:  BL31: Built : 10:14:29, May  9 2023
INFO:    spec: 0x13
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
ERROR:   dfs get fsp_params[0] error, 0xfead0004 != 0xfead0003
ERROR:   dfs get fsp_params[1] error, 0x1111 != 0xfead0003
ERROR:   dfs get fsp_params[2] error, 0x0 != 0xfead0003
ERROR:   dfs get fsp_params[3] error, 0x60241520 != 0xfead0003
ERROR:   loader&trust unmatch!!! Please update trust if need enable dmc
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE init                                                                              ialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2017.09-armbian (May 21 2024 - 16:21:13 +0000)

Model: NanoPi R6S
MPIDR: 0x81000000
PreSerial: 2, raw, 0xfeb50000
DRAM:  8 GiB
Sysmem: init
Relocation Offset: eda3f000
Relocation fdt: eb9f9960 - eb9fecd0
CR: M/C/I
Using default environment

optee check api revision fail: -1.0
optee api revision is too low
### ERROR ### Please RESET the board ###
`

MichaIng, 2024-06-15:
Many thanks for reporting. I can replicate it, and latest mainline U-Boot seems to have broken it, when using the previously common R6S U-Boot build. Hence Armbian added dedicated support for this board with explicit R6C U-Boot. I'll adjust our installer and rebuild the image.

As well we need to send a patch next DietPi update, or better a live-patch, to migrate to the new U-Boot package.

labmaster, 2024-06-15:
Many thanks for the quick response and the confirmation work. 

So what do you think would be the estimated timeline for having a working image.
Just ask as i have to get my system up and running again asap and one option would be to use the old image (Bullseye) as underlying system, or wait for getting all up on the newer version. (what i would prefer but time might be the showstopper)

MichaIng, 2024-06-15:
2. Implementation of new APT component done: https://github.com/MichaIng/DietPi/commit/5db7d27
3. New image build based on `dev` branch: https://github.com/MichaIng/DietPi/actions/runs/9527948341
    Reviewing the commit at Armbian, it seems they use a generic RK3588 config, so that the R6C device tree might need to be defined explicitly. So that image might not boot yet, in case I'll revert that part in the installer and do a rebuild.
4. If it works, I'll implement a live patch, which allows us to generate images based on DietPi `master`/stable branch.

closed 2024-06-15 by MichaIng

MichaIng, 2024-06-15:
I update all images. Should work now, but please give it a try 🙂.