MichaIng/DietPi #7112: adguard only runs on ipv6 on proxmox

hanscees, 2024-06-15:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
951

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm

- Kernel version | `uname -a`
Linux DietPi 6.1.0-21-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.90-1 (2024-05-03) x86_64 GNU/Linux

I downloaded proxmox dietpi bookworm and use it on proxmox running on debian. 

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Adguard

- Was the software title installed freshly or updated/migrated?
freshly installed

- Can this issue be replicated on a fresh installation of DietPi?
probably, mine was fresh

#### Steps to reproduce

1. install adguard from dietpi-software
2. apt install net-tools
3. netstat -tunlp

shows only tcp6 :53 

I have corrected this by doing ```

vi /mnt/dietpi_userdata/adguardhome/AdGuardHome.yaml 

dns:
  bind_hosts:
    - 192.168.0.185
```

Not a big deal but I thought I should mention it.

now all is well
```
tcp        0      0 0.0.0.0:8200            0.0.0.0:*               LISTEN      332/minidlnad       
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      351/sshd: /usr/sbin 
tcp        0      0 192.168.0.185:53        0.0.0.0:*               LISTEN      327/AdGuardHome     
tcp6       0      0 :::22                   :::*                    LISTEN      351/sshd: /usr/sbin 
tcp6       0      0 :::5252                 :::*                    LISTEN      328/dietpi-dashboar 
tcp6       0      0 :::8083                 :::*                    LISTEN      327/AdGuardHome     
udp        0      0 192.168.0.185:60579     0.0.0.0:*                           332/minidlnad       
udp        0      0 239.255.255.250:1900    0.0.0.0:*                           332/minidlnad       
udp        0      0 192.168.0.185:53        0.0.0.0:*                           327/AdGuardHome     
udp        0      0 0.0.0.0:67              0.0.0.0:*                           354/dhcpd        
```

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2024-06-16:
I'm not sure if your interpretation is correct. I did a quick test on RPI4B and AGH is working ootb without any configuration changes needed.

Basically, our configuration has not been changed for 3 years and AGH should listen to all network interfaces as we bind to `0.0.0.0` be default

https://github.com/MichaIng/DietPi/blob/2b5b0e7943366edad7991a1897502fa6cd6b0ec3/.conf/dps_126/AdGuardHome.yaml#L14-L17

Basically my test `netstat` is showing same result as yours

```
root@DietPi4:~# netstat -tunlp
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      446/dropbear
tcp6       0      0 :::8083                 :::*                    LISTEN      2230/AdGuardHome
tcp6       0      0 :::53                   :::*                    LISTEN      2230/AdGuardHome
tcp6       0      0 :::22                   :::*                    LISTEN      446/dropbear
udp        0      0 0.0.0.0:68              0.0.0.0:*                           350/dhclient
udp6       0      0 :::53                   :::*                                2230/AdGuardHome
root@DietPi4:~#
```

But `dig` is able to resolve a DNS request without issue

```
root@DietPi4:~# dig @127.0.0.1 -p 53 dietpi.com

; <<>> DiG 9.18.24-1-Debian <<>> @127.0.0.1 -p 53 dietpi.com
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 54122
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;dietpi.com.                    IN      A

;; ANSWER SECTION:
dietpi.com.             300     IN      A       188.114.97.3
dietpi.com.             300     IN      A       188.114.96.3

;; Query time: 80 msec
;; SERVER: 127.0.0.1#53(127.0.0.1) (UDP)
;; WHEN: Sun Jun 16 12:12:09 CEST 2024
;; MSG SIZE  rcvd: 71

root@DietPi4:~#
```

As well, no issues using the pre-defined upstream DNS

[image]

MichaIng, 2024-06-16:
Note that `net-tools` are deprecated. Use the `ss` command from the `iproute2` package to get accurate results:
```console
root@VM-Bookworm:~# netstat -tlpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      426/dropbear
tcp        0      0 127.0.0.1:3306          0.0.0.0:*               LISTEN      1955/mariadbd
tcp6       0      0 :::3000                 :::*                    LISTEN      2104/forgejo
tcp6       0      0 :::8083                 :::*                    LISTEN      2131/AdGuardHome
tcp6       0      0 :::80                   :::*                    LISTEN      2027/apache2
tcp6       0      0 :::53                   :::*                    LISTEN      2131/AdGuardHome
tcp6       0      0 :::22                   :::*                    LISTEN      426/dropbear
tcp6       0      0 :::5252                 :::*                    LISTEN      423/dietpi-dashboar
root@VM-Bookworm:~# ss -tlpn
State  Recv-Q Send-Q Local Address:Port Peer Address:PortProcess
LISTEN 0      1000         0.0.0.0:22        0.0.0.0:*    users:(("dropbear",pid=426,fd=3))
LISTEN 0      80         127.0.0.1:3306      0.0.0.0:*    users:(("mariadbd",pid=1955,fd=18))
LISTEN 0      4096               *:3000            *:*    users:(("forgejo",pid=2104,fd=13))
LISTEN 0      4096               *:8083            *:*    users:(("AdGuardHome",pid=2131,fd=9))
LISTEN 0      511                *:80              *:*    users:(("apache2",pid=2028,fd=4),("apache2",pid=2027,fd=4))
LISTEN 0      4096               *:53              *:*    users:(("AdGuardHome",pid=2131,fd=12))
LISTEN 0      1000            [::]:22           [::]:*    users:(("dropbear",pid=426,fd=4))
LISTEN 0      1024               *:5252            *:*    users:(("dietpi-dashboar",pid=423,fd=9))
```
Not sure why `netstat` is listing them with `tcp6` protocol, but most of those entries with `:::xyz` all listen on both, IPv4 and IPv6, same as the `*:xyz` entries in `ss`. `ss` shows IPv6-only listeners with `[::]:xyz` instead.

closed 2025-07-18 by MichaIng