MichaIng/DietPi #7116: TP-Link Archer T2U Nano

actraiser30, 2024-06-18:
Hello

I have a USB wifi Tplink Archer T2U Nano
With chipset Realtek RTL8811AU

Using a raspberry pi 2 with latest dietpi bookworm 

Installed wifi hotspot mode but after I realized that it does not see the USB WiFi device,in the network-adapters section it says not found,are you going to include drivers for this chipset in a feature update??

Thanks

Joulinar, 2024-06-18:
Usually we don't add individual drivers.

actraiser30, 2024-06-18:
Since not all raspberry pi's have internal wifi ,i must use an external USB wifi,but both of them I mentioned before are not working..

If possible in a future release to add a package with all the missing wifi chipset drivers to be installed if necessary if wifi hotspot mode is selected from the software list to be installed.

MichaIng, 2024-06-18:
It should work with the new kernel version. Please try our migration script. However, note that this cannot be (easily) reverted: #6676

actraiser30, 2024-06-22:
All this are a little to much for me,tried an old atheros AR92xx it worked,but the performance is very poor,is the bookworm os too much/heavy for the very aged Raspberry Pi 2?  Should I put dietpi bullseye?

Thanks

Joulinar, 2024-06-22:
Normally the performance difference between Bullseye and Bookworm should not be that big. Do you have the poor performance with the plain operating system or a specific application?

actraiser30, 2024-06-22:
No only with the hotspot,I checked with htop there not big CPU usage.strange.can be my atheros AR92xx,what pi is suggested for best hotspot performance??

Joulinar, 2024-06-22:
We never tested WiFi performance, just some adapter in general https://dietpi.com/blog/?p=3266#issues_with_external_wifi_adapters

probably @StephanStS could comment on this.

actraiser30, 2024-06-22:
Thanks,you know what county can i put to have channel 12 and 13??

actraiser30, 2024-06-22:
Nevermind I put channel 11 channels 12,13 are not working the hostapd is not starting 

With channel 11 I get the max 20mbps 

Not bad

Thanks for your help

actraiser30, 2024-07-09:
Hi again,in armbian this chipset is working if you install the armbian-firmware-full package,maybe you can add this package to dietpi to install it if hotspot mode is selected,or just the wifi drivers from this package to be installed only
Thanks

MichaIng, 2024-07-10:
I am trying to find out which firmware binary is missing. There are multiple ones which seem matching. Can you please check for related kernel errors? There should be one about a missing firmware binary, when the WiFi driver loads:
```sh
dmesg -l 0,1,2,3`
```

actraiser30, 2024-07-11:
That command does not show/do anything 

I can see the wifi adapter with lsusb command though

MichaIng, 2024-07-11:
Ah, Linux 6.1 does not contain the `rtw88` drivers yet. If would be a one-way ticket, unless you create a full SD card/drive backup, but you could try whether upgrading to the new kernel/firmware stack with Linux 6.6 helps, which I suggested above already: #6676
```sh
bash <(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')
```
Alternatively, if you have a spare SD card, we host images with this new kernel:
- For RPi 2 PCB v1.2: https://dietpi.com/downloads/images/testing/DietPi_RPi234-ARMv8-Bookworm.img.xz
- For the older 32-bit-only RPi 2 model: https://dietpi.com/downloads/images/testing/DietPi_RPi2-ARMv7-Bookworm.img.xz