MichaIng/DietPi #7117: Diet-Pi Update failed

chrisgillings, 2024-06-18:
#### Details:
- Date           | Wed Jun 19 00:05:04 AEST 2024
- DietPi version | v9.2.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux workshop-clock 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run dietpi-update
#### Expected behaviour:

- Update should succeed
#### Actual behaviour:

- [FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.
#### Extra details:

- ...
#### Additional logs:
```
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

chrisgillings, 2024-06-18:
Full log:

```
# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.5.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.2.1
[ INFO ] DietPi-Update | Latest version  : v9.5.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=11522 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:3 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:4 https://deb.debian.org/debian bookworm-backports InRelease [56.5 kB]
Get:5 https://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:6 https://deb.debian.org/debian bookworm-updates/main armhf Packages.diff/Index [10.6 kB]
Get:7 https://deb.debian.org/debian bookworm-updates/main armhf Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:7 https://deb.debian.org/debian bookworm-updates/main armhf Packages T-2024-04-23-2036.10-F-2024-04-23-2036.10.pdiff [1597 B]
Get:8 https://deb.debian.org/debian-security bookworm-security/main armhf Packages [154 kB]
Get:9 https://deb.debian.org/debian bookworm-backports/main armhf Packages.diff/Index [63.3 kB]
Ign:9 https://deb.debian.org/debian bookworm-backports/main armhf Packages.diff/Index
Get:10 https://deb.debian.org/debian bookworm-backports/contrib armhf Packages.diff/Index [30.8 kB]
Get:11 https://deb.debian.org/debian bookworm-backports/non-free armhf Packages.diff/Index [4995 B]
Get:12 https://deb.debian.org/debian bookworm-backports/contrib armhf Packages T-2024-06-05-0204.17-F-2024-04-16-1405.22.pdiff [2604 B]
Get:12 https://deb.debian.org/debian bookworm-backports/contrib armhf Packages T-2024-06-05-0204.17-F-2024-04-16-1405.22.pdiff [2604 B]
Get:13 https://archive.raspberrypi.com/debian bookworm/main armhf Packages [418 kB]
Get:14 https://deb.debian.org/debian bookworm-backports/non-free armhf Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:14 https://deb.debian.org/debian bookworm-backports/non-free armhf Packages T-2024-04-22-2005.41-F-2024-04-22-2005.41.pdiff [257 B]
Get:15 https://deb.debian.org/debian bookworm-backports/main armhf Packages [217 kB]
Fetched 1087 kB in 12s (91.4 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  libopenjp2-7 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'apt autoremove' to remove them.
[... 211 lines cut ...]
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/etc/logrotate.d/alternatives': Read-only file system
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/etc/dpkg/dpkg.cfg': Read-only file system
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/etc/cron.daily/dpkg': Read-only file system
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/etc/alternatives/README': Read-only file system
dpkg: error while cleaning up:
 unable to securely remove '/var/lib/dpkg/tmp.ci': Read-only file system
dpkg: error: unable to create new file '/var/lib/dpkg/status-new': Read-only file system
E: Sub-process /usr/bin/dpkg returned an error code (2)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
[ INFO ] DietPi-Update | APT upgrade, please wait...
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
[ INFO ] DietPi-Update | APT upgrade, please wait...
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Update | APT upgrade

chrisgillings, 2024-06-18:
So it's likely to be a failing SD card?  The Achilles Heel of Pis.  :-(
```
# dmesg -l 0,1,2,3
[61431.680881] EXT4-fs error (device mmcblk0p2): __ext4_find_entry:1675: inode #154: comm dpkg: checksumming directory block 0
[61431.680963] Aborting journal on device mmcblk0p2-8.
[61431.684705] EXT4-fs (mmcblk0p2): Remounting filesystem read-only

Joulinar, 2024-06-18:
Can you try following 

```
> /forcefsck
reboot
```

```
# then after reboot
journalctl -t systemd-fsck
```

chrisgillings, 2024-06-19:
I had to run `e2fsck` on the SD card on another server to fix the file system.  After that, the upgrade worked.
The lesson here is to recall that any time one sees a "readonly filesystem" error it's almost certainly due to corruption.

Thanks for the swift and helpful guidance!

closed 2024-06-19 by chrisgillings