MichaIng/DietPi #7119: Job for mariadb.service failed

fagron, 2024-06-20:
#### Details:
- Date           | Thu Jun 20 21:13:16 CEST 2024
- Bug report     | 8e424e6c-8032-4f33-aec4-211861b43660
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `systemctl restart mariadb`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xeu mariadb.service" for details.
```

fagron, 2024-06-20:
root@DietPi:~# readlink -f /var/lib/mysql
/mnt/dietpi_userdata/mysql

root@DietPi:~# readlink /var/lib/mysql
/mnt/dietpi_userdata/mysql

root@DietPi:~# cat /var/log/mysql/error.log
cat: /var/log/mysql/error.log: Datei oder Verzeichnis nicht gefunden

root@DietPi:~# journalctl -u mariadb
Jun 20 19:39:41 DietPi systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
Jun 20 19:39:42 DietPi mariadbd[675]: 2024-06-20 19:39:42 0 [Note] Starting MariaDB 10.11.6-MariaDB-0+deb12u1 source revision  as process 675
Jun 20 19:39:42 DietPi mariadbd[675]: 2024-06-20 19:39:42 0 [Note] InnoDB: Compressed tables use zlib 1.2.13
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: Number of transaction pools: 1
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: Using ARMv8 crc32 instructions
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: Using liburing
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: Initializing buffer pool, total size = 128.000MiB, chunk size = 2.000MiB
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: Completed initialization of buffer pool
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: File system buffers for log disabled (block size=512 bytes)
Jun 20 19:39:43 DietPi mariadbd[675]: 2024-06-20 19:39:43 0 [Note] InnoDB: End of log at LSN=5619048737
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: 128 rollback segments are active.
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: Setting file './ibtmp1' size to 12.000MiB. Physically writing the file full; Please wait ...
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: preallocating 12582912 bytes for file ./ibtmp1 failed with error 28
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: Could not set the file size of './ibtmp1'. Probably out of disk space
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: Unable to create the shared innodb_temporary
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: Plugin initialization aborted with error Generic error
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: FTS optimize thread exiting.
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: Starting shutdown...
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: Removed temporary tablespace data file: "./ibtmp1"
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] Plugin 'InnoDB' registration as a STORAGE ENGINE failed.
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] Plugin 'FEEDBACK' is disabled.
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] Unknown/unsupported storage engine: InnoDB
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] Aborting
Jun 20 19:39:44 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jun 20 19:39:44 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jun 20 19:39:44 DietPi systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.
Jun 20 19:40:51 DietPi systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] Starting MariaDB 10.11.6-MariaDB-0+deb12u1 source revision  as process 1737
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Compressed tables use zlib 1.2.13
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Number of transaction pools: 1
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Using ARMv8 crc32 instructions
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Using liburing
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Initializing buffer pool, total size = 128.000MiB, chunk size = 2.000MiB
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: Completed initialization of buffer pool
Jun 20 19:40:51 DietPi mariadbd[1737]: 2024-06-20 19:40:51 0 [Note] InnoDB: File system buffers for log disabled (block size=512 bytes)
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [Note] InnoDB: End of log at LSN=5619048737
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [Note] InnoDB: 128 rollback segments are active.
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [Note] InnoDB: Setting file './ibtmp1' size to 12.000MiB. Physically writing the file full; Please wait ...
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [ERROR] InnoDB: preallocating 12582912 bytes for file ./ibtmp1 failed with error 28
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [ERROR] InnoDB: Could not set the file size of './ibtmp1'. Probably out of disk space
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [ERROR] InnoDB: Unable to create the shared innodb_temporary
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [ERROR] InnoDB: Plugin initialization aborted with error Generic error
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [Note] InnoDB: FTS optimize thread exiting.
Jun 20 19:40:52 DietPi mariadbd[1737]: 2024-06-20 19:40:52 0 [Note] InnoDB: Starting shutdown...
[... cut ...]
Jun 20 21:13:16 DietPi mariadbd[24098]: 2024-06-20 21:13:16 0 [ERROR] Aborting
Jun 20 21:13:16 DietPi mariadbd[24098]: 2024-06-20 21:13:16 0 [ERROR] mariadbd: Error on close of '(null)' (Errcode: 9 "Bad file descriptor")
Jun 20 21:13:16 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jun 20 21:13:16 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jun 20 21:13:16 DietPi systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.

Joulinar, 2024-06-20:
You file system seems to be full

> Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [Note] InnoDB: Setting file './ibtmp1' size to 12.000MiB. Physically writing the file full; Please wait ...
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: preallocating 12582912 bytes for file ./ibtmp1 failed with error 28
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: Could not set the file size of './ibtmp1'. Probably out of disk space
Jun 20 19:39:44 DietPi mariadbd[675]: 2024-06-20 19:39:44 0 [ERROR] InnoDB: Unable to create the shared innodb_temporary

and

> Jun 20 21:13:16 DietPi mariadbd[24098]: 2024-06-20 21:13:16 0 [ERROR] mariadbd: Error writing file './ddl_recovery.log' (Errcode: 28 "No space left on device")

Can you check following 

```
df -h
```

fagron, 2024-06-20:
root@DietPi:~# df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root        30G     29G     0  100% /
devtmpfs        1,7G       0  1,7G    0% /dev
tmpfs           1,9G    2,9M  1,9G    1% /dev/shm
tmpfs           769M    8,7M  761M    2% /run
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           1,9G       0  1,9G    0% /tmp
tmpfs            50M    2,4M   48M    5% /var/log
/dev/mmcblk0p1  127M     33M   95M   26% /boot
/dev/sda1       1,8T    201G  1,6T   11% /mnt/HDD

Joulinar, 2024-06-20:
> /dev/root        30G     29G     0  100% /

Your root file system is completely full.

fagron, 2024-06-20:
I know, I did a nextcloudupdate two days ago. After that, the SD card slowly filled up. 
What can I delete?

Joulinar, 2024-06-20:
First step would be to find largest files/folder 

```
du -aBM / --exclude /mnt/HDD | sort -n -r | head -n 20
```

fagron, 2024-06-20:
root@DietPi:~# du -aBM / --exclude /mnt/HDD | sort -n -r | head -n 20
du: Zugriff auf '/proc/24369/task/24369/fd/3' nicht möglich: Datei oder Verzeichnis nicht gefunden
du: Zugriff auf '/proc/24369/task/24369/fdinfo/3' nicht möglich: Datei oder Verzeichnis nicht gefunden
du: Zugriff auf '/proc/24369/fd/4' nicht möglich: Datei oder Verzeichnis nicht gefunden
du: Zugriff auf '/proc/24369/fdinfo/4' nicht möglich: Datei oder Verzeichnis nicht gefunden
29445M	/
23782M	/mnt/dietpi_userdata
23782M	/mnt
23030M	/mnt/dietpi_userdata/nextcloud_data
19684M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj
19670M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview
2661M	/root/logfile_storage
2661M	/root
2626M	/root/logfile_storage/pihole
2623M	/root/logfile_storage/pihole/pihole.log
2199M	/mnt/dietpi_userdata/nextcloud_data/updater-ocshgkvqztuj/backups
2199M	/mnt/dietpi_userdata/nextcloud_data/updater-ocshgkvqztuj
1781M	/usr
1307M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/5
1267M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/1
1266M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/f
1260M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/7
1256M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/a
1246M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/d
1241M	/mnt/dietpi_userdata/nextcloud_data/appdata_ocshgkvqztuj/preview/0

Joulinar, 2024-06-20:
On short notice, you could delete an old Pihole log to gain 2.5GB

```
rm /root/logfile_storage/pihole/pihole.log
```

On a long-term, you might need to move `/mnt/dietpi_userdata` to your HDD. Can be done via drive manager.

fagron, 2024-06-20:
Ok, I have deleted the pihole.log.
restart mariadb.service
nextcloud installed

Thanks for the help. It is working.

Joulinar, 2024-06-21:
I guess it has been fixed, right? Therefore I will close the topic for now.

closed 2024-06-21 by Joulinar