MichaIng/DietPi #7120: [Warning] Can't create test file '/var/lib/mysql/DietPi.lower-test

fagron, 2024-06-21:
root@DietPi:~# journalctl -xeu mariadb.service
Jun 21 08:50:25 DietPi systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
░░ Subject: A start job for unit mariadb.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit mariadb.service has begun execution.
░░ 
░░ The job identifier is 106.
Jun 21 08:50:26 DietPi mariadbd[665]: 2024-06-21  8:50:26 0 [Warning] Can't create test file '/var/lib/mysql/DietPi.lower-test' (Errcode: 2 "No such file or directory")
Jun 21 08:50:26 DietPi mariadbd[665]: [99B blob data]
Jun 21 08:50:26 DietPi mariadbd[665]: 2024-06-21  8:50:26 0 [ERROR] Aborting
Jun 21 08:50:26 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit mariadb.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jun 21 08:50:26 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit mariadb.service has entered the 'failed' state with result 'exit-code'.
Jun 21 08:50:26 DietPi systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.
░░ Subject: A start job for unit mariadb.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit mariadb.service has finished with a failure.
░░ 
░░ The job identifier is 106 and the job result is failed.
Jun 21 08:50:46 DietPi systemd[1]: Starting mariadb.service - MariaDB 10.11.6 database server...
░░ Subject: A start job for unit mariadb.service has begun execution
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit mariadb.service has begun execution.
░░ 
░░ The job identifier is 147.
Jun 21 08:50:46 DietPi mariadbd[1486]: 2024-06-21  8:50:46 0 [Warning] Can't create test file '/var/lib/mysql/DietPi.lower-test' (Errcode: 2 "No such file or directory")
Jun 21 08:50:46 DietPi mariadbd[1486]: [99B blob data]
Jun 21 08:50:46 DietPi mariadbd[1486]: 2024-06-21  8:50:46 0 [ERROR] Aborting
Jun 21 08:50:46 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStart= process belonging to unit mariadb.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jun 21 08:50:46 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit mariadb.service has entered the 'failed' state with result 'exit-code'.
Jun 21 08:50:46 DietPi systemd[1]: Failed to start mariadb.service - MariaDB 10.11.6 database server.
░░ Subject: A start job for unit mariadb.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit mariadb.service has finished with a failure.
░░ 
░░ The job identifier is 147 and the job result is failed.

Joulinar, 2024-06-21:
It would be really nice if you could write a few more words about your problem. What did you do before or were there any changes? A little more information would be appreciated.

for testing

```
sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
ls -la /var/lib/mysql/
```

fagron, 2024-06-21:
Sorry.

I would like to move dietpi_userdata to HDD later.
My first step was to copy dietpi_userdata to HDD.
My second step was to rename dietpi_userdata to dietpi_userdata_old. (Test if the nextcloud still works)
Then I restarted all services.

"[ OK ] DietPi-Services | restart : redis-server
[FAILED] DietPi-Services | restart : mariadb
[ OK ] DietPi-Services | restart : php8.2-fpm
[ OK ] DietPi-Services | restart : nginx"

Then I have "systemctl restart mariadb.service"

"Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xeu mariadb.service" for details."

fagron, 2024-06-21:
root@DietPi:~# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
ls -la /var/lib/mysql/
touch: '/var/lib/mysql/demo.file' kann nicht berührt werden: Datei oder Verzeichnis nicht gefunden
ls: Zugriff auf '/var/lib/mysql/' nicht möglich: Datei oder Verzeichnis nicht gefunden

Joulinar, 2024-06-21:
> My second step was to rename dietpi_userdata to dietpi_userdata_old. (Test if the nextcloud still works)

This is causing your issue.

Pls revert all changes you have done. To move `dietpi_userdata` use drive manager. There is a build-in function to handle this scenario.

fagron, 2024-06-21:
Ok, I have undone all the changes.

How can I now move the dietpi_userdata directory to HHD without mariadb.service and the nextcloud causing problems?

fagron, 2024-06-21:
root@DietPi:~# sudo -u mysql -g mysql touch /var/lib/mysql/demo.file
ls -la /var/lib/mysql/
insgesamt 188920
drwxr-xr-x 6 mysql  mysql       4096 21. Jun 12:51 .
drwxrwxr-x 9 dietpi dietpi      4096  3. Aug 2023  ..
-rw-rw---- 1 mysql  mysql     425984 21. Jun 12:37 aria_log.00000001
-rw-rw---- 1 mysql  mysql         52 21. Jun 12:37 aria_log_control
-rw-rw---- 1 mysql  mysql          9 21. Jun 12:28 ddl_recovery-backup.log
-rw-rw---- 1 mysql  mysql          9 21. Jun 12:51 ddl_recovery.log
-rw-r--r-- 1 root   root           0 12. Feb 17:25 debian-10.11.flag
-rw-r--r-- 1 mysql  mysql          0 21. Jun 12:51 demo.file
-rw-rw---- 1 mysql  mysql      16649 21. Jun 12:37 ib_buffer_pool
-rw-rw---- 1 mysql  mysql   79691776 21. Jun 12:36 ibdata1
-rw-rw---- 1 mysql  mysql  100663296 21. Jun 12:36 ib_logfile0
-rw-rw---- 1 mysql  mysql   12582912 21. Jun 12:51 ibtmp1
-rw-rw---- 1 mysql  mysql          0 13. Jul 2023  multi-master.info
drwx------ 2 mysql  mysql       4096 13. Jul 2023  mysql
-rw-r--r-- 1 root   mysql         15 13. Jul 2023  mysql_upgrade_info
drwx------ 2 mysql  mysql      16384 20. Jun 22:23 nextcloud
drwx------ 2 mysql  mysql       4096 13. Jul 2023  performance_schema
drwx------ 2 mysql  mysql      12288 13. Jul 2023  sys

Joulinar, 2024-06-21:
> How can I now move the dietpi_userdata directory to HHD without mariadb.service and the nextcloud causing problems?

As stated more than once, use drive manager

fagron, 2024-06-21:
Only use the drive manager?
What about the parameter "SOFTWARE_NEXTCLOUD_DATADIR" in the dietpi.txt?
When do I have to change it?
Before the reboot?

Joulinar, 2024-06-21:
Simply go into drive manger, select your HDD and select the `User data` option. Nothing else to change.

fagron, 2024-06-21:
Thank you, it works.

closed 2024-06-21 by Joulinar