MichaIng/DietPi #7125: a2enmod command not found

nageshshetty, 2024-06-23:
#### Details:
- Date           | Sun Jun 23 21:30:07 BST 2024
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux CT-MACHINA 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `a2enmod http2`
- Exit code      | 127
- Software title | DietPi-LetsEncrypt
#### Steps to reproduce:

1. ...LEts encrypt has stopped working from the dietpi-luancher or or the command line
2. ...
#### Expected behaviour:

- ...run and install certitificate
#### Actual behaviour:

- ...
#### Extra details:
I have lighttpd and all was working till now so dont know what triggered the error
- ...
#### Additional logs:
```
/boot/dietpi/func/dietpi-globals: line 852: a2enmod: command not found

nageshshetty, 2024-06-23:
its resolved. for some reason apache2.service file was created in systemd. removed it and all working now..thanks

closed 2024-06-23 by nageshshetty

MichaIng, 2024-06-24:
> for some reason apache2.service file was created in systemd.

systemd does not create services by itself. Are you sure Apache is not somehow installed as well?
```sh
dpkg -l | grep apache
```

nageshshetty, 2024-06-26:
> > for some reason apache2.service file was created in systemd.
> 
> systemd does not create services by itself. Are you sure Apache is not somehow installed as well?
> 
> ```shell
> dpkg -l | grep apache
> ```

had never installed apache from day 1 but for some reason only apache configuration file was created from last reboot. had to remove it as every reboot the init.d file was created. ..thanks

MichaIng, 2024-06-26:
Then you should find out who or what is creating this file on your system ASAP. Might be a serious security incidence or bug. E.g.:
```sh
apt install inotify-tools lsof
{ inotifywait /etc/init.d/apache2; lsof /etc/init.d/apache2; } &> /root/inotify.log &
```
In the hope `lsof` runs fast enough to catch the process. If not, then there are other way with `loggedfs` or `auditd`.

nageshshetty, 2024-07-02:
I have done now..so far nothing but will keep an eye. Thanks @MichaIng!