MichaIng/DietPi #7126: Unable to update DietPi

mss1980au, 2024-06-24:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
### Details:
- Date           | Mon Jun 24 11:30:10 BST 2024
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-mark unhold rpi-eeprom`
- Exit code      | 100
- Software title | DietPi-Pre-patches
- SD Card - SanDisk

#### Steps to reproduce
Sudo dietpi-update

#### Expected behaviour
Should update to latest version, but instead throws this error

#### Actual behaviour
Update fails with below message:
[FAILED] DietPi-Pre-patches | Unable to continue, DietPi-Pre-patches will now terminate.
[FAILED] DietPi-Update | An error occurred during pre-patching. Please check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors, and rerun "dietpi-update" after the cause has been solved.

#### Extra details
Waited for a long time thinking that the next update versions might fix this issue, but no luck, always get this same issue

MichaIng, 2024-06-24:
Can you run these commands and paste the output here:
```sh
dpkg-query -s rpi-eeprom
apt-mark showhold rpi-eeprom
apt-mark unhold rpi-eeprom
```

mss1980au, 2024-06-25:
Hello

Please see results of the commands you asked me to run, cut and paste
below, says syntax error for last 2 commands

***@***.***:~$ dpkg-query -s rpi-eeprom
Package: rpi-eeprom
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 28850
Maintainer: Serge Schneider ***@***.***>
Architecture: arm64
Version: 16.0-1
Replaces: rpi-eeprom-images (<< 7.2)
Provides: rpi-eeprom-images
Depends: libraspberrypi-bin, python3, binutils, raspberrypi-bootloader (>=
1.20190819), pciutils
Recommends: flashrom
Breaks: rpi-eeprom-images (<< 7.2)
Conffiles:
 /etc/default/rpi-eeprom-update 0d657c25176302dace9384ca77aa3328
Description: Raspberry Pi 4 boot EEPROM updater
 Checks whether the Raspberry Pi bootloader EEPROM is up-to-date and updates
 the EEPROM.
Homepage: https://github.com/raspberrypi/rpi-eeprom/

***@***.***:~$ apt-mark showhold rpi-eeprom
E: Syntax error /etc/apt/apt.conf.d/01autoremove-kernels:1: Extra junk at
end of file

***@***.***:~$ apt-mark unhold rpi-eeprom
E: Syntax error /etc/apt/apt.conf.d/01autoremove-kernels:1: Extra junk at
end of file

Thanks for looking into this.

MichaIng, 2024-06-25:
> `E: Syntax error /etc/apt/apt.conf.d/01autoremove-kernels:1: Extra junk at end of file`

Oh, this file seems to be corrupted. Can you paste its content?
```sh
cat /etc/apt/apt.conf.d/01autoremove-kernels
```
However, you can just remove it. It has no effect on RPi anyway, and is not really needed at all:
```sh
rm /etc/apt/apt.conf.d/01autoremove-kernels
dietpi-update 1
```

mss1980au, 2024-06-26:
below is the result i got

***@***.***:~$ cat /etc/apt/apt.conf.d/01autoremove-kernels
�������������������������������������������������������������������������������������������������������������������������������������������������������������
***@***.***:~$

I ran the rm command to remove that file and then tried updating again but
got below

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
/boot/dietpi/.version: line 1: unexpected EOF while looking for matching ``'
/boot/dietpi/.version: line 2: syntax error: unexpected end of file
[ INFO ] DietPi-Pre-patches | Reverting now obsolete rpi-eeprom hold
[FAILED] DietPi-Pre-patches | apt-mark unhold rpi-eeprom

[FAILED] DietPi-Pre-patches | Unable to continue, DietPi-Pre-patches will
now terminate.
[FAILED] DietPi-Update | An error occurred during pre-patching. Please
check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors,
and rerun "dietpi-update" after the cause has been solved.

Please advise how to resolve this issue.

mss1980au, 2024-06-26:
hi

i just noticed it actually did not let me delete that file, says it is
"write protected", see below

***@***.***:~$ rm /etc/apt/apt.conf.d/01autoremove-kernels
rm: remove write-protected regular file
'/etc/apt/apt.conf.d/01autoremove-kernels'? y
rm: cannot remove '/etc/apt/apt.conf.d/01autoremove-kernels': Permission
denied
***@***.***:~$

mss1980au, 2024-06-26:
Hello

apologies for multiple emails, i logged in as root so I could remove
the file you mention,this helped remove the file. The update attempt got
further ahead, but I'm getting the below error now
screenshots and the last it that shows errors I've cut and paste, can you
help resolve?

[image: image.png]

[image: image.png]

debconf: delaying package configuration, since apt-utils is not installed

                         │ Fetched 99.2 MB in 25s (3913 kB/s)

                         │ (Reading database ... ^M(Reading database ...
5%^M(Reading database ... 10%^M(Reading database ... 15%^M(Reading datab

                         │ ... 20%^M(Reading database ... 25%^M(Reading
database ... 30%^M(Reading database ... 35%^M(Reading database ... 40%^M(

                         │ database ... 45%^M(Reading database ...
50%^M(Reading database ... 55%^M(Reading database ... 60%^M(Reading
database .
                         │ (Reading database ... 70%^Mdpkg: unrecoverable
fatal error, aborting:^M

                         │  files list file for package
'linux-libc-dev:arm64' is missing final newline^M

                         │ E: Sub-process /usr/bin/dpkg returned an error
code (2)

Joulinar, 2024-06-26:
for me it looks your system has quite some data corruption. Nearly every file we touch is damaged. 

Btw: if possible, login to GitHub directly instead of using mail. Not every information is show correctly using mail, like pictures are missing.

mss1980au, 2024-06-26:
please see images below

[image]

[image]

Hope this helps find the issue. Also below is the text cut and paste
The update process started well this time and got further ahead, but got stuck like below

APT upgrade                                                                                                                                
                     │  - Command: apt-get -y --with-new-pkgs upgrade                                                                                             
                     │  - Exit code: 100                                                                                                                          
                     │  - DietPi version: v8.16.2 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6                                                        
                     │  - Error log:                                                                                                                              
                     │ Reading package lists...                                                                                                                   
                     │ Building dependency tree...                                                                                                                
                     │ Reading state information...                                                                                                               
                     │ Calculating upgrade...                                                                                                                     
                     │ The following packages will be upgraded:                                                                                                   
                     │   adduser base-files bsdutils curl debian-archive-keyring dpkg dpkg-dev fdisk                                                              
                     │   file firmware-atheros firmware-brcm80211 firmware-iwlwifi                                                                                
                     │   firmware-misc-nonfree firmware-realtek grep isc-dhcp-client libblkid1                                                                    
                     │   libbsd0 libc-bin libc-dev-bin libc-l10n libc6 libc6-dev libcurl3-gnutls                                                                  
                     │   libcurl4 libdbus-1-3 libdpkg-perl libfdisk1 libglib2.0-0 libgnutls30                                                                     
                     │   libgssapi-krb5-2 libjson-c5 libk5crypto3 libkrb5-3 libkrb5support0                                                                       
                     │   libmagic-mgc libmagic1 libmount1 libncurses-dev libncurses5-dev libncurses6                                                              
                     │   libncursesw5-dev libncursesw6 libnghttp2-14 libperl5.32 libsmartcols1                                                                    
                     │   libssl-dev libssl1.1 libsystemd0 libtinfo6 libudev1 libuuid1 libx11-6                                                                    
                     │   libx11-data libx11-dev locales mount ncurses-base ncurses-bin openssh-client                                                             
                     │   openssh-server openssh-sftp-server openssl perl perl-base perl-modules-5.32                                                              
                     │   rpi-eeprom systemd systemd-sysv tar tzdata udev util-linux uuid-dev                                                                      
                     │ 74 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.                                                                            
                     │ Need to get 99.2 MB of archives.                                                                                                           
                     │ After this operation, 54.3 MB disk space will be freed.                                                                                    
                     │ Get:1 http://ftp.au.debian.org/debian bullseye/main arm64 base-files arm64 11.1+deb11u9 [70.2 kB]                                          
                     │ Get:2 https://deb.debian.org/debian-security bullseye-security/main arm64 bsdutils arm64 1:2.36.1-8+deb11u2 [145 kB]                       
                     │ Get:3 http://ftp.au.debian.org/debian bullseye/main arm64 tar arm64 1.34+dfsg-1+deb11u1 [830 kB]                                           
                     │ Get:4 https://deb.debian.org/debian-security bullseye-security/main arm64 libblkid1 arm64 2.36.1-8+deb11u2 [187 kB]                        
                     │ Get:5 https://deb.debian.org/debian-security bullseye-security/main arm64 uuid-dev arm64 2.36.1-8+deb11u2 [99.6 kB]                        
                     │ Get:6 https://deb.debian.org/debian-security bullseye-security/main arm64 libuuid1 arm64 2.36.1-8+deb11u2 [83.4 kB]                        
                     │ Get:7 https://deb.debian.org/debian-security bullseye-security/main arm64 libfdisk1 arm64 2.36.1-8+deb11u2 [224 kB]                        
[... cut ...]
                     │ database ... 45%^M(Reading database ... 50%^M(Reading database ... 55%^M(Reading database ... 60%^M(Reading database .                     
                     │ (Reading database ... 70%^Mdpkg: unrecoverable fatal error, aborting:^M                                                                    
                     │  files list file for package 'linux-libc-dev:arm64' is missing final newline^M                                                             
                     │ E: Sub-process /usr/bin/dpkg returned an error code (2)

Joulinar, 2024-06-26:
can you share kernel error log

```
dmesg -l 0,1,2,3
```

would be enough to share the console output (copy/paste). No need to do screen shots

mss1980au, 2024-06-26:
root@DietPi:~# dmesg -1 0,1,2,3
dmesg: invalid option -- '1'
Try 'dmesg --help' for more information.
root@DietPi:~#

MichaIng, 2024-06-26:
Do you have a backup where individual files can be easily restored from or even the whole system of more/other files turn out to be affected?

However, you can restore an individual package's file list like that:
```sh
cd /tmp
apt download linux-libc-dev
dpkg-deb -R linux-libc-dev*.deb package
find package -mindepth 1 ! \( -path 'package/DEBIAN*' -prune \) | sed 's/^package//' > /var/lib/dpkg/info/linux-libc-dev:arm64.list
rm -R package*
apt update
apt upgrade
```

mss1980au, 2024-06-26:
Unfortunately don't have a backup. I ran the commands you mentioned, all worked well until last command "apt upgrade", see output below. Say "apt-utils is not installed", seems we are close to resolving this...hopefully.

root@DietPi:~# cd /tmp
root@DietPi:/tmp# apt download linux-libc-dev
Get:1 https://archive.raspberrypi.com/debian bullseye/main arm64 linux-libc-dev arm64 1:1.20230405-1 [1135 kB]
Fetched 1135 kB in 4s (303 kB/s)         
root@DietPi:/tmp# dpkg-deb -R linux-libc-dev*.deb package
root@DietPi:/tmp# find package -mindepth 1 ! \( -path 'package/DEBIAN*' -prune \) | sed 's/^package//' > /var/lib/dpkg/info/linux-libc-dev:arm64.list
root@DietPi:/tmp# rm -R package*
root@DietPi:/tmp# apt update
Hit:1 http://ftp.au.debian.org/debian bullseye InRelease
Get:2 http://ftp.au.debian.org/debian bullseye-updates InRelease [44.1 kB]                                                      
Get:3 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]                                                          
Get:4 http://ftp.au.debian.org/debian bullseye-backports InRelease [49.0 kB]                                       
Get:5 https://deb.debian.org/debian-security bullseye-security/main arm64 Packages [271 kB]                               
Get:6 https://archive.raspberrypi.com/debian bullseye InRelease [23.6 kB]
Get:7 https://archive.raspberrypi.com/debian bullseye/main arm64 Packages [307 kB]
Fetched 743 kB in 5s (159 kB/s)    
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
74 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  adduser base-files bsdutils curl debian-archive-keyring dpkg dpkg-dev fdisk file firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree
  firmware-realtek grep isc-dhcp-client libblkid1 libbsd0 libc-bin libc-dev-bin libc-l10n libc6 libc6-dev libcurl3-gnutls libcurl4 libdbus-1-3 libdpkg-perl libfdisk1
  libglib2.0-0 libgnutls30 libgssapi-krb5-2 libjson-c5 libk5crypto3 libkrb5-3 libkrb5support0 libmagic-mgc libmagic1 libmount1 libncurses-dev libncurses5-dev libncurses6
  libncursesw5-dev libncursesw6 libnghttp2-14 libperl5.32 libsmartcols1 libssl-dev libssl1.1 libsystemd0 libtinfo6 libudev1 libuuid1 libx11-6 libx11-data libx11-dev
  locales mount ncurses-base ncurses-bin openssh-client openssh-server openssh-sftp-server openssl perl perl-base perl-modules-5.32 rpi-eeprom systemd systemd-sysv tar
  tzdata udev util-linux uuid-dev
74 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/99.2 MB of archives.
After this operation, 54.3 MB disk space will be freed.
Do you want to continue? [Y/n] y
debconf: delaying package configuration, since apt-utils is not installed
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libxau-dev:arm64' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

MichaIng, 2024-06-27:
Still the same issue, just another package. Repeat above steps, but replace `linux-libc-dev` with `libxau-dev`.

The `debconf` warning is no error and unrelated. It is expected on DietPi, since we do not see `debconf` pre-generation as a real benefit, and script-internal APT calls skip the rare `debconf` dialogs, using reasonable defaults.

mss1980au, 2024-06-28:
I repeated those steps and got another issue with another package again, followed same steps to update that package also.
Once these packages were sorted, the pi upgraded to latest version of Dietpi. You can now close this issue as you have successfully resolved it.

Thanks so much for your help, it's very much appreciated !!

closed 2024-06-28 by Joulinar