MichaIng/DietPi #7127: Wifi Hotspot Extender

actraiser30, 2024-06-24:
Hello

Is it possible to add to the wifi hotspot software an extra option(optional) 
If a user has a second USB wifi connected that can use it as a wifi extender/repeater??

For example the internal/onboard wifi as a hotspot and an external USB wifi as an extender/repeater to increase the wifi coverage.

Thanks

MichaIng, 2024-06-24:
Does it make any sense to use a hotspot and an extender on the same device? If onboard WiFi has a worse range than the USB WiFi adapter, I would use the latter only and disable the onboard WiFi adapter.

But an option to to configure the DietPi system as repeater _only_ would be neat.

actraiser30, 2024-06-24:
would be great such an option even if it's in the same device nevertheless I can always get a long usb extension cable and put the extender usb wifi to another room.

MichaIng, 2024-06-24:
Okay, I am currently not sure how to set this up correctly, probably best with a bridge, running `hostapd` and the DHCP server on the bridge interface. So the extender would not really be an extender, but just a clone of the hotspot itself. This has the benefit that the WiFi signal is not further forwarded between the two adapters, but via USB between "extender" and DietPi system directly, resulting in better performance.

... while writing, I had the word "repeater" in mind, and was wondering about the difference: https://www.howtogeek.com/774725/wi-fi-extender-vs-booster-vs-repeater-whats-the-difference/
Okay, so "extender" is in fact this solution with signal transport via USB (though it is probably more meant with a standalone device, instead of a USB network adapter only, but should work either way), while a "repeater" would be one which did not make so much sense for me, when managed from the same host system. So then this all makes sense:
- One adapter for Internet/LAN access for the host.
- One adapter as WiFi hotspot.
- Optionally another adapter as WiFi extender, practically implemented by moving the hotspot (software) to a bridge between the two WiFi interfaces.

However, in any case, this would be an addition to the `dietpi-network` script which is somewhat planned without any ETA.

actraiser30, 2024-06-24:
Sounds good the wifi clone "extender" option if it's possible, as far I know the repeater, repeat the wifi and the Extender, extends it.

thanks.

Joulinar, 2024-06-24:
this can be done using RaspAP. There you can specify a 2nd WiFi interface https://docs.raspap.com/repeater/

MichaIng, 2024-06-24:
By the word "extend" alone, I would not know the difference to "repeat" in this context 😄. However, the article made it clear.

When using RaspAP, just take care to not use `dietpi-config` anymore, since both naturally conflict. But it is good that we can check how RaspAP implements this, so we can verify whether what we are doing is somewhat correct or nonsense 😄.

actraiser30, 2024-06-25:
Ok thanks,I used in the past RasAp it locked everything in my Raspbian I didn't have access to anything so I wouldn't think is a good choice to use it.

Another thing now,sometimes when trying to connect to the hotspot it stays in the optaining ip address message when connecting to it and eventually disconnects with a failed to connect message,if I try to connect again it does the same,all I can do is to turn off the raspberry pi by removing the power,this can happen 4,6 times in a row,is it a bug??