MichaIng/DietPi #7128: Node-red reinstall issue

thinkbig1979, 2024-06-25:
Hit an error after running `dietpi-software reinstall 122` 
Muy goal was to upgrade my existing nod-red installation to version 4.0.

All was proceeding well until here:

```bash
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Node-RED: tool for wiring devices, APIs and online services

[ INFO ] DietPi-Software | APT install python3, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
python3 is already the newest version (3.9.2-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
[  OK  ] DietPi-Software | APT install python3
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/node-red
[  OK  ] DietPi-Software | usermod -aG dialout,gpio,i2c,spi -d /mnt/dietpi_userdata/node-red -s /usr/sbin/nologin nodered
[  OK  ] DietPi-Software | eval echo 'nodered ALL=NOPASSWD: ALL' > /etc/sudoers.d/nodered
[  OK  ] DietPi-Software | chown -R nodered:nodered /mnt/dietpi_userdata/node-red
[  OK  ] DietPi-Software | cd /mnt/dietpi_userdata/node-red
[ INFO ] DietPi-Software | sudo -u nodered npm i --no-audit node-red, please wait...
npm error Class extends value undefined is not a constructor or null
npm error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-debug-0.log
[FAILED] DietPi-Software | sudo -u nodered npm i --no-audit node-red
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 47eb0dca-f49f-4498-a3bd-19126e1227f0

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

```

I used the built in bug report function, and as you can see it created a reference code for the issue.

-------------
System info:

- Date           | Tue Jun 25 13:23:58 CEST 2024
- Bug report     | 47eb0dca-f49f-4498-a3bd-19126e1227f0
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `sudo -u nodered npm i --no-audit node-red`

MichaIng, 2024-06-25:
Can you show the content of the log file:
```sh
cat /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-debug-0.log
```

thinkbig1979, 2024-06-25:
Sure, here you go:

```bash
~ ❯ cat /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-debug-0.log
0 verbose cli /usr/local/bin/node /usr/local/bin/npm
1 info using npm@10.8.1
2 info using node@v22.3.0
3 silly config load:file:/usr/local/lib/node_modules/npm/npmrc
4 silly config load:file:/mnt/dietpi_userdata/node-red/.npmrc
5 silly config load:file:/usr/local/etc/npmrc
6 verbose title npm i node-red
7 verbose argv "i" "--no-audit" "node-red"
8 verbose logfile logs-max:10 dir:/mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-
9 verbose logfile /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-debug-0.log
10 silly logfile start cleaning logs, removing 2 files
11 verbose stack TypeError: Class extends value undefined is not a constructor or null
11 verbose stack     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/minipass-collect/index.js:4:23)
11 verbose stack     at Module._compile (node:internal/modules/cjs/loader:1460:14)
11 verbose stack     at Module._extensions..js (node:internal/modules/cjs/loader:1544:10)
11 verbose stack     at Module.load (node:internal/modules/cjs/loader:1275:32)
11 verbose stack     at Module._load (node:internal/modules/cjs/loader:1091:12)
11 verbose stack     at wrapModuleLoad (node:internal/modules/cjs/loader:212:19)
11 verbose stack     at Module.require (node:internal/modules/cjs/loader:1297:12)
11 verbose stack     at require (node:internal/modules/helpers:123:16)
11 verbose stack     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/cacache/lib/get.js:3:17)
11 verbose stack     at Module._compile (node:internal/modules/cjs/loader:1460:14)
12 error Class extends value undefined is not a constructor or null
13 verbose cwd /mnt/dietpi_userdata/node-red
14 verbose os Linux 6.1.21-v7+
15 verbose node v22.3.0
16 verbose npm  v10.8.1
17 verbose exit 1
18 verbose code 1
19 error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T11_23_57_442Z-debug-0.log
```

MichaIng, 2024-06-25:
Hmm, `npm` does a pretty bad job resolving conflicts between global and local modules. Looks like the `minipass-collect` (which is part of the Node.js core installation) is too new for some of the locally installed Node-RED dependencies. I just do not get why this matters when one explicitly asks to install=update those local modules. However, you are not the first one who runs into such issue, when upgrading from certain older Node-RED versions. So let's solve this once and for all.

Can you test this:
```sh
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm i --no-audit node-red@latest
```
And if it does not work, this:
```sh
cd /mnt/dietpi_userdata/node-red
sudo -u nodered npm up --no-audit node-red
```
The hammer method is to remove all local modules and install them freshly. But if one of the above works, I would prefer it, causing less traffic and disk I/O:
```sh
cd /mnt/dietpi_userdata/node-red
sudo rm -R node_modules
npm i --no-audit node-red
```

thinkbig1979, 2024-06-25:
Tested the first 2 options:

```bash
/mnt/dietpi_userdata/node-red ❯ sudo -u nodered npm i --no-audit node-red@latest
npm error Class extends value undefined is not a constructor or null
npm error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_11_26_351Z-debug-0.log

/mnt/dietpi_userdata/node-red ❯ sudo -u nodered npm up --no-audit node-red
npm error Class extends value undefined is not a constructor or null
npm error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_11_59_706Z-debug-0.log
``` 

Will look into the third option asap. 
First want to make sure I do not have any manually installed stuff that could trip the installation up.

MichaIng, 2024-06-25:
> First want to make sure I do not have any manually installed stuff that could trip the installation up.

Actually, when plugins were installed manually, they should show up in `/mnt/dietpi_userdata/node-red/package.json`. Can you test this first?
```sh
cd /mnt/dietpi_userdata/node-red
sudo rm -R node_modules
sudo -u nodered npm up
```
That should install and update all packages defined in `package.json` and all their dependencies.

thinkbig1979, 2024-06-25:
Yes, exactly what I just did. :-)
Everything in there was installed through node-red, so I deleted the node_modules directory and tried again. 
No luck:

```bash
/mnt/dietpi_userdata/node-red ❯ cat /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_22_15_416Z-debug-0.log
0 verbose cli /usr/local/bin/node /usr/local/bin/npm
1 info using npm@10.8.1
2 info using node@v22.3.0
3 silly config load:file:/usr/local/lib/node_modules/npm/npmrc
4 silly config load:file:/mnt/dietpi_userdata/node-red/.npmrc
5 silly config load:file:/usr/local/etc/npmrc
6 verbose title npm i node-red
7 verbose argv "i" "--no-audit" "node-red"
8 verbose logfile logs-max:10 dir:/mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_22_15_416Z-
9 verbose logfile /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_22_15_416Z-debug-0.log
10 silly logfile start cleaning logs, removing 2 files
11 verbose stack TypeError: Class extends value undefined is not a constructor or null
11 verbose stack     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/minipass-collect/index.js:4:23)
11 verbose stack     at Module._compile (node:internal/modules/cjs/loader:1460:14)
11 verbose stack     at Module._extensions..js (node:internal/modules/cjs/loader:1544:10)
11 verbose stack     at Module.load (node:internal/modules/cjs/loader:1275:32)
11 verbose stack     at Module._load (node:internal/modules/cjs/loader:1091:12)
11 verbose stack     at wrapModuleLoad (node:internal/modules/cjs/loader:212:19)
11 verbose stack     at Module.require (node:internal/modules/cjs/loader:1297:12)
11 verbose stack     at require (node:internal/modules/helpers:123:16)
11 verbose stack     at Object.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/cacache/lib/get.js:3:17)
11 verbose stack     at Module._compile (node:internal/modules/cjs/loader:1460:14)
12 error Class extends value undefined is not a constructor or null
13 verbose cwd /mnt/dietpi_userdata/node-red
14 verbose os Linux 6.1.21-v7+
15 verbose node v22.3.0
16 verbose npm  v10.8.1
17 verbose exit 1
18 verbose code 1
19 error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-06-25T14_22_15_416Z-debug-0.log
```

MichaIng, 2024-06-25:
Hmm, last idea is the cache:
```sh
sudo rm -R .npm/_cacache
sudo -u nodered npm up --no-audit
```

thinkbig1979, 2024-06-25:
So after some searching online I found that this is an error that sometimes occurs when using the `n` tool to manage nod versions, which is what I have been doing for years. I deleted the n directory, then removed and reinstalled node-js using the dietpi-software tool. 
Then used option 1 from your earlier post (https://github.com/MichaIng/DietPi/issues/7128#issuecomment-2189004611) to install the latest node-red.

Node and node-red are now successfully running again, and node-red is on the latest version. I have not lost any data in node-red, so that's a win. Doing this does seem to have broken my pm2 install though, but I was only using pm2 for a single service, so fixing that should be relatively easy.

Happy I got this far and thanks for your help on this!

MichaIng, 2024-06-25:
Ah yeah that makes sense, so the Node version did not match the `npm` version. If `n` is already installed, we could actually just use it to update Node, instead of running our installer.

However, there were as well issues in the past related to incompatible local Node modules/dependencies. So if a reinstall is done, which includes a Node.js update, it makes sense to all modules from scratch. I also disabled the local cache now: https://github.com/MichaIng/DietPi/commit/611c047
This increases traffic and processing. However, it makes sense for a reinstall, while for a Node-RED update only, it can be done with much less overhead, according to our docs: https://dietpi.com/docs/software/hardware_projects/#__tabbed_6_5

Additionally, our Node.js installer does not remove the global `npm` and `corepack` modules, before unpacking the new archive. This alone might have solved your issue already. Of course, it basically replaces the any Node version which may have been installed via `n`. I do not remember exactly how it works, but I think it creates symlinks for modules and executables, which are replaced with actual files/dirs when the archive is unpacked.

closed 2024-06-25 by MichaIng