MichaIng/DietPi #7129: Amlogic S802 support

DealsBeam, 2024-06-26:
I have an Amlogic S802 Android TV box that I'd like to run Linux and Pi-Hole on and I'm wondering if DietPi would ever get support for it? I've been doing research on how I could get Linux running on it, I've come across some informative pages but none that say they support Amlogic S802 SoC

https://linux-meson.com/index.html

The M8 Amlogic S802 Android TV box specs:

CPU	Amlogic S802 Cortex A9 r4 2GHz  Quad Core
GPU	Octa-core Mali-450MP GPU @ 600MHz
RAM	2GB DDR3
ROM	8GB NAND FLASH
Wifi Connectivity	2.4G/5G, wireless mouse and keyboard via 2.4GHz USB dongle
Ethernet	10/100M, standard RJ-45
Bluetooth	Support  
HDD File System 	FAT16/FAT32/NTFS
Card reader 	SD/SDHC/MMC cards
HDMI Port	Standard HDMI female, 1.4b
Other Interfaces	2 x High speed USB 2.0, support USB DISK and USB HDD ;1 x RJ45 ;1 x AV port ;1 x Optical ;1 x SD card slot
Power	DC 5V/2A

MichaIng, 2024-10-19:
A natural problem with these TV boxes. They are extremely badly supported. The SoC ia not the problem, but the SBC/board around it.

The manufacturer does not provide any Linux image, but Android only?

MichaIng, 2024-10-24:
Yes, Armbian and based on it DietPi support the Orange Pi Zero 3.

I'll mark the issue as closed then. Direct support for TV boxes is out of scope for us.

closed 2024-10-24 by MichaIng