MichaIng/DietPi #7131: installing coolify

NagaMerahMeriah, 2024-06-28:
i have orange pi 5 plus board installed with dietpi latest version. This board have 2 storage, **SSD NVME** and **Micro SD** (i install dietpi from Micro SD ).

i change user data location to **SSD NVME**, but when i install coolify ([with this method](https://coolify.io/docs/installation)) + deploy some apps, i check the storage left with `df -h`, it shows:

```bash
Filesystem      Size  Used Avail Use% Mounted on
udev            7.6G     0  7.6G   0% /dev
tmpfs           1.6G   83M  1.5G   6% /run
/dev/mmcblk1p1   30G   12G   17G  41% /
tmpfs           7.7G     0  7.7G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           7.7G     0  7.7G   0% /tmp
tmpfs            50M   12K   50M   1% /var/log
/dev/nvme0n1p1  229G   48K  229G   1% /mnt/10cdeef9-d836-4cfa-a457-313c7ef23e35
```

i expected the coolify and my deployed apps is located on **SSD NVME** (/dev/nvme0n1p1).
1. how can i move those installed app from **Micro SD** to **SSD NMVE**? 
1. or how can i install by default to **SSD NMVE**? 

is there any step can i achieve that?

thank youuu

MichaIng, 2024-06-28:
The installer uses `/data/coolify` as hardcoded directory. You could either create `/data/coolify` or `/data` itself as symlink to the NVMe mount, or mount the NVMe right to `/data/coolify` or `/data`.

oxwazz, 2024-06-28:
you need both Micro SD and NVMe for the storage?

if you only need NVMe, you can move your system from Micro SD to NVMe ([tutorial](https://dietpi.com/blog/?p=1236)), and then all applications that you will install will be directly stored in NVMe.

NagaMerahMeriah, 2024-06-29:
> The installer uses `/data/coolify` as hardcoded directory. You could either create `/data/coolify` or `/data` itself as symlink to the NVMe mount, or mount the NVMe right to `/data/coolify` or `/data`.
> 
> Shall the NVMe drive contain any other data, or just coolify and nothing else?

@MichaIng you have any tutorial i can follow? thanks

> you need both Micro SD and NVMe for the storage?
> 
> if you only need NVMe, you can move your system from Micro SD to NVMe ([tutorial](https://dietpi.com/blog/?p=1236)), and then all applications that you will install will be directly stored in NVMe.

@oxwazz ok i will try this first, i dont know if DietPi can installed on NVMe only, without Micro SD. thanks

MichaIng, 2024-06-29:
> @MichaIng you have any tutorial i can follow? thanks

The question is how you want it. I think cleanest and with max flexibility and least change to cause issues is to mount the NVMe to `/data` and let the installer create `coolify` directory within it. Respectively now, that it has been installed already, you can just move it there. The commands below depend on the assumptions about existing dirs and mount point from our info above. Please do not run a next command, when a previous one failed. For sure, you'll need to stop the coolify container first, but I don't know its name or ID.
```sh
mv /data/coolify /coolify_bak
umount /mnt/10cdeef9-d836-4cfa-a457-313c7ef23e35
sed -i 's|/mnt/10cdeef9-d836-4cfa-a457-313c7ef23e35|/data|' /etc/fstab
systemctl daemon-reload
mount /data
mv /coolify_bak /data/coolify
```

Running the whole system on NVMe surely is the better solution, for performance and since an NVMe SSD is usually much more durable and reliable compared to an SD card. It works if the SPI storage contains a compatible bootloader. IIRC, the one shipped with OPi 5 Plus works. However, you can also update it to use the same that currently boots the SD card via `dietpi-config` > Advanced Options > Update SPI bootloader.

NagaMerahMeriah, 2024-06-29:
@MichaIng @oxwazz thank you for the help, i ended up with running the whole system on NVMe, and i think the booting SOO MUCH faster than SD Card.

closed 2024-06-29 by NagaMerahMeriah

MichaIng, 2024-06-29:
Perfect!

Out of interest, did it work OOTB or did you need to flash our bootloader first?

NagaMerahMeriah, 2024-06-29:
I just follow the [tutorial](https://dietpi.com/blog/?p=1236), only using `dd` command and it works perfectly.

MichaIng, 2024-06-29:
So it works with the original Orange Pi SPI bootloader, that is good. I am not sure anymore whether I tested it, when I received mine. I flashed ours in the meantime, for testing.