MichaIng/DietPi #7138: ODROID N2+ can't find USB DAC with latest packages

Kreeblah, 2024-07-07:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.5.1
- Distro version | bookworm
- Kernel version | Linux odroidn2p-1 6.6.32-current-meson64 #1 SMP PREEMPT Sat May 25 14:22:56 UTC 2024 aarch64 GNU/Linux
- SBC model | Odroid N2 (aarch64)
- Power supply used | 12V/2A from AmeriDroid
- SD card used | eMMC Module

#### Additional Information (if applicable)
- Software title | dietpi-config
- Was the software title installed freshly or updated/migrated? Updated
- Can this issue be replicated on a fresh installation of DietPi? Haven't tried yet

- Bug report ID | b67ce171-0ae2-401a-af5d-0fc7afb8b582

#### Steps to reproduce
1. Install DietPi
2. Install the Roon bridge software, or something else that installs audio packages
3. Update all packages to latest versions and reboot
4. Attempt to set the sound card to a USB DAC in dietpi-config (Audio Options -> Sound card)

#### Expected behaviour
- DietPi should set the sound card to the selected USB DAC

#### Actual behaviour
- DietPi is failing to find the DAC with the following output in the terminal:

```
[ SUB1 ] DietPi-Set_hardware > soundcard (usb-dac)
[ INFO ] DietPi-Set_hardware | Checking for required APT packages: alsa-utils
[ INFO ] DietPi-Set_hardware | Resetting all sound card settings...
Found hardware: "axg-sound-card" "" "" "" ""
Hardware is initialized using a generic method
[ INFO ] DietPi-Set_hardware | Applying new sound card settings...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=usb-dac
[FAILED] DietPi-Set_hardware | Unable to find any USB DAC on your system, reverting to "none"...
[  OK  ] DietPi-Set_hardware | Setting in /boot/dietpi.txt adjusted: CONFIG_SOUNDCARD=none
[FAILED] soundcard usb-dac | Exited with error
```

#### Extra details
- This DAC worked fine before updating these packages earlier today.  I use this SBC as a Roon bridge, and noticed that Roon didn't see it when it came back up, so when I SSHd into the device, I found that it didn't have the USB DAC selected any more, and I can no longer select it for audio output.

MichaIng, 2024-07-07:
> This DAC worked fine before updating these packages earlier today.

Do you remember which kernel version you upgraded from?

I guess it did not work with out latest image from last month, since the kernel version is already `6.6.32`. Other packages can hardly have an effect on USB sound card detection.

Also can you please show the output of this:
```sh
aplay -l
```

Kreeblah, 2024-07-07:
> Do you remember which kernel version you upgraded from?

Looks like I still have the output from the package update in my terminal buffer, so here's what it was.  This was from an `apt-get dist-upgrade` since I was already on 9.5.1 from when I updated to that last month.  I don't actually see a kernel update, though, unless I'm missing something.

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-firmware base-files bash curl libcurl4 libglib2.0-0 libgnutls30
  libgssapi-krb5-2 libk5crypto3 libkrb5-3 libkrb5support0 libseccomp2 libssl3
  libsystemd-shared libsystemd0 libtommath1 libudev1 nano openssl
  shairport-sync systemd systemd-sysv systemd-timesyncd tailscale u-boot-tools
  udev wpasupplicant
27 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 135 MB of archives.
After this operation, 1590 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 https://deb.debian.org/debian bookworm/main arm64 base-files arm64 12.4+deb12u6 [70.8 kB]
Get:2 https://deb.debian.org/debian bookworm/main arm64 bash arm64 5.2.15-2+b7 [1444 kB]
Get:3 https://deb.debian.org/debian bookworm/main arm64 libseccomp2 arm64 2.5.4-1+deb12u1 [45.5 kB]
Get:4 https://deb.debian.org/debian bookworm/main arm64 libssl3 arm64 3.0.13-1~deb12u1 [1808 kB]
Get:5 https://deb.debian.org/debian bookworm/main arm64 systemd-timesyncd arm64 252.26-1~deb12u2 [61.5 kB]
Get:6 https://deb.debian.org/debian bookworm/main arm64 systemd arm64 252.26-1~deb12u2 [2832 kB]
Get:7 https://dietpi.com/apt all/odroidn2 all armbian-firmware all 24.8.0-trunk-dietpi1 [91.2 MB]
Get:8 https://deb.debian.org/debian bookworm/main arm64 libsystemd-shared arm64 252.26-1~deb12u2 [1509 kB]
Get:10 https://deb.debian.org/debian bookworm/main arm64 libsystemd0 arm64 252.26-1~deb12u2 [308 kB]
Get:11 https://deb.debian.org/debian bookworm/main arm64 systemd-sysv arm64 252.26-1~deb12u2 [42.3 kB]
Get:12 https://deb.debian.org/debian bookworm/main arm64 libgnutls30 arm64 3.7.9-2+deb12u3 [1313 kB]
Get:13 https://deb.debian.org/debian bookworm/main arm64 udev arm64 252.26-1~deb12u2 [1619 kB]
Get:9 https://pkgs.tailscale.com/stable/debian bookworm/main arm64 tailscale arm64 1.68.2 [25.9 MB]
Get:14 https://deb.debian.org/debian bookworm/main arm64 libudev1 arm64 252.26-1~deb12u2 [102 kB]
Get:15 https://deb.debian.org/debian bookworm/main arm64 nano arm64 7.2-1+deb12u1 [680 kB]
Get:16 https://deb.debian.org/debian-security bookworm-security/main arm64 libgssapi-krb5-2 arm64 1.20.1-2+deb12u2 [124 kB]
Get:17 https://deb.debian.org/debian-security bookworm-security/main arm64 libkrb5-3 arm64 1.20.1-2+deb12u2 [313 kB]
Get:18 https://deb.debian.org/debian-security bookworm-security/main arm64 libkrb5support0 arm64 1.20.1-2+deb12u2 [31.8 kB]
Get:19 https://deb.debian.org/debian-security bookworm-security/main arm64 libk5crypto3 arm64 1.20.1-2+deb12u2 [79.4 kB]
Get:20 https://deb.debian.org/debian bookworm/main arm64 curl arm64 7.88.1-10+deb12u6 [308 kB]
Get:21 https://deb.debian.org/debian bookworm/main arm64 libcurl4 arm64 7.88.1-10+deb12u6 [366 kB]
Get:22 https://deb.debian.org/debian bookworm/main arm64 libglib2.0-0 arm64 2.74.6-2+deb12u3 [1313 kB]
Get:23 https://deb.debian.org/debian bookworm/main arm64 libtommath1 arm64 1.2.0-6+deb12u1 [62.8 kB]
Get:24 https://deb.debian.org/debian bookworm/main arm64 openssl arm64 3.0.13-1~deb12u1 [1385 kB]
Get:25 https://deb.debian.org/debian bookworm/main arm64 u-boot-tools arm64 2023.01+dfsg-2+deb12u1 [189 kB]
Get:26 https://deb.debian.org/debian bookworm/main arm64 wpasupplicant arm64 2:2.10-12+deb12u1 [1304 kB]
Get:27 https://dietpi.com/apt bookworm/main arm64 shairport-sync arm64 4.3.3-dietpi1 [160 kB]
Fetched 135 MB in 5s (29.7 MB/s)          
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 23376 files and directories currently installed.)
Preparing to unpack .../base-files_12.4+deb12u6_arm64.deb ...
Unpacking base-files (12.4+deb12u6) over (12.4+deb12u5) ...
Setting up base-files (12.4+deb12u6) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 23376 files and directories currently installed.)
Preparing to unpack .../bash_5.2.15-2+b7_arm64.deb ...
Unpacking bash (5.2.15-2+b7) over (5.2.15-2+b2) ...
Setting up bash (5.2.15-2+b7) ...
update-alternatives: using /usr/share/man/man7/bash-builtins.7.gz to provide /usr/share/man/man7/builtins.7.gz (builtins.7.gz) in auto mode
(Reading database ... 23376 files and directories currently installed.)
Preparing to unpack .../libseccomp2_2.5.4-1+deb12u1_arm64.deb ...
Unpacking libseccomp2:arm64 (2.5.4-1+deb12u1) over (2.5.4-1+b3) ...
Setting up libseccomp2:arm64 (2.5.4-1+deb12u1) ...
(Reading database ... 23375 files and directories currently installed.)
Preparing to unpack .../libssl3_3.0.13-1~deb12u1_arm64.deb ...
Unpacking libssl3:arm64 (3.0.13-1~deb12u1) over (3.0.11-1~deb12u2) ...
Preparing to unpack .../systemd-timesyncd_252.26-1~deb12u2_arm64.deb ...
Unpacking systemd-timesyncd (252.26-1~deb12u2) over (252.22-1~deb12u1) ...
Setting up libssl3:arm64 (3.0.13-1~deb12u1) ...
(Reading database ... 23375 files and directories currently installed.)
Preparing to unpack .../systemd_252.26-1~deb12u2_arm64.deb ...
[... 79 lines cut ...]
W: No zstd in /usr/bin:/sbin:/bin, using gzip
update-initramfs: Converting to U-Boot format
[... cut ...]
Load Address: 00000000
Entry Point:  00000000
'/boot/uInitrd' -> 'uInitrd-6.6.32-current-meson64'
```

As far as `aplay -l`, I get this:

```
**** List of PLAYBACK Hardware Devices ****
card 0: ODROIDN2 [ODROID-N2], device 0: fe.dai-link-0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ODROIDN2 [ODROID-N2], device 1: fe.dai-link-1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ODROIDN2 [ODROID-N2], device 2: fe.dai-link-2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

MichaIng, 2024-07-07:
Hmm, the packages which were upgraded should not have affected this. But the USB DAC is indeed not listed at all. Probably it is some indirect effect of insufficient power/voltage. Are there any kernel errors, especially when you detach and re-attach the USB DAC?
```sh
dmesg -l 0,1,2,3
```
Or do you see it being recognised as USB device at all?
```sh
dmesg | tail -10
lsusb
```

Kreeblah, 2024-07-08:
In case it was port-specific, I tried that with all four ports.  With one of them (the lower-right port), I got this in `dmesg`:

```
[43496.627305] usb 1-1.4: new high-speed USB device number 3 using xhci-hcd
[43496.751595] usb 1-1.4: New USB device found, idVendor=2fc6, idProduct=f829, bcdDevice= 2.01
[43496.751619] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[43496.751628] usb 1-1.4: Product: Hidizs S9Pro
[43496.751635] usb 1-1.4: Manufacturer: Hidizs
[43496.751642] usb 1-1.4: SerialNumber: Hidizs-S9Por
[43496.940620] usb 1-1.4: 1:4 : unsupported format bits 0x100000000
[43497.037766] usbcore: registered new interface driver snd-usb-audio
[43507.725104] usb 1-1.4: USB disconnect, device number 3
```

However, it had that immediate disconnect there and didn't show up for `lsusb`.  I'm also noticing that the light on it is off when it's plugged into the SBC, but it turns on (and is recognized/works fine) when plugged into my laptop.  So, I think it is some sort of power/voltage issue, which is odd since it never had a problem with that before with the N2+.

Kreeblah, 2024-07-08:
With the potential for a power issue, I double-checked my power supply, since I copied/pasted my power supply info from an old issue I filed (everything else was current, though).  Sure enough, I forgot that I'd upgraded the power supply a while back (when I got the external DAC) to a Mean Well GST60A12-P1J, which is a 12V/5A power supply.  I'd think that should be enough, given that the DAC itself doesn't even come close to drawing a watt, and I don't have anything else plugged into the N2+.

MichaIng, 2024-07-08:
Strange. The packages upgrades are all definitely unrelated, especially since it is on raw USB device detection. I have also no idea why it would start to have connectivity or detection issues just suddenly.

However, I compiled new kernel packages, if you want to give them a try:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-meson64.deb
dpkg -i linux-{image,dtb}-current-meson64.deb
reboot
```

Kreeblah, 2024-07-08:
Huh.  Those new kernel packages actually resolve it.  The DAC is consistently being detected/initialized now, I can assign it as the sound output for the SBC, and it works just fine again.

kartikynwa, 2024-07-10:
I am not sure if this is related but I updated the kernel for Odroid N2+ today and since the reboot my USB wifi adapter does not work anymore. I get an error message saying:

```Unable to handle kernel NULL pointer deference at virtual address 000000000000```

Is there a way I could try installing the previous kernel package? I see that the packages mentioned in https://github.com/MichaIng/DietPi/issues/7138#issuecomment-2215496278 have been removed. Presumably moved to https://dietpi.com/downloads/binaries but I can't say for sure. I tried reinstalling the deb packages located at `https://dietpi.com/downloads/binaries/linux-{image,dtb}-current-meson64.deb` but I still get the same error.

The USB wifi adapter in question is a `TP-Link Archer T2U PLUS [RTL8821AU]` as listed by `lsusb`.

MichaIng, 2024-07-10:
> The USB wifi adapter in question is a TP-Link Archer T2U PLUS [RTL8821AU] as listed by lsusb.

When it is listed in `lsusb`, then it is a different issue. It does not appear in `ip list`, I guess? Can you check for kernel errors?
```sh
dmesg -l 0,1,2,3
```
... I am currently checking the kernel modules, I think I found it already.

kartikynwa, 2024-07-10:
I can't copy paste the output since I can't ssh into the machine right now. But I have taken a picture of it. I hope it helps. 

[image]

The adapter shows up under lsusb. It always shows up under `iwctl station list` output. It just seems to not be able to detect any wifi networks at all. I suggested it could be related because it happened after updating the `linux-{current,dtb}*` packages today and rebooting.

kartikynwa, 2024-07-10:
Additionally, in the output for `ip link`, the device's state is shown as `DORMANT`. Google tells me this is related to power saving but I am a bit out of my depth here. Thanks for looking into it.

MichaIng, 2024-07-10:
This seems to be a power savings mode. What does this show?
```sh
iw dev wlan0 get power_save
```
And does this help?
```sh
iw dev wlan0 set power_save off
```

kartikynwa, 2024-07-10:
Says `Power save: off` 😅

kartikynwa, 2024-07-10:
Sharing some outputs.

This is for `ip link`:

```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP mode DORMANT group default qlen 1000         link/ether 98:48:27:9c:e8:4d brd ff:ff:ff:ff:ff:ff
```

This is for `ip a`:

```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP group default qlen 1000                      link/ether 98:48:27:9c:e8:4d brd ff:ff:ff:ff:ff:ff      inet6 fe80::9a48:27ff:fe9c:e84d/64 scope link              valid_lft forever preferred_lft forever
```

This is for `iw dev wlan0 info`:

```
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 98:48:27:9c:e8:4d
        type managed
        wiphy 0
        channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
        txpower 20.00 dBm
```

MichaIng, 2024-07-10:
Linking some explanation of the DORMANT mode: https://askubuntu.com/questions/1194048/what-does-mean-dormant-mode-in-ip-link-show-wlan0
But possible that this is not the type pf power savings mode controlled by `iw dev IFACE set power_save`.

Found a way to change that type of mode:
- https://bbs.archlinux.org/viewtopic.php?id=169302
- https://raspberrypi.stackexchange.com/questions/107463/raspberry-pi-3-b-with-buster-cant-be-pinged-wlan0-stuck-in-dormant

=>
```sh
ip l set wlan0 mode default
```
The question is how it went into DORMANT mode in the first place 🤔.

kartikynwa, 2024-07-10:
That does change the mode from DORMANT to DEFAULT in the ip link output. But it still isn't able to find any networks.

Thank you for trying to help. I really appreciate it and I don't want to take up any more of your time. This issue is getting cluttered and sidetracked. I will check the adapter with other devices to make sure it works and then make a post on the forum.

MichaIng, 2024-07-10:
Please do so. And if the issue as really DietPi-specific, please create a new issue. We can test the old kernel then and check for changes in the driver. I'll mark this issue as closed, as OP issue has been solved.

Thanks for testing and feedback @Kreeblah btw 🙂.

closed 2024-07-10 by MichaIng