MichaIng/DietPi #7139: Will more version be supported for nextcloud ?

champymarty, 2024-07-07:
Right now, the latest supported version for nextcloud by dietpi is 25.0.13.
That version is marked as old unsupported version I can cause security risk using it.
The oldest supported major version is 27.

Thank you !

champymarty, 2024-07-07:
Hi ! Sorry abour the template. When I did the questions did not show any template, I would Ihave fill it up for sure.
Yes I am running older debian version. I just did the script https://dietpi.com/blog/?p=811 to update to bulleye

Is there any step to upgrade to bookworm or even newer ?
Also if I do fresh install, what debian version it will be ?
I am running on raspberry 4

- DietPi version:
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=5
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applicable'

- Distro version : bullseye 0
- Kernel version | Linux DietPi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Official raspberry pi power supply for pi 4
- SD card used: SanDisk ultra

Thank you for your help !

Joulinar, 2024-07-07:
Simply follow next step to upgrade to Bookworm https://dietpi.com/blog/?p=3128

champymarty, 2024-07-08:
Thank you for your help !

closed 2024-07-08 by champymarty