MichaIng/DietPi #7140: Update cannot be finished

diendter, 2024-07-07:
RPi 4 Model B (aarch64) | IP: 192.168.158.149                                                                                                       RPi 4 Model B (aarch64) | IP: 192.168.158.149                                                                                                                     ┌──────────────────────────────────────────────────┤ DietPi-Update ├───────────────────────────────────────────────────┐                            │ APT update                                                                                                           │                            │  - Command: apt-get -y -eany update                                                                                  │                            │  - Exit code: 100                                                                                                    │                            │  - DietPi version: v9.5.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 7                                   │                            │  - Error log:                                                                                                        │                            │ Hit:1 https://deb.debian.org/debian bookworm InRelease                                                               │                            │ Hit:2 https://dietpi.com/apt bookworm InRelease                                                                      │                            │ Hit:3 https://deb.debian.org/debian bookworm-updates InRelease                                                       │                            │ Hit:4 https://dietpi.com/apt all InRelease                                                                           │                            │ Hit:5 https://deb.debian.org/debian-security bookworm-security InRelease                                             │                            │ Hit:6 https://archive.raspberrypi.com/debian bookworm InRelease                                                      │                            │ Hit:7 https://deb.debian.org/debian bookworm-backports InRelease                                                     │                            │ Reading package lists...                                                                                             │                            │ E: lzma_read: Read error (10)                                                                                        │                            │ E: The package lists or status file could not be parsed or opened.                                                   │                            │                                                                                                                      │                            │                         Retry          : Re-run the last command that failed                                         │                            │                         DietPi-Config  : Edit network, APT/NTP mirror settings etc                                   │                            │                         Open subshell  : Open a subshell to investigate or solve the issue                           │                            │                         Send report    : Uploads bugreport containing system info to DietPi                          │                            │                                        ●─ Devs only ──────────────────────────────────────●                          │                            │                         Change command : Adjust and rerun the command                                                │                            │                                                                                                                      │                            │                                                                                                                      │                            │                                  <Ok>                                      <Exit>                                    │                            │                                                                                                                      │                            └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘              

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Sun Jul  7 11:36:11 CEST 2024
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://dietpi.com/apt bookworm InRelease
Hit:3 https://deb.debian.org/debian bookworm-updates InRelease
Hit:4 https://dietpi.com/apt all InRelease
Hit:5 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:6 https://archive.raspberrypi.com/debian bookworm InRelease
Hit:7 https://deb.debian.org/debian bookworm-backports InRelease
Reading package lists...
E: lzma_read: Read error (10)
E: The package lists or status file could not be parsed or opened.
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

root@DietPi:~# apt upgrade
Paketlisten werden gelesen… Fehler!
E: lzma_read: Lesefehler (10)
E: Die Paketliste oder die Statusdatei konnte nicht eingelesen oder geöffnet werden.
root@DietPi:~#

Joulinar, 2024-07-07:
Should be similar to this one https://github.com/MichaIng/DietPi/issues/4076#issuecomment-770454020

MichaIng, 2025-06-02:
I'll close this issue. Feel free to reopen it the problem persists.

closed 2025-06-02 by MichaIng