MichaIng/DietPi #7141: Bazarr not working (probably after 9.5.1 upgrade)

sureshk75, 2024-07-07:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.5.1`
- Distro version | `bookworm`
- Kernel version | `Linux MyRaspberryPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | 5V
- SD card used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | (Bazarr)
- Was the software title installed freshly or updated/migrated? It may have been updated recently if there was an update but I didn't pay attention when running APT UPGRADE
- Can this issue be replicated on a fresh installation of DietPi? Don't have an alternate device to test..

- Bug report ID | `eea16101-83e0-4c48-b5d6-aabdaa3094c3`

#### Steps to reproduce

1. Bazarr web service not accessible

#### Expected behaviour

- Bazarr should be accessible via web browser at port 6767 of the RPi4 IP address.

#### Actual behaviour

- A 'Site can't be reached' error is displayed when attempting to access it

#### Extra details

- Running 'systemctl status bazarr.service' results in the following:
```
● bazarr.service - Bazarr (DietPi)
     Loaded: loaded (/etc/systemd/system/bazarr.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-07-08 04:20:33 +08; 53s ago
   Main PID: 270010 (python3)
      Tasks: 2 (limit: 9352)
        CPU: 18.620s
     CGroup: /system.slice/bazarr.service
             └─270010 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr

Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 2335, in _handle_dbapi_exception
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:     raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1960, in _exec_single_context
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:     self.dialect.do_execute(
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/default.py", line 924, in do_execute
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:     cursor.execute(statement, parameters)
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: [SQL: UPDATE system SET updated=?]
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: [parameters: ('0',)]
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: (Background on this error at: https://sqlalche.me/e/20/e3q8)
```

Joulinar, 2024-07-07:
Can you share following 

```
journalctl -u bazarr.service
```

MichaIng, 2024-07-07:
> `sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database`

Please check also permissions of the database directory, and for kernel errors:
```sh
ls -al /mnt/dietpi_userdata/bazarr
dmesg -l 0,1,2,3
```

sureshk75, 2024-07-07:
> journalctl -u bazarr.service
Jul 05 16:56:22 MyRaspberryPi systemd[1]: Started bazarr.service - Bazarr (DietPi).
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: Traceback (most recent call last):
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1960, in _exec_single_context
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     self.dialect.do_execute(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/default.py", line 924, in do_execute
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     cursor.execute(statement, parameters)
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: sqlite3.OperationalError: attempt to write a readonly database
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: The above exception was the direct cause of the following exception:
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: Traceback (most recent call last):
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/main.py", line 58, in <module>
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     database.execute(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/orm/scoping.py", line 778, in execute
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return self._proxied.execute(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/orm/session.py", line 2306, in execute
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return self._execute_internal(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/orm/session.py", line 2191, in _execute_internal
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     result: Result[Any] = compile_state_cls.orm_execute_statement(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/orm/bulk_persistence.py", line 1617, in orm_execute_statement
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return super().orm_execute_statement(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/orm/context.py", line 293, in orm_execute_statement
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     result = conn.execute(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:              ^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1408, in execute
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return meth(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/sql/elements.py", line 513, in _execute_on_connection
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return connection._execute_clauseelement(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1630, in _execute_clauseelement
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     ret = self._execute_context(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:           ^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1839, in _execute_context
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     return self._exec_single_context(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1979, in _exec_single_context
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     self._handle_dbapi_exception(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 2335, in _handle_dbapi_exception
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/base.py", line 1960, in _exec_single_context
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     self.dialect.do_execute(
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/default.py", line 924, in do_execute
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]:     cursor.execute(statement, parameters)
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: [SQL: UPDATE system SET updated=?]
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: [parameters: ('0',)]
Jul 05 16:56:49 MyRaspberryPi Bazarr[692]: (Background on this error at: https://sqlalche.me/e/20/e3q8)
Jul 06 00:53:30 MyRaspberryPi systemd[1]: Stopping bazarr.service - Bazarr (DietPi)...
Jul 06 00:53:30 MyRaspberryPi systemd[1]: bazarr.service: Deactivated successfully.
Jul 06 00:53:30 MyRaspberryPi systemd[1]: Stopped bazarr.service - Bazarr (DietPi).
Jul 06 00:53:30 MyRaspberryPi systemd[1]: bazarr.service: Consumed 16.778s CPU time.
Jul 06 00:54:51 MyRaspberryPi systemd[1]: Started bazarr.service - Bazarr (DietPi).
[... cut ...]
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:   File "/opt/bazarr/bazarr/../libs/sqlalchemy/engine/default.py", line 924, in do_execute
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]:     cursor.execute(statement, parameters)
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) attempt to write a readonly database
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: [SQL: UPDATE system SET updated=?]
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: [parameters: ('0',)]
Jul 08 04:20:56 MyRaspberryPi Bazarr[270025]: (Background on this error at: https://sqlalche.me/e/20/e3q8)

sureshk75, 2024-07-07:
> ls -al /mnt/dietpi_userdata/bazarr
drwxr-xr-x dietpi dietpi 4.0 KB Mon Jun 10 20:11:30 2024  .
drwxrwxr-x dietpi dietpi 4.0 KB Sat Jul  6 00:54:42 2024  ..
drwxrwxr-x dietpi dietpi 4.0 KB Mon Jun 10 20:11:17 2024  .cache
drwxrwxr-x dietpi dietpi 4.0 KB Sun Jun 30 03:00:01 2024  backup
drwxrwxr-x dietpi dietpi 4.0 KB Thu Jul  4 06:32:36 2024  cache
drwxrwxr-x dietpi dietpi 4.0 KB Wed Jun 12 21:24:05 2024  config
drwxrwxr-x dietpi dietpi 4.0 KB Fri Jul  5 16:46:40 2024  db
lrwxrwxrwx dietpi dietpi  15 B  Mon Jun 10 20:11:12 2024  log ⇒ /var/log/bazarr
drwxrwxr-x dietpi dietpi 4.0 KB Mon Jun 10 20:11:30 2024  restore

> dmesg -l 0,1,2,3
[    0.617684] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
[    6.004836] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    6.004846] [vc_sm_connected_init]: failed to initialize shared memory service
[    6.197442] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)

MichaIng, 2024-07-07:
Interesting permissions, definitely not as intended. This should all be owned by `bazarr:root`, not `dietpi:dietpi`. The `775` mode allows `bazarr` to edit files inside the directory, since it is member of the `dietpi` group, but it cannot create new files and directories at least, since the directory has `755` mode. The database files should be however within the `db` dir, and we do not see its mode here.

However, not sure how this happened, but at first, please adjust ownership to what it should be:
```sh
chown -R bazarr:0 /mnt/dietpi_userdata/bazarr /opt/bazarr
systemctl restart bazarr
```

sureshk75, 2024-07-08:
> ```shell
> systemctl restart bazarr
> ```

This seems to fix the problem. I can now access Bazarr.. not sure if the output from journalctl makes sense though but I'll consider this a win for now :P

_journalctl -u bazarr.service_ truncated output:Jul 09 00:14:44 MyRaspberryPi systemd[1]: Stopping bazarr.service - Bazarr (DietPi)...
Jul 09 00:14:44 MyRaspberryPi systemd[1]: bazarr.service: Deactivated successfully.
Jul 09 00:14:44 MyRaspberryPi systemd[1]: Stopped bazarr.service - Bazarr (DietPi).
Jul 09 00:14:44 MyRaspberryPi systemd[1]: bazarr.service: Consumed 22.002s CPU time.
Jul 09 00:14:44 MyRaspberryPi systemd[1]: Started bazarr.service - Bazarr (DietPi).
Jul 09 00:15:04 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:04,299 - root                             (7fb71384c0) :  INFO (server:79) - BAZARR is started and waiting for request on http://0.0.0.0:6767
Jul 09 00:15:04 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:04,308 - root                             (7ef3bff180) :  INFO (signalr_client:122) - BAZARR trying to connect to Sonarr SignalR feed...
Jul 09 00:15:04 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:04,332 - root                             (7ef33ef180) :  INFO (signalr_client:189) - BAZARR trying to connect to Radarr SignalR feed...
Jul 09 00:15:04 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:04,385 - root                             (7ef2bdf180) :  INFO (signalr_client:150) - BAZARR SignalR client for Sonarr is connected and waiting for events.
Jul 09 00:15:04 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:04,522 - root                             (7ef3bff180) :  INFO (signalr_client:217) - BAZARR SignalR client for Radarr is connected and waiting for events.
Jul 09 00:15:08 MyRaspberryPi Bazarr[400303]: 2024-07-09 00:15:08,132 - root                             (7ef13af180) :  ERROR (movies:51) - BAZARR cannot update movie /volume1/data/media/movies/Alice Through the Looking Glass (2016) [imdbid-tt2567026]/Alice Through the Looking Glass (2016) [imdbid-tt2567026] - [Bluray-1080p][AAC 2.0][x264]-YTS.AG.mp4 because of (sqlite3.IntegrityError) FOREIGN KEY constraint failed
Jul 09 00:15:08 MyRaspberryPi Bazarr[400303]: [SQL: UPDATE table_movies SET "alternativeTitles"=?, audio_codec=?, audio_language=?, fanart=?, file_size=?, format=?, "imdbId"=?, monitored=?, movie_file_id=?, overview=?, path=?, poster=?, "radarrId"=?, resolution=?, "sceneName"=?, "sortTitle"=?, tags=?, title=?, "tmdbId"=?, video_codec=?, year=? WHERE table_movies."tmdbId" = ?]

MichaIng, 2024-07-08:
Okay great. I'll close the issue. Not sure how the dir owner changed like this, but assuming it to be a single case. If someone else reports the same, we need to have a deeper look into how it happened.

closed 2024-07-08 by MichaIng