MichaIng/DietPi #7151: USB WiFi (TP-Link Archer T2U PLUS) adapter not working on Odroid N2+ since recent kernel upgrade

kartikynwa, 2024-07-11:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.6, G_DIETPI_VERSION_RC=1, G_GITBRANCH='master', G_GITOWNER='MichaIng'`
- Distro version | `G_DISTRO_NAME=bookwork, G_RASPBIAN=`
- Kernel version | `6.6.37-current-meson64`
- SBC model | `G_HW_MODEL_NAME="Odroid N2 (aarch64)"`
- Power supply used | None. Direct connection used.
- SD card used | SanDisk Extreme

#### Additional Information (if applicable)
- Software title | iwd
- Was the software title installed freshly or updated/migrated? No.
- Can this issue be replicated on a fresh installation of DietPi? I haven't tried yet.

#### Steps to reproduce
1. Have an Odroid N2+ with DietPi 9.5. Use an Archer T2U Plus USB wifi adapter with it.
2. It works out of the box.
3. Upgrade to DietPi 9.6 and upgrade the `linux-{current,stb}-*` packages alongside it.
4. The wifi adapter is not functional anymore.

#### Expected behaviour
It should work.

#### Actual behaviour
I cannot use iwd to connect to a wifi network with the USB adapter. Specifically, the adapter appears in the outputs of commands like `ip link`, `iwctl station list`. But it does not detect any wifi networks. I have tested the adapter with a Windows laptop and can confirm that it still works. This happened after I upgraded `linux-{current,stb}-*` through apt and then rebooted which makes me think this is related to this kernel upgrade.

#### Extra details

Discussion regarding this took place here: https://github.com/MichaIng/DietPi/issues/7138#issuecomment-2221119916

I am seeing this error output in `dmesg`. It is also dumped in the terminal at boot. I cannot make sense of it so I can't confidently say if this is relevant. I have only been accessing the machine over SSH over the past few months. Now I am seeing this error but I am also now accessing the machine directly so I can't say if this was occurring before and it is only now that I am seeing it:

[image]

Output for `ip link`:

```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP mode DORMANT group default qlen 1000         link/ether 98:48:27:9c:e8:4d brd ff:ff:ff:ff:ff:ff
```

Output for `ip a`:

```
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2312 qdisc mq state UP group default qlen 1000                      link/ether 98:48:27:9c:e8:4d brd ff:ff:ff:ff:ff:ff      inet6 fe80::9a48:27ff:fe9c:e84d/64 scope link              valid_lft forever preferred_lft forever
```

Output for `iw dev wlan0 info`:

```
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 98:48:27:9c:e8:4d
        type managed
        wiphy 0
        channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
        txpower 20.00 dBm
```

Following a previous suggestion, I tried `iw dev wlan0 set power_save off ` and `ip l set wlan0 mode default`. Neither helped.

I would like to first try downgrading the aforementioned kernel packages and see if it fixes the issue before trying to test this issue with a fresh install.

MichaIng, 2024-07-11:
Hmm, after running `ip l set wlan0 mode default`, `ip link` still shows `mode DORMANT`? In case, is there any error output?

And does downgrading the kernel help?
```sh
apt install linux-{image,dtb}-current-meson64=24.5.0-trunk-dietpi1
reboot
```

kartikynwa, 2024-07-11:
No. Sorry for not being clear. The state does change from DORMANT to DEFAULT. But the adapter is still unable to find any wifi networks.

I just downgraded the psckages following the commands you suggested and I can resolutely confirm that the adapter has started working again.

You have no idea how happy this makes me.

MichaIng, 2024-07-11:
Okay, then we can at least narrow it down and it works again for you for now. To prevent another upgrade of the kernel, mark then on hold:
```sh
apt-mark hold linux-{image,dtb}-current-meson64
```

MichaIng, 2024-07-11:
I found a commit from 2 days before I triggered our last kernel update, which seems very suspicious: https://github.com/armbian/build/pull/6888
> wifi: rtw88: coex: Prevent doing I/O during Wi-Fi power saving

I triggered new builds with this patch removed. Let's see whether it helps: https://github.com/MichaIng/DietPi/actions/runs/9897823517

MichaIng, 2024-07-11:
Okay, it is ready for testing:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-meson64.deb
dpkg -i linux-{image,dtb}-current-meson64.deb
reboot
```

kartikynwa, 2024-07-12:
Btw let me know if you need any more info. Happy to provide.

MichaIng, 2025-05-31:
Next iteration, as we are now in Linux 6.12, the driver should be really available, right? 😄

kartikynwa, 2025-06-07:
I installed the latest DietPi version available as of today and the wifi adapter works out of the box. Thank you very much.

kartikynwa, 2025-06-16:
It's the same one as before. `TP-Link Archer T2U PLUS [RTL8821AU]` as reported by lsusb.

MichaIng, 2025-06-19:
Oh my bad, different version of that adapter have different chips from different chip manufacturers 😄: https://forums.linuxmint.com/viewtopic.php?t=311501

That directory `rtw88` in Linux source code contains drivers for RTL88xx chip family. New chip drivers were added from Linux version to Linux version. Earlier, Armbian patched drivers for certain chips into the kernel ... ah for that chip they still do 🤔: https://github.com/armbian/build/blob/main/lib/functions/compilation/patch/drivers_network.sh#L171

Makes sense, the RTL8812 are still missing in Linux 6.12. But latest Linux has it, so the patch could be skipped for Linux 6.15.

Hmm, even that the driver is patched into sources, it does not seem to be enabled at compile time. At least my Linux 6.15.2 build does not have it. And it does not have any of the mainline RTL8812 drivers either.

Can you check whether it exists for you in Linux 6.12:
```sh
ls -l /lib/modules/*/kernel/drivers/net/wireless/rtl8812au
modinfo rtl8812au
```
Maybe they disabled the own driver too early, when there was no sane mainline driver yet, and additionally did not enable the additional mainline driver in edge kernel.

Also, please show the driver actually loaded in your case:
```sh
lsmod
```

... ah, you installed the 3rd party driver already. Maybe you find time to uninstall or blacklist that one to use and show whatever the kernel ships.

kartikynwa, 2025-06-22:
```
root@ksodroid:~# ls -l /lib/modules/*/kernel/drivers/net/wireless/rtl8812au
total 4032
-rw-r--r-- 1 root root 4127112 May  8 15:14 88XXau.ko

root@ksodroid:~# modinfo rtl8812au
modinfo: ERROR: Module rtl8812au not found.
```

```
root@ksodroid:~# lsmod | awk '{print $1,$3}' | grep 88
rtw_8821au 0
rtw_8821a 1
rtw_88xxa 1

root@ksodroid:~# lsmod | awk '{print $1,$3}' | grep rt
xt_addrtype 4
rtw_8821au 0
rtw_8821a 1
rtw_88xxa 1
rtw_usb 1
rtw_core 3
rtc_pcf8563 0
rtc_meson_vrtc 1
```

FYI I have blacklisted the 88XX module:

```
root@ksodroid:~# cat /etc/modprobe.d/disable_rtl88XXau.conf
blacklist 88XXau
```

MichaIng, 2025-06-22:
In my case, on edge kernel Linux 6.15.3, `rtw88_8821cu` is sed, since `88XXau` is not installed anymore, and there is no `rtw88_8821a*` module. In my case it works, but AFAIK it is not a 100% matching driver.

... and on Linux 6.12 it is the same for me. Yeah, this module is not enabled for Rockchip, but for Amlogic:
- https://github.com/armbian/build/blob/main/config/kernel/linux-rockchip64-current.config
- https://github.com/armbian/build/blob/main/config/kernel/linux-meson64-current.config#L1126

No idea why. Testing on my Odroid N2+ now. Okay, there is another Realtek driver additionally enabled: `8821cu` loaded in my case. Yeah now I remember my chip is cu not au. So for me it does not make a difference, but both, mainline driver and this patched driver work reliably.

However, I have hence overlapping drivers on Amlogic, but a missing RTL88xxau driver on Rockchip. Found another patch: https://github.com/armbian/build/blob/main/patch/misc/rtw88/6.12/001-drivers-net-realtek-rtw88-upstream-wireless.patch

So that one adds the AU modules to the mainline driver ... On Amlogic, the 3rd party driver is enabled, on Rockchip the (patched) mainline driver. Pretty confusing inconsistency.

Since the 3rd party driver is causing issues for you, and I generally prefer mainline drivers, and the fact that with upcoming LTS (or edge), mainline supports all those chips natively, I will align this, for all SoCs to use all related mainline drivers. So you can test whether this solves the problem for you as well without adding a 3rd party driver yourself.

MichaIng, 2025-06-22:
Try this one. Works well here, now with mainline driver used for my RTL8812CU/RTL8821CU whatever:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-meson64.deb
sudo dpkg -i linux-{image,dtb}-current-meson64.deb
sudo reboot
```

kartikynwa, 2025-06-23:
Do I still need to do 👆 or should just doing a dietpi-update suffice? Asking since I see a new release has been tagged.

MichaIng, 2025-06-23:
Yeah right, the packages are now in the APT repository, hence just update DietPi, or `apt upgrade`.

closed 2025-07-28 by MichaIng