MichaIng/DietPi #7152: Dietpi-update fails to complete due to apt-key cannot create /dev/null

psyburr, 2024-07-11:
### Creating a bug report/issue
- [*] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 1`
- Kernel version | 6.1.21-v7+
- SBC model | RPi 3B+
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra 32 GB)

- Bug report ID | d8fd3fe5-bbd1-4970-a5ce-7e520d61e20a

#### Steps to reproduce
`sudo dietpi-update`

#### Expected behaviour
Successful update/patching from  v9.2.1 to v9.6.1

#### Actual behaviour
`/usr/bin/apt-key: 95: cannot create /dev/null:
Permission denied
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied                              /usr/bin/apt-key: 95: cannot create /dev/null:
Permission denied
E: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed                Sub-process apt-key returned an error code (29)Err:1 https://raspbian.mirror.constant.com/raspbian bookworm InRelease
…`

#### Extra details
After the update fails, my shell prompt includes three lines of 
`-bash: /dev/null: Permission denied`
with the only remedy being to restore using dietpi-backup.

I will admit I have quite a bit of additional software installed, including the starship shell prompt, that may be giving me this error. I have found that a manual `sudo apt-get update/upgrade` yields no issues, and other software installed via `dietpi-softwate` such as pihole has no issues updating.

MichaIng, 2024-07-11:
Hmm, I faced a very similar issue some weeks ago. The `/dev/null` was somehow replaced by an actual file, which of course breaks everything which tries to redirect output to it. Can you check this:
```sh
ls -l /dev/null
```

psyburr, 2024-07-11:
I restored to my most recent backup and this was the result pre dietpi-update 
`crw-rw-rw- 1 root root 1, 3 Jul 11 19:35 /dev/null`

After the update
`crw-r--r-- 1 root root 1, 3 Jul  7 22:50 /dev/null`

MichaIng, 2024-07-11:
Can you share the full update log:
```sh
cat /var/tmp/dietpi/logs/dietpi-update.log
```

psyburr, 2024-07-11:
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 https://raspbian.mirror.constant.com/raspbian bookworm InRelease [15.0 kB]
Get:2 https://dietpi.com/apt bookworm InRelease [3525 B]
Get:3 https://dietpi.com/apt all InRelease [25.3 kB]
Get:4 https://raspbian.mirror.constant.com/raspbian bookworm/main armhf Packages [14.5 MB]
Get:5 https://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:6 https://dietpi.com/apt bookworm/main armhf Packages [2032 B]
Get:7 https://archive.raspberrypi.com/debian bookworm/main armhf Packages [419 kB]
Get:8 https://archive.raspberrypi.com/debian bookworm/main armhf Contents (deb) [3396 kB]
Fetched 18.4 MB in 13s (1460 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages will be upgraded:
  vaultwarden
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 11.5 MB of archives.
After this operation, 557 kB of additional disk space will be used.
Get:1 https://dietpi.com/apt bookworm/main armhf vaultwarden armhf 1.31.0-dietpi1 [11.5 MB]
Fetched 11.5 MB in 2s (5829 kB/s)
(Reading database ... 96414 files and directories currently installed.)
Preparing to unpack .../vaultwarden_1.31.0-dietpi1_armhf.deb ...
Unpacking vaultwarden (1.31.0-dietpi1) over (1.30.5-dietpi3) ...
Setting up vaultwarden (1.31.0-dietpi1) ...

Configuration file '/mnt/dietpi_userdata/vaultwarden/vaultwarden.env'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
 ==> Keeping old config file as default.
Configuring vaultwarden service user ...
usermod: no changes
Setting vaultwarden userdata owner ...
Configuring vaultwarden systemd service ...
[  OK  ] DietPi-Update | APT upgrade

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

[  OK  ] DietPi-Update | Downloading update archive
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Removing unused files
[  OK  ] DietPi-Update | Hardening update archive mode
[  OK  ] DietPi-Update | Installing new DietPi scripts
[  OK  ] DietPi-Update | Installing new DietPi system files
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] verify_dietpi.txt  | Completed
[  OK  ] DietPi-Update | sync
[  OK  ] DietPi-Update | systemctl daemon-reload

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches
[... 26 lines cut ...]
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied
E: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Sub-process apt-key returned an error code (29)Err:3 https://dietpi.com/apt all InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied
/usr/bin/apt-key: 95: cannot create /dev/null: Permission denied
E: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Sub-process apt-key returned an error code (29)Err:4 https://archive.raspberrypi.com/debian bookworm InRelease
  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://dietpi.com/apt bookworm InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://raspbian.mirror.constant.com/raspbian bookworm InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://dietpi.com/apt all InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://archive.raspberrypi.com/debian bookworm InRelease: gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: Failed to fetch https://raspbian.mirror.constant.com/raspbian/dists/bookworm/InRelease  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: Failed to fetch https://dietpi.com/apt/dists/bookworm/InRelease  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
E: Failed to fetch https://dietpi.com/apt/dists/all/InRelease  gpgv, gpgv2 or gpgv1 required for verification, but neither seems installed
[... cut ...] ired for verification, but neither seems installed
E: Some index files failed to download. They have been ignored, or old ones used instead.
[FAILED] DietPi-Patch | APT update
 - Command: apt-get -y -eany update
[FAILED] DietPi-Patch | Unable to continue, DietPi-Patch will now terminate.
[FAILED] DietPi-Update | An error occurred during incremental patching. Please check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors, and rerun "dietpi-update" after the cause has been solved.

MichaIng, 2024-07-12:
Just to be sure, after restoring the backup, you ran `dietpi-update` just a single time and did not repeat it? Because I need to find the exact pre-patch, package upgrade or patch which caused this, but am unable to find the culprit so far.

We have a 2nd report: #7157
Weirdly, in this case, `/dev/null` was indeed replaced with a regular file, like it happened some time ago on my system as well, while in your case, it is still a node, just with wrong permissions.

This will re-create the node with the correct permissions:
```sh
sudo mknod -m 0666 /dev/null c 1 3
```
Since it is a correct node in your case already:
```sh
sudo chmod 0666 /dev/null
```

MichaIng, 2024-07-12:
> I restored to my most recent backup and this was the result pre dietpi-update 
> `crw-rw-rw- 1 root root 1, 3 Jul 11 19:35 /dev/null`
> 
> After the update
> `crw-r--r-- 1 root root 1, 3 Jul  7 22:50 /dev/null`

What I want to verify:
- The first output was right before the `dietpi-update` you posted the logs from, without any other step done in between?
- And the 2nd was right after the failure?

Because interesting is that the timestamp is now older: `Jul 7` vs `Jul 11`. This would mean the `/dev/null` was not removed and re-created during the update, like because of a missing `>` redirect character when some file is removed, like
```sh
rm /path/to/file /dev/null
```
Such mistake would however result in a later/current timestamp. Means that e.g. a `null` file was accidentally contained in an archive and restored from there, preserving its timestamp, or just a file `mv`ed to `/dev/null`.

psyburr, 2024-07-13:
yes, that is correct. i had restored to a working backup. i obtained the output of ls on /dev/null immediately after rebooting. i then attempted dietpi-update again to obtain the second output of ls.

[  OK  ] DietPi-Services | nmbd                 active (running) since Fri 2024-07-12 11:51:01 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | smbd                 active (running) since Fri 2024-07-12 11:51:03 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | php8.2-fpm           active (running) since Fri 2024-07-12 11:51:02 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | apache2              active (running) since Sat 2024-07-13 03:15:45 UTC; 19h ago
[  OK  ] DietPi-Services | vaultwarden          active (running) since Fri 2024-07-12 11:50:58 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | cron                 active (running) since Fri 2024-07-12 11:50:56 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | ssh                  active (running) since Fri 2024-07-12 11:50:59 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | pihole-FTL           active (running) since Fri 2024-07-12 11:50:58 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | unbound              active (running) since Fri 2024-07-12 11:51:00 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | fail2ban             active (running) since Fri 2024-07-12 11:50:58 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Fri 2024-07-12 11:50:57 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Fri 2024-07-12 11:50:57 UTC; 1 day 11h ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Fri 2024-07-12 11:50:57 UTC; 1 day 11h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive x⸑x

_this all occurred while at work and unable to respond as i don't have regular access to any computer during those hours, thus it was done using my mobile and a detached tmux session so i may let things run in the background while i will more than likely lose my ssh connection. i appreciate your quick replies and help in the matter so far!_

MichaIng, 2024-07-16:
Let's go through some possible culprits:
```sh
ls -l /dev/null # check that it is still has right permissions
apt update
ls -l /dev/null
apt upgrade
ls -l /dev/null
dietpi-services restart
ls -l /dev/null
```

MichaIng, 2025-06-02:
I ran into this issue when trying to disable the obsolete MariaDB PID file with `pid_file=/dev/null` in another config file: https://github.com/MichaIng/hacks/commit/7c453c9

This worked in the past, but recent MariaDB then replaces the node with an actual file. Problem is there is no real way to unset a setting, also empty string value does not work but causes failure. And I try to avoid editing config files shipped by APT packages. However, this is my personal configs, was never shipped with a DietPi system.

I'll close this issue. Feel free to reopen it the problem persists.

closed 2025-06-02 by MichaIng