MichaIng/DietPi #7154: Radxa Zero 3W - Format eMMC fails Report 62215187-eddb-4247-9feb-69f6d693802e

Clumpton, 2024-07-11:
### Creating a bug report/issue
- [ YES] I have searched the existing open and closed issues

#### Required Information
#### Details:
- Date           | 2024-07-11 20:36:45
- Bug report     | 62215187-eddb-4247-9feb-69f6d693802e
- Program name   | DietPi-Drive_Manager
- Command        | `dd if=/dev/zero of=/dev/mmcblk0boot0 bs=4K count=10`
- Exit code      | 1
- DietPi version | v9.6.1 (MichaIng/master)
- Distro version | bookworm (ID=7)
- Kernel version | `Linux RadxaDietPi 5.10.160-legacy-rk35xx #1 SMP Wed May 15 03:04:45 UTC 2024 aarch64 GNU/Linux`
- Architecture   | `arm64`
- Hardware model | Radxa ZERO 3 (aarch64) (ID=90)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)

#### Steps to reproduce:

1. Run dietpi-Drive Manager as root
2. Request format of on-board eMMc
#### Expected behaviour:

- Drive should be formatted
#### Actual behaviour:

- Operation not permitted
#### Extra details:

- This is the latest Radxa Zero 3 W, has wifi but no Ethernet
#### Additional logs:
```
dd: error writing '/dev/mmcblk0boot0': Operation not permitted

MichaIng, 2024-07-11:
Hmm, `/dev/mmcblk0boot0` AFAIK is not the actual eMMC, those are some additional storages which appear with some SBCs and kernel versions which I also didn't fully understand. Probably uncommon SPI devices.

Can you try to switch to from partition format mode to drive format mode? Then `dev/mmcblk0` as a whole should be formatted, with a new partition table. I'll try to exclude those `*bootN` devices when formatting in partition mode.

Clumpton, 2024-07-12:
Hi,
Thanks for the quick reply.
Ran the command on dev/mmcblk0, first time it gave an error on 
dev/mmcblk0boot1 (not boot0), second try worked fine and created 
/dev/mmcblk0p1.
:thumbsup:

62215187-eddb-4247-9feb-69f6d693802e (Issue #7154)

MichaIng, 2024-10-19:
This was solved in `dietpi-drive_manager` with last release.

closed 2024-10-19 by MichaIng