MichaIng/DietPi #7155: Koel not working anymore after updating to version 7.0.2

bartolus39, 2024-07-12:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.6.1
- Distro version | bookworm 0
- Kernel version | 6.1.21-v8+
- SBC model | Rpi 4 Model B
- Power supply used | 5.1V 3A Official
- SD card used | 2.5" SSD via USB

#### Additional Information (if applicable)
- Software title | Koel
- It was updated
- Can this issue be replicated on a fresh installation of DietPi? -> do not know.

#### Steps to reproduce

1. Update/install koel
2. visit the webpage at x.x.x.x:8003

#### Actual behaviour
The webpage is saying "Untrusted Host"
[image]

further down:
[image]

#### Extra details
- update was done from version 6.12.1 to 7.0.2.

MichaIng, 2024-07-12:
Does clearing the browser cache help, e.g. with a forced page reload `CTRL+C`? And did/do you have HTTPS enabled for Koel? Trusted hosts seem to be set/checked only when HTTPS is enforced: https://github.com/koel/koel/blob/master/app/Http/Middleware/ForceHttps.php

MichaIng, 2024-07-12:
I can replicate it. It works on a fresh install, but upgrading from v6 to v7 leads to this error. Erasing all browser data does not help, it must be some database migration issue.

MichaIng, 2024-07-12:
Okay, not sure why it did work well on my very first attempt. Now, I cannot get it working anymore with fresh installs on any system or with any other browser.

But I found out one way to get it working: In the config file `/mnt/dietpi_userdata/koel/.env`, change the host in `APP_URL` from `localhost` to the IP or hostname you use to access Koel. The port number does not matter, not even the protocol does, but the host only. That one seems to be added to the list of trusted hosts now. The question is how other hosts can be added to this list.

bartolus39, 2024-07-13:
Thanks a lot for your efforts!!
I am seeing now same behavior. With changing the APP_URL it loads the page, but no UI elements.
I put it also into an apache webserver. This gives the same problem.
However, putting it into a "real" webserver environment (with access from outside home with DDNS, with https...) it loads the UI elements. Maybe the access via https makes the difference?

MichaIng, 2024-07-13:
Yes, I guess that there is something messed up with these trusted host checks. Since there is no (documented) way to append to this list it must not be checked, but it is. Likely, internal URLs/requests generated by the backend to load further assets, are also affected by this, and hence fail for the same reason, unless one indirectly adds another entry by enforcing HTTPS. The script I posted above then adds the current client request host to the list of trusted hosts explicitly, and I guess also Koel itself then generates URLs internally differently.

There was a PR to change the behaviour around this: https://github.com/koel/koel/pull/1706
But as can be seen, it was reverted later. Looks like it was again implemented, probably a different way, but still breaks things. I know this in general from ownCloud and Nextcloud: In their configs, you can define an array with all trusted hosts (including a wildcard `*` to disable the feature), and a CLI URL, to be used by requests from the backend itself. Koel however does not have any of these, at least not documented, hence this cannot go well. Maybe Laravel itself actually does support it.

We need to report this to the Koel: https://github.com/koel/koel/issues
Let me do this, since I can list the steps how we install Koel (still all right, according to docs), and hence an easy way to replicate.

MichaIng, 2024-07-16:
There have been several updates, now v7.0.6, but the issue persists.
Reported here: https://github.com/koel/koel/issues/1796

bartolus39, 2024-07-16:
I tried out the v7.0.6 now, and here it is now working as expected. So all UI elements are visible and seem to work. What is the difference to your setup?

MichaIng, 2024-07-16:
We found the reason and a nice way to make the trusted hosts feature transparently available, but allow all hosts by default (pre-v7.0.0 behaviour): https://github.com/koel/koel/pull/1797

MichaIng, 2024-08-26:
This has been released with [Koel v7.0.10](https://github.com/koel/koel/releases/tag/v7.0.10)

Update via reinstall:
```sh
dietpi-software reinstall 143
```

closed 2024-08-26 by MichaIng