MichaIng/DietPi #7156: Error while updating diet Pi version from 9.4.2 to 9.6.1

adityaa2b, 2024-07-12:
#### Details:
- Date           | Fri Jul 12 14:36:52 BST 2024
- DietPi version | v9.4.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
E: Could not get lock /var/lib/apt/lists/lock. It is held by process 361859 (apt-get)
E: Unable to lock directory /var/lib/apt/lists/
```

Joulinar, 2024-07-12:
Did you start package update twice? Because it seems already running. At least a Debian package update.

adityaa2b, 2024-07-12:
No..while running 1st time itself it gave me the  error..I got option to retry the command but still same.. Even after closing the session and relogin it gave same error

Joulinar, 2024-07-12:
Can you share following 

`ps -ef | grep 361859`

adityaa2b, 2024-07-13:
root      361859  361858  0 Jul06 ?        00:02:25 /usr/bin/apt-get update
_apt      361862  361859  0 Jul06 ?        00:00:00 /usr/lib/apt/methods/https
_apt      361863  361859  0 Jul06 ?        00:00:00 /usr/lib/apt/methods/https
_apt      361864  361859  0 Jul06 ?        00:00:00 /usr/lib/apt/methods/https
root      766098  765921  0 06:55 pts/0    00:00:00 grep 361859

Joulinar, 2024-07-13:
Ok seems something hass Ben started in July 6th. Let's check 

`pstree -s -a 361859`

adityaa2b, 2024-07-13:
systemd
  └─cron -f
      └─sh -c /usr/bin/apt-get update > /dev/null 2>&1
          └─apt-get update
              ├─https
              ├─https
              └─https

Joulinar, 2024-07-13:
Ok it has been started by cron. Maybe the nightly package check was failing. We would need the process id.

`pstree -s -p -a -G -l 361859`

MichaIng, 2024-07-13:
Could be also killed via `htop`, or @Joulinar do you want to find out something else?

Looks like APT hung up completely, while trying to download package lists 🤔.

Joulinar, 2024-07-13:
Nope just trying to find the process to kill

MichaIng, 2024-07-13:
In `htop`, one can scroll the process list (up/down buttons), select the hanging process, then hit F9 and RETURN/ENTER to kill the selected process with SIGTERM. If it does not disappear, one can retry with SIGKILL: After pressing F9, a list appears on the right side, which can be again navigated with up/down buttons, to select "9 SIGKILL", then RETURN/ENTER. But of course one must be careful to not accidentally select the wrong process, and keep it statically sorted by name or PID, instead of CPU usage, so that they do not swap places all the time.

With `pstree` and `kill` command, it is of course a little safer. Another attempt is to kill it by process name:
```sh
killall apt-get
```
or if it is still there:
```sh
killall -9 apt-get
```
after which the lock file however needs to be removed manually:
```sh
rm /var/lib/apt/lists/lock
```
And it is rare but possible that processes hang on kernel level, which makes it impossible to really kill them without a reboot.

adityaa2b, 2024-08-02:
"killall -9 apt-get" this command solved the issue. Thanks for your help.

closed 2024-08-02 by adityaa2b