MichaIng/DietPi #7157: unable to update dietpi to latest version

hcr2020, 2024-07-12:
#### Details:
- Date           | Fri Jul 12 19:40:13 BST 2024
- Bug report     | 6dbbd6b4-823d-4de3-b099-33ee20fb991d
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | `Linux DietPi 6.1.63-current-rockchip64 #1 SMP PREEMPT Mon Nov 20 10:52:19 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up exim4-config (4.94.2-7+deb11u3) ...
/usr/sbin/update-exim4.conf: 467: cannot create /dev/null: File exists
Invalid new configfile /var/lib/exim4/config.autogenerated.tmp, not installing 
/var/lib/exim4/config.autogenerated.tmp to /var/lib/exim4/config.autogenerated
dpkg: error processing package exim4-config (--configure):
 installed exim4-config package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of exim4-base:
 exim4-base depends on exim4-config (>= 4.94) | exim4-config-2; however:
  Package exim4-config is not configured yet.
  Package exim4-config-2 is not installed.
  Package exim4-config which provides exim4-config-2 is not configured yet.

dpkg: error processing package exim4-base (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of exim4-daemon-light:
 exim4-daemon-light depends on exim4-base (>= 4.94.2); however:
  Package exim4-base is not configured yet.

dpkg: error processing package exim4-daemon-light (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 exim4-config
 exim4-base
 exim4-daemon-light
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I am unable to update dietpi to latest version, due to exim4-config; exim4-base; exim-daemon-light upgrade error and 
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied

Thank you in advance

Joulinar, 2024-07-12:
Probably similar to https://github.com/MichaIng/DietPi/issues/7152

hcr2020, 2024-07-12:
here is the output of:
ls -l /dev/null

-rw-r--r-- 1 root root 0 Jul 12 21:06 /dev/null

MichaIng, 2024-07-12:
Can you share the full update log:
```sh
cat /var/tmp/dietpi/logs/dietpi-update.log
```

hcr2020, 2024-07-12:
root@DietPi:~# cat /var/tmp/dietpi/logs/dietpi-update.log

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[      ] DietPi-Update | Downloading pre-patches/boot/dietpi/func/dietpi-globals                                                                                                                     : line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Update | Downloading pre-patches
[      ] DietPi-Update | Applying execute permission/boot/dietpi/func/dietpi-glo                                                                                                                     bals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://packages.azlux.fr/debian bookworm InRelease
Hit:2 http://ftp.de.debian.org/debian bullseye InRelease
Hit:3 http://ftp.de.debian.org/debian bullseye-updates InRelease
Hit:4 http://ftp.de.debian.org/debian bullseye-backports InRelease
Hit:5 https://download.mono-project.com/repo/debian buster InRelease
Hit:6 https://apt.sonarr.tv/debian buster InRelease
Hit:7 https://download.docker.com/linux/debian bullseye InRelease
Hit:8 https://dietpi.com/apt bullseye InRelease
Hit:9 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:10 https://repo.nextdns.io/deb stable InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up exim4-config (4.94.2-7+deb11u3) ...
/usr/sbin/update-exim4.conf: 467: cannot create /dev/null: File exists
Invalid new configfile /var/lib/exim4/config.autogenerated.tmp, not installing
/var/lib/exim4/config.autogenerated.tmp to /var/lib/exim4/config.autogenerated
dpkg: error processing package exim4-config (--configure):
 installed exim4-config package post-installation script subprocess returned err                                                                                                                     or exit status 1
dpkg: dependency problems prevent configuration of exim4-base:
 exim4-base depends on exim4-config (>= 4.94) | exim4-config-2; however:
  Package exim4-config is not configured yet.
  Package exim4-config-2 is not installed.
  Package exim4-config which provides exim4-config-2 is not configured yet.

dpkg: error processing package exim4-base (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of exim4-daemon-light:
 exim4-daemon-light depends on exim4-base (>= 4.94.2); however:
  Package exim4-base is not configured yet.

dpkg: error processing package exim4-daemon-light (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 exim4-config
 exim4-base
 exim4-daemon-light
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
[... 36 lines cut ...]
/var/lib/exim4/config.autogenerated.tmp to /var/lib/exim4/config.autogenerated
dpkg: error processing package exim4-config (--configure):
 installed exim4-config package post-installation script subprocess returned err                                                                                                                     or exit status 1
dpkg: dependency problems prevent configuration of exim4-base:
 exim4-base depends on exim4-config (>= 4.94) | exim4-config-2; however:
  Package exim4-config is not configured yet.
  Package exim4-config-2 is not installed.
  Package exim4-config which provides exim4-config-2 is not configured yet.

dpkg: error processing package exim4-base (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of exim4-daemon-light:
 exim4-daemon-light depends on exim4-base (>= 4.94.2); however:
  Package exim4-base is not configured yet.

dpkg: error processing package exim4-daemon-light (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 exim4-config
 exim4-base
 exim4-daemon-light
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
---------------------------------------------------------------------
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2024-07-12:
Do you have a backup you can restore from before the update? I would like to see really all pre-patches and package upgrades applied, to be able to find the cultprit.

hcr2020, 2024-07-12:
I run the last two commands and dietpi-update , but it failed
this is the new id 6dbbd6b4-823d-4de3-b099-33ee20fb991d

this is   cat /var/tmp/dietpi/logs/dietpi-update.log

root@DietPi:~# cat /var/tmp/dietpi/logs/dietpi-update.log

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[      ] DietPi-Update | Downloading pre-patches/boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Update | Downloading pre-patches
[      ] DietPi-Update | Applying execute permission/boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://dietpi.com/apt bullseye InRelease
Hit:2 http://packages.azlux.fr/debian bookworm InRelease
Hit:3 http://ftp.de.debian.org/debian bullseye InRelease
Hit:4 https://download.mono-project.com/repo/debian buster InRelease
Hit:5 https://deb.debian.org/debian-security bullseye-security InRelease
Get:6 http://ftp.de.debian.org/debian bullseye-updates InRelease [44.1 kB]
Hit:7 https://apt.sonarr.tv/debian buster InRelease
Hit:8 https://download.docker.com/linux/debian bullseye InRelease
gpgv: Signature made Tue Jul  9 19:43:33 2024 BST
gpgv:                using RSA key FFB3546A0FEE5516E659B6E31DF45FDB9C6CE851
gpgv: Good signature from "MichaIng <micha@dietpi.com>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
Hit:9 https://repo.nextdns.io/deb stable InRelease
Get:10 http://ftp.de.debian.org/debian bullseye-backports InRelease [49.0 kB]
gpgv: Signature made Tue Jul  9 07:49:09 2024 BST
gpgv:                using RSA key 98B824A5FA7D3A10FDB225B7CA548A0A0312D8E6
gpgv: Good signature from "Azlux <github@azlux.fr>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
gpgv: Signature made Sat Jun 29 11:27:34 2024 BST
gpgv:                using RSA key A7236886F3CCCAAD148A27F80E98404D386FA1D9
gpgv: Good signature from "Debian Archive Automatic Signing Key (11/bullseye) <ftpmaster@debian.org>"
gpgv: Signature made Sat Jun 29 11:27:34 2024 BST
gpgv:                using RSA key 4CB50190207B4758A3F73A796ED0E7B82643E131
gpgv: Good signature from "Debian Archive Automatic Signing Key (12/bookworm) <ftpmaster@debian.org>"
gpgv: Signature made Sat Jun 29 11:31:04 2024 BST
gpgv:                using RSA key A4285295FC7B1A81600062A9605C66F00D6C9793
gpgv: Good signature from "Debian Stable Release Key (11/bullseye) <debian-release@lists.debian.org>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
gpgv: Signature made Wed Jul 12 16:18:10 2023 BST
gpgv:                using RSA key 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
gpgv: Good signature from "Xamarin Public Jenkins (auto-signing) <releng@xamarin.com>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
gpgv: Signature made Fri Jul 12 17:19:40 2024 BST
gpgv:                using RSA key 5237CEEEF212F3D51C74ABE0112695A0E562B32A
gpgv: Good signature from "Debian Security Archive Automatic Signing Key (10/buster) <ftpmaster@debian.org>"
gpgv: Signature made Fri Jul 12 17:19:41 2024 BST
gpgv:                using RSA key ED541312A33F1128F10B1C6C54404762BBB6E853
gpgv: Good signature from "Debian Security Archive Automatic Signing Key (11/bullseye) <ftpmaster@debian.org>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
gpgv: Signature made Mon Mar 13 07:13:33 2023 GMT
gpgv:                using RSA key EBFF6B99D9B78493
gpgv: Can't check signature: No public key
gpgv: Signature made Mon Mar 13 07:13:33 2023 GMT
gpgv:                using RSA key 1F4F90DE2A9B4BF8
gpgv: Good signature from "Sonarr <hello@sonarr.tv>"
/usr/bin/apt-key: 57: cannot create /dev/null: Permission denied
Sub-process apt-key returned an error code (2)gpgv: Signature made Fri Jul 12 10:28:03 2024 BST
gpgv:                using RSA key 7EA0A9C3F273FCD8
[... 53 lines cut ...]
 exim4-base
 exim4-daemon-light
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[      ] DietPi-BugReport | Packing upload archive/boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-BugReport | Packing upload archive
[      ] DietPi-BugReport | Sending bug report/boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 6dbbd6b4-823d-4de3-b099-33ee20fb991d

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...root@DietPi:~#

I have backups, but how to  do?

hcr2020, 2024-07-12:
dietpi-backup 
list

 Please select the backup from the archive you wish to restore:                                                       
                                      │                                                                                                                                                   │
                                      │                                        1 2024-05-13 19:27:01.000000000 +0100                                         │
                                      │                                        2 2024-06-04 11:13:22.215547183 +0100                                         │
                                      │                                        3 2024-05-13 19:27:01.000000000 +0100                                         │
                                      │                                        4 2023-08-21 20:47:29.073313629 +0100                                         │

What is the one that you suggest I restore?

MichaIng, 2024-07-12:
Okay your last output shows the broken `/dev/null` before `dietpi-update` even started to do anything. Now I also see the capped lines in your first paste, which contains the same error message already from `dietpi-globals`, when it tried to download the the pre-patches. The `dietpi-update` script has not changed since v9.0, so it is unlikely to be the culprit.

Can you try the commands again and check the node right afterwards?
```sh
sudo rm /dev/null
sudo mknod -m 0666 /dev/null c 1 3
ls -l /dev/null
```
Then wait for a while, check again, and run only the update check (without the actual update), before checking agian:
```sh
sleep 10
ls -l /dev/null
dietpi-update 2
ls -l /dev/null
```
So far to assure `dietpi-update` until this part does not cause the issue.

hcr2020, 2024-07-12:
The first commands
root@DietPi:~# sudo rm /dev/null    
root@DietPi:~# sudo mknod -m 0666 /dev/null c 1 3root@DietPi:~# ls -l /dev/null                   crw-rw-rw- 1 root root 1, 3 Jul 12 22:46 /dev/null
root@DietPi:~#

I waited 1 minute
And here it is

root@DietPi:~# sleep 10                          root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 12 22:46 /dev/null
root@DietPi:~# dietpi-update 2
                                                  DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity                                       [  OK  ] DietPi-Update | Checking DNS resolver   [ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.6.1
[  OK  ] DietPi-Update | Update available:       [ INFO ] DietPi-Update | Current version : v9.5.1[ INFO ] DietPi-Update | Latest version  : v9.6.1root@DietPi:~# ls -l /dev/null                   crw-rw-rw- 1 root root 1, 3 Jul 12 22:46 /dev/null
root@DietPi:~#

MichaIng, 2024-07-12:
Okay so far so good, probably it is one of the stopped services:
```sh
ls -l /dev/null
dietpi-services stop
ls -l /dev/null
```

hcr2020, 2024-07-12:
root@DietPi:~# ls -l /dev/null                   crw-rw-rw- 1 root root 1, 3 Jul 12 22:46 /dev/null
root@DietPi:~# dietpi-services stop                                                                DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop 

[  OK  ] DietPi-Services | stop : cron           [  OK  ] DietPi-Services | stop : docker         [  OK  ] DietPi-Services | stop : radarr         [  OK  ] DietPi-Services | stop : sonarr         [  OK  ] DietPi-Services | stop : medusa         [  OK  ] DietPi-Services | stop : qbittorrent    [  OK  ] DietPi-Services | stop : minidlna       [  OK  ] DietPi-Services | stop : nginx          [  OK  ] DietPi-Services | stop : smbd           [  OK  ] DietPi-Services | stop : nmbd           [  OK  ] DietPi-Services | stop : avahi-daemon   root@DietPi:~# ls -l /dev/null                   crw-rw-rw- 1 root root 1, 3 Jul 12 22:46 /dev/null
root@DietPi:~#

hcr2020, 2024-07-12:
Hi 
I launched dietpi-update and it passed @MichaIng 
🥇

hcr2020, 2024-07-12:
Here is after reboot

Linux DietPi 6.6.34-current-rockchip64 #2 SMP PREEMPT Sun Jun 16 
11:47:49 UTC 2024 aarch64

The programs included with the Debian GNU/Linux system are free s
oftware;
the exact distribution terms for each program are described in th
e
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jul 12 23:10:41 2024 from 192.168.1.200
 ─────────────────────────────────────────────────────
 DietPi v9.6.1 : 23:11 - Fri 07/12/24
 ─────────────────────────────────────────────────────
 - Device model : ROCK64 (aarch64)
 - CPU temp : WARNING: 75 °C / 167 °F : Reducing the life of your device
 - LAN IP : 192.168.1.159 (eth0)
 - MOTD : DietPi v9.6 is here. Check out all changes:
          https://dietpi.com/docs/releases/v9_6/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# cat /var/tmp/dietpi/logs/dietpi-update.log
                                                                                        DietPi-Update                                                                         ─────────────────────────────────────────────────────                                   Phase: Applying pre-patches                                                           [  OK  ] DietPi-Update | Applying execute permission                                   [  OK  ] DietPi-Update | Successfully applied pre-patches                                                                                                                      DietPi-Update                                                                         ─────────────────────────────────────────────────────                                   Phase: Upgrading APT packages                                                         
[ INFO ] DietPi-Update | APT update, please wait...                                    Hit:1 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:2 https://dietpi.com/apt bullseye InRelease
Hit:3 https://repo.nextdns.io/deb stable InRelease
Hit:4 http://packages.azlux.fr/debian bookworm InRelease
Hit:5 https://download.docker.com/linux/debian bullseye InRelease
Hit:6 https://apt.sonarr.tv/debian buster InRelease
Hit:7 https://download.mono-project.com/repo/debian buster InRelease
Hit:8 http://ftp.de.debian.org/debian bullseye InRelease
Hit:9 http://ftp.de.debian.org/debian bullseye-updates InRelease
Hit:10 http://ftp.de.debian.org/debian bullseye-backports InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update                                                    [ INFO ] DietPi-Update | APT upgrade, please wait...                                   Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
3 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up exim4-config (4.94.2-7+deb11u3) ...
Setting up exim4-base (4.94.2-7+deb11u3) ...
exim4-base.service is a disabled or a static unit not running, not starting it.
Setting up exim4-daemon-light (4.94.2-7+deb11u3) ...
[  OK  ] DietPi-Update | APT upgrade                                                                                                                                           DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt                          [  OK  ] verify_dietpi.txt  | Completed                                                [  OK  ] DietPi-Update | systemctl daemon-reload                                                                                                                               DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v9.5.1                                      [  OK  ] DietPi-Set_software | eval echo 'deb https://dietpi.com/apt all rock64' >> /etc/apt/sources.list.d/dietpi.list                                                       [  OK  ] apt-mirror dietpi | Completed                                                 [ INFO ] DietPi-Patch | APT update, please wait...                                     Hit:1 http://packages.azlux.fr/debian bookworm InRelease                               Hit:2 http://ftp.de.debian.org/debian bullseye InRelease
Hit:3 http://ftp.de.debian.org/debian bullseye-updates InRelease
Hit:4 https://deb.debian.org/debian-security bullseye-security InRelease
[... 65 lines cut ...]
Armbian 'linux-image-current-rockchip64' for '6.6.34-current-rockchip64': 'postinst' finishing.
Setting up linux-u-boot-rock64-current (24.8.0-trunk-dietpi1) ...
[... cut ...]
root@DietPi:~# ls -l /dev/null-rw-r--r-- 1 root root 0 Jul 12 23:11 /dev/null      
root@DietPi:~# stat /dev/null   File: /dev/null               Size: 0               Blocks: 0          IO Block: 4096   regular empty file
Device: 5h/5dInode: 375         Links: 1
Access: (0644/-rw-r--r--)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2024-07-12 23:10:50.340945456 +0100
Modify: 2024-07-12 23:11:57.188945488 +0100
Change: 2024-07-12 23:11:57.188945488 +0100
 Birth: 2024-07-12 23:09:46.484945425 +0100
root@DietPi:~#

hcr2020, 2024-07-12:
Thank you very much

MichaIng, 2024-07-12:
The `/dev/null` file however is again broken. More importantly, it is broken now, after the reboot. Since `/dev` is recreated from scratch, this means that not the DietPi update, but something else, after the reboot, broke it. Another test:
```sh
ls -l /dev/null
dietpi-services restart
ls -l /dev/null
```

hcr2020, 2024-07-12:
Linux DietPi 6.6.34-current-rockchip64 #2 SMP PREEMPT Sun Jun 16 
11:47:49 UTC 2024 aarch64

The programs included with the Debian GNU/Linux system are free s
oftware;
the exact distribution terms for each program are described in th
e
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jul 12 23:25:47 2024 from 192.168.1.200
 ─────────────────────────────────────────────────────
 DietPi v9.6.1 : 23:29 - Fri 07/12/24
 ─────────────────────────────────────────────────────
 - Device model : ROCK64 (aarch64)
 - CPU temp : 62 °C / 143 °F : Running hot, not recommended
 - LAN IP : 192.168.1.159 (eth0)
 - MOTD : DietPi v9.6 is here. Check out all changes:
          https://dietpi.com/docs/releases/v9_6/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# ls -l /dev/null        -rw-r--r-- 1 root root 0 Jul 12 23:29 /dev/null
root@DietPi:~# dietpi-services restart                                       DietPi-Services
─────────────────────────────────────────────────────
 Mode: restart 

[      ] Di/boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Services | restart : avahi-daemon
[  /boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Services | restart : nmbd
[  /boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Services | restart : smbd
[   /boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Services | restart : nginx
[      /boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file
[  OK  ] DietPi-Services | restart : minidlna
[ INFO ] DietPi-Services | skip : qbittorrent (disabled)
[ INFO ] DietPi-Services | skip : medusa (disabled)
[ INFO ] DietPi-Services | skip : sonarr (disabled)
[ INFO ] DietPi-Services | skip : radarr (disabled)
[  OK  ] DietPi-Services | restart : docker                                 [  /boot/dietpi/func/dietpi-globals: line 288: /dev/null: cannot overwrite existing file                          [  OK  ] DietPi-Services | restart : cron                                   root@DietPi:~# ls -l /dev/null        -rw-r--r-- 1 root root 0 Jul 12 23:29 /dev/null
root@DietPi:~#

MichaIng, 2024-07-12:
Ah sorry, the node needs to be repaired first, of course:
```sh
rm /dev/null
mknod -m 0666 /dev/null c 1 3
ls -l /dev/null
dietpi-services restart
ls -l /dev/null

hcr2020, 2024-07-12:
root@DietPi:~# rm /dev/null
root@DietPi:~# mknod -m 0666 /dev/null c 1 3
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 12 23:38 /dev/null
root@DietPi:~# dietpi-services restart
                                                      DietPi-Services
───────────────────────────────────────────────────── Mode: restart 

[  OK  ] DietPi-Services | restart : avahi-daemon    [  OK  ] DietPi-Services | restart : nmbd            [  OK  ] DietPi-Services | restart : smbd            [  OK  ] DietPi-Services | restart : nginx           [  OK  ] DietPi-Services | restart : minidlna        [ INFO ] DietPi-Services | skip : qbittorrent (disabled)
[ INFO ] DietPi-Services | skip : medusa (disabled)  [ INFO ] DietPi-Services | skip : sonarr (disabled)  [ INFO ] DietPi-Services | skip : radarr (disabled)  [  OK  ] DietPi-Services | restart : docker          [  OK  ] DietPi-Services | restart : cron            root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 12 23:38 /dev/null
root@DietPi:~#

hcr2020, 2024-07-12:
I have made modification like this last comment

https://www.reddit.com/r/dietpi/comments/ttnaw2/dietpi_auto_update/
https://dietpi.com/forum/t/automate-dietpi-update-with-no-user-input/16678

MichaIng, 2024-07-12:
Okay, service restarts did not break it. Can you do another `reboot` and see if it is broken again? If so, let's check journal logs whether we can see related errors from some point on:
```sh
journalctl
```

hcr2020, 2024-07-13:
`root@DietPi:~# ls -l /dev/null        
-rw-r--r-- 1 root root 0 Jul 13 06:21 
/dev/null
root@DietPi:~# uptime                                 06:23:01 up  6:35,  2 users,  load average: 0.48, 0.59, 0.62
root@DietPi:~# journalctl | grep null                Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
root@DietPi:~#
`

hcr2020, 2024-07-13:
`checkgw.sh)
Jul 13 01:25:02 DietPi CRON[15220]: (root) CMD (test -x /usr/sbin/anacron || { cd / && run-parts --report /etc/cron.daily; })
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: Starting Network Time Synchronization...
Jul 13 01:25:02 DietPi systemd[15237]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/dev: Invalid argument
Jul 13 01:25:02 DietPi systemd[15237]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Invalid argument
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 1.
Jul 13 01:25:02 DietPi systemd[1]: Stopped Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: Starting Network Time Synchronization...
Jul 13 01:25:02 DietPi systemd[15242]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/dev: Invalid argument
Jul 13 01:25:02 DietPi systemd[15242]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Invalid argument
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 2.
Jul 13 01:25:02 DietPi systemd[1]: Stopped Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: Starting Network Time Synchronization...
Jul 13 01:25:02 DietPi systemd[15248]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/dev: Invalid argument
Jul 13 01:25:02 DietPi systemd[15248]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Invalid argument
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 3.
Jul 13 01:25:02 DietPi systemd[1]: Stopped Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: Starting Network Time Synchronization...
Jul 13 01:25:02 DietPi systemd[15252]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/dev: Invalid argument
Jul 13 01:25:02 DietPi systemd[15252]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Invalid argument
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 4.
Jul 13 01:25:02 DietPi systemd[1]: Stopped Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: /dev/null is not a device.
Jul 13 01:25:02 DietPi systemd[1]: Starting Network Time Synchronization...
Jul 13 01:25:02 DietPi systemd[15260]: systemd-timesyncd.service: Failed to set up mount namespacing: /run/systemd/unit-root/dev: Invalid argument
Jul 13 01:25:02 DietPi systemd[15260]: systemd-timesyncd.service: Failed at step NAMESPACE spawning /lib/systemd/systemd-timesyncd: Invalid argument
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Main process exited, code=exited, status=226/NAMESPACE
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Scheduled restart job, restart counter is at 5.
Jul 13 01:25:02 DietPi systemd[1]: Stopped Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Start request repeated too quickly.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Start request repeated too quickly.
Jul 13 01:25:02 DietPi systemd[1]: systemd-timesyncd.service: Failed with result 'exit-code'.
Jul 13 01:25:02 DietPi systemd[1]: Failed to start Network Time Synchronization.
Jul 13 01:25:12 DietPi CRON[15219]: pam_unix(cron:session): session closed for user root
Jul 13 01:25:35 DietPi CRON[15218]: pam_unix(cron:session): session closed for user root
`
It self rebooted when I was asleep, because of cron job I wrote if no internet is detected, because it froze after 20 days of uptime wih no reason that I could found.

hcr2020, 2024-07-13:
After reboot
````console
root@DietPi:~# ls -l /dev/null        -rw-r--r-- 1 root root 0 Jul 13 17:01 /dev/null
root@DietPi:~# journalctl | grep null root@DietPi:~#
````

MichaIng, 2024-07-13:
The daily cron job throws this error, but we don't know yet whether it also causes is, or whether it is done at boot only. Can you try this:
```sh
rm /dev/null
mknod -m 0666 /dev/null c 1 3
cd /
run-parts --report /etc/cron.hourly
ls -l /dev/null
run-parts --report /etc/cron.daily
ls -l /dev/null
```
The daily cron job might trigger a `dietpi-sync` and/or `dietpi-backup`, just in case you configures either to be done daily.

hcr2020, 2024-07-13:
```console
root@DietPi:~# rm /dev/null                                                            root@DietPi:~# mknod -m 0666 /dev/null c 1 3
root@DietPi:~# cd /
root@DietPi:/# run-parts --report /etc/cron.hourly
root@DietPi:/# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:/# run-parts --report /etc/cron.daily

root@DietPi:/# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:/#
````

hcr2020, 2024-07-13:
This is the setting that I changed to 2

https://www.reddit.com/r/dietpi/s/fJpMbtLuUY

MichaIng, 2024-07-13:
Since the daily cron job did not cause it, the daily update check obviously does not cause the issue. So it seems to be something at boot then. Let's try some potential candidates:
```sh
ls -d /dev/null
systemctl start systemd-tmpfiles-setup-dev-early
ls -d /dev/null
systemctl start systemd-tmpfiles-setup-dev
ls -d /dev/null
systemctl start systemd-tmpfiles-setup
ls -d /dev/null
systemctl start systemd-tmpfiles-clean
ls -d /dev/null
```

hcr2020, 2024-07-13:
```console
root@DietPi:~# ls -d /dev/null                                                         /dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup-dev-early
Failed to start systemd-tmpfiles-setup-dev-early.service: Unit systemd-tmpfiles-setup-dev-early.service not found.
root@DietPi:~# ls -d /dev/null
/dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup-dev
root@DietPi:~# ls -d /dev/null
/dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup
root@DietPi:~# ls -d /dev/null
/dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-clean
root@DietPi:~# ls -d /dev/null
/dev/null
root@DietPi:~#
````

MichaIng, 2024-07-13:
Oh typo, `ls -l` of course, to see the details of the file:
```sh
ls -l /dev/null
systemctl start systemd-tmpfiles-setup-dev-early
ls -l /dev/null
systemctl start systemd-tmpfiles-setup-dev
ls -l /dev/null
systemctl start systemd-tmpfiles-setup
ls -l /dev/null
systemctl start systemd-tmpfiles-clean
ls -l /dev/null
```

hcr2020, 2024-07-13:
````console
root@DietPi:~# ls -l /dev/null                                                         crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup-dev-early
Failed to start systemd-tmpfiles-setup-dev-early.service: Unit systemd-tmpfiles-setup-dev-early.service not found.
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup-dev
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-setup
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~# systemctl start systemd-tmpfiles-clean
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~#
`````

MichaIng, 2024-07-13:
I just recognised that the setup jobs "remain after exit", so they need to be restarted to do something. Let's test this and some more:
```sh
ls -l /dev/null
systemctl restart systemd-tmpfiles-setup-dev
ls -l /dev/null
systemctl restart systemd-tmpfiles-setup
ls -l /dev/null
systemctl restart dietpi-ramlog
ls -l /dev/null
systemctl restart dietpi-preboot
ls -l /dev/null
systemctl restart dietpi-postboot
ls -l /dev/null
. /etc/bashrc.d/dietpi.bash
ls -l /dev/null
```

hcr2020, 2024-07-13:
````console
root@DietPi:~# ls -l /dev/null                                 crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null             root@DietPi:~# systemctl restart dietpi-ramlog                 root@DietPi:~# ls -l /dev/null                                 crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null             root@DietPi:~# systemctl restart dietpi-preboot                root@DietPi:~# ls -l /dev/null                                 crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null             root@DietPi:~# systemctl restart dietpi-postboot               root@DietPi:~# ls -l /dev/null                                 crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null             root@DietPi:~# source /etc/bashrc.d/dietpi.bash                 ─────────────────────────────────────────────────────          DietPi v9.6.1 : 21:25 - Sat 07/13/24                           ─────────────────────────────────────────────────────
 - Device model : ROCK64 (aarch64)
 - CPU temp : 60 °C / 140 °F : Running hot, not recommended
 - LAN IP : 192.168.1.159 (eth0)
 - MOTD : DietPi v9.6 is here. Check out all changes:
          https://dietpi.com/docs/releases/v9_6/
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://twitter.com/DietPi_
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 18:51 /dev/null
root@DietPi:~#

`````

MichaIng, 2024-07-13:
Did you restart the first to as well?
```sh
rm /dev/null
mknod -m 0666 /dev/null c 1 3
systemctl restart systemd-tmpfiles-setup-dev
ls -l /dev/null
systemctl restart systemd-tmpfiles-setup
ls -l /dev/null
```
I intentionally re-added them, since preciously it was just `start`, which is a no-op.

hcr2020, 2024-07-13:
Yed and I restarted my box
````console
root@DietPi:~# rm /dev/null                   
root@DietPi:~# mknod -m 0666 /dev/null c 1 3
root@DietPi:~# systemctl restart systemd-tmpfiles-setup-dev
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 21:46 /dev/null
root@DietPi:~# systemctl restart systemd-tmpfiles-setup
Failed to restart systemd-tmpfiles-setup.service: Operation refused, unit systemd-tmpfi
les-setup.service may be requested by dependency only (it is configured to refuse manua
l start/stop).
See system logs and 'systemctl status systemd-tmpfiles-setup.service' for details.
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 13 21:46 /dev/null
root@DietPi:~#
`````

MichaIng, 2024-07-13:
I wonder whether there is a better way to find the culprit than testing each and every service. There is a `loggedfs`: https://github.com/rflament/loggedfs

Let's try first to narrow it down first, and at the same time check which DietPi script we can use to setup the `loggedfs`:
```sh
sed -i '1i\ls -l /dev/null | logger -t devnull' /boot/dietpi/func/dietpi-ramlog
sed -i '1i\ls -l /dev/null | logger -t devnull' /boot/dietpi/preboot
sed -i '1i\ls -l /dev/null | logger -t devnull' /boot/dietpi/postboot
reboot
```
Then after reboot:
```sh
journalctl -t devnull
```

hcr2020, 2024-07-13:
````console
root@DietPi:~# journalctl -t devnull             
-- Journal begins at Sat 2024-07-13 21:57:36 BST, ends at Sat 2024-07-13 22:01:44 BST.
--
Jul 13 21:57:38 DietPi devnull[380]: crw-rw-rw- 1 root root 1, 3 Jan  1  1970 /dev/null
root@DietPi:~# ls -l /dev/null                            -rw-r--r-- 1 root root 0 Jul 13 22:01 /dev/null
root@DietPi:~#
````

MichaIng, 2024-07-13:
Hmm, are you sure you ran all three commands above? There should be 3 log entries, not just one 🤔.

hcr2020, 2024-07-13:
````console

rrent-rockchip64] (local build)
Jul 13 21:57:39 DietPi smartd[370]: Copyrig
ht (C) 2002-20, Bruce Allen, Christian Fran
ke, www.smartmontools.org
Jul 13 21:57:39 DietPi kernel: Adding 26214
0k swap on /dev/zram0.  Priority:100 extent
s:1 across:262140k SS
Jul 13 21:57:39 DietPi smartd[370]: Opened
configuration file /etc/smartd.conf
Jul 13 21:57:39 DietPi smartd[370]: Drive:
DEVICESCAN, implied '-a' Directive on line
21 of file /etc/smartd.conf
Jul 13 21:57:39 DietPi smartd[370]: Configu
ration file /etc/smartd.conf was parsed, fo
und DEVICESCAN, scanning devices
Jul 13 21:57:39 DietPi smartd[370]: In the 
system's table of devices NO devices found 
to scan
Jul 13 21:57:39 DietPi smartd[370]: Monitor
ing 0 ATA/SATA, 0 SCSI/SAS and 0 NVMe devic
es
Jul 13 21:57:39 DietPi kernel: cfg80211: Lo
ading compiled-in X.509 certificates for re
gulatory database
Jul 13 21:57:39 DietPi kernel: Loaded X.509
 cert 'sforshee: 00b28ddf47aef9cea7'
Jul 13 21:57:39 DietPi kernel: Loaded X.509
 cert 'wens: 61c038651aabdcf94bd0ac7ff06c72
48db18c600'
Jul 13 21:57:39 DietPi kernel: brcmfmac: F1
 signature read @0x18000000=0x16034334
Jul 13 21:57:39 DietPi systemd[1]: Started
Self Monitoring and Reporting Technology (S
MART) Daemon.
Jul 13 21:57:39 DietPi systemd[1]: Finished
 Linux zramswap setup.
Jul 13 21:57:39 DietPi systemd[1]: Found de
vice /sys/subsystem/net/devices/eth0.
Jul 13 21:57:39 DietPi kernel: brcmfmac: br
cmf_fw_alloc_request: using brcm/brcmfmac43
34-sdio for chip BCM4334/3
Jul 13 21:57:39 DietPi kernel: usbcore: reg
istered new interface driver brcmfmac
Jul 13 21:57:39 DietPi kernel: brcmfmac mmc
1:0001:1: Direct firmware load for brcm/brc
mfmac4334-sdio.rockchip,rk3318-box.bin fail
ed with error -2
Jul 13 21:57:39 DietPi kernel: brcmfmac mmc
1:0001:1: Direct firmware load for brcm/brc
mfmac4334-sdio.clm_blob failed with error -
2
Jul 13 21:57:39 DietPi systemd[1]: Reached
target Sound Card.
Jul 13 21:57:39 DietPi systemd[1]: Listenin
g on Load/Save RF Kill Switch Status /dev/r
fkill Watch.
Jul 13 21:57:39 DietPi systemd[1]: Finished
 Helper to synchronize boot up for ifupdown
.
Jul 13 21:57:39 DietPi kernel: brcmfmac: br
cmf_c_process_clm_blob: no clm_blob availab
le (err=-2), device may have limited channe
ls available
Jul 13 21:57:39 DietPi kernel: brcmfmac: br
cmf_c_process_txcap_blob: no txcap_blob ava
ilable (err=-2)
Jul 13 21:57:39 DietPi kernel: brcmfmac: br
cmf_c_preinit_dcmds: Firmware: BCM4334/3 wl
[... 413 lines cut ...]
ory
Jul 13 21:58:12 DietPi systemd[1]: Started
Samba SMB Daemon.
Jul 13 21:58:12 DietPi systemd[1]: Reached
target Multi-User System.
Jul 13 21:58:13 DietPi systemd[669]: dietpi
-postboot.service: Failed to execute /boot/
dietpi/postboot: Exec format error
Jul 13 21:58:13 DietPi systemd[669]: dietpi
-postboot.service: Failed at step EXEC spaw
ning /boot/dietpi/postboot: Exec format err
or
Jul 13 21:58:13 DietPi systemd[1]: dietpi-p
ostboot.service: Main process exited, code=
exited, status=203/EXEC
Jul 13 21:58:13 DietPi systemd[1]: dietpi-p
ostboot.service: Failed with result 'exit-c
ode'.
Jul 13 21:58:13 DietPi login[685]: pam_unix
(login:session): session opened for user ro
ot(uid=0) by LOGIN(uid=0)
Jul 13 21:58:14 DietPi login[987]: ROOT LOG
IN  on '/dev/tty1'

````

hcr2020, 2024-07-13:
````console
 mawk 'NR==1' /boot/dietpi/func/dietpi-ramlog /boot/dietpi/preboot /boot/dietpi/postboot                     ls -l /dev/null | logger -t devnull

````

MichaIng, 2024-07-13:
Ah shit, I added them wrong. New attempt 😄:
```sh
sed -i '/logger -t devnull/d' /boot/dietpi/func/dietpi-ramlog /boot/dietpi/preboot /boot/dietpi/postboot
sed -i '1a\ls -l /dev/null | logger -t devnull' /boot/dietpi/func/dietpi-ramlog
sed -i '1a\ls -l /dev/null | logger -t devnull' /boot/dietpi/preboot
sed -i '1a\ls -l /dev/null | logger -t devnull' /boot/dietpi/postboot
reboot
```
The first line is the shebang, which needs to stay there, so the logger command needs to be added after the first line, not before/as first line.

hcr2020, 2024-07-13:
````console
root@DietPi:~# journalctl -t devnull
-- Journal begins at Sat 2024-07-13 22:27:50 BST, ends at Sat 2024-07-13 22:30:42 BST.
--
Jul 13 22:27:52 DietPi devnull[385]: crw-rw-rw- 1 root root 1, 3 Jan  1  1970 /dev/null
Jul 13 22:27:54 DietPi devnull[467]: crw-rw-rw- 1 root root 1, 3 Jul 13 22:27 /dev/null
Jul 13 22:28:30 DietPi devnull[1018]: crw-rw-rw- 1 root root 1, 3 Jul 13 22:27 /dev/nul
l
root@DietPi:~# ls -l /dev/null    -rw-r--r-- 1 root root 0 Jul 13 22:30 /dev/null
root@DietPi:~#
`````

MichaIng, 2024-07-13:
Okay, so everything is fine at least until `dietpi-postboot` runs. One thing it does, which we did not test yet and which is spawned into background:
```sh
rm /dev/null
mknod -m 0666 /dev/null c 1 3
/boot/dietpi/func/run_ntpd
ls -l /dev/null
/boot/dietpi/func/run_ntpd 1
ls -l /dev/null
```
If the node is still fine, then let's check `journalctl` for all services/targets which start or finish after `dietpi-postboot`:
```sh
journalctl -e
```
Try to locate "Started dietpi-postboot.service - DietPi-PostBoot" and copy&paste everything afterwards, or just the whole output.

hcr2020, 2024-07-13:
`````console 
not found"
Jul 13 22:30:04 DietPi dockerd[2294]: time="2024-07-1
3T22:30:04.826898629+01:00" level=warning msg="error
locating sandbox id b632da880f83172b60c786df7291bb6ee
5ffb229d3519a2856c5e874e17d0f84: sandbox b632da880f83
172b60c786df7291bb6ee5ffb229d3519a2856c5e874e17d0f84
not found"
Jul 13 22:30:04 DietPi dockerd[2294]: time="2024-07-1
3T22:30:04.827791421+01:00" level=warning msg="error
locating sandbox id 73b65266af78196d0b8f14b9965e0d8a3
39c13c2c90cff72a275f296d1160aeb: sandbox 73b65266af78
196d0b8f14b9965e0d8a339c13c2c90cff72a275f296d1160aeb
not found"
Jul 13 22:30:04 DietPi dockerd[2294]: time="2024-07-1
3T22:30:04.828528171+01:00" level=warning msg="error
locating sandbox id f28d9e2f720038dc48a7a6d17b98c7363
98c3c6b443374aca43e12d1cb9f1a5f: sandbox f28d9e2f7200
38dc48a7a6d17b98c736398c3c6b443374aca43e12d1cb9f1a5f
not found"
Jul 13 22:30:04 DietPi dockerd[2294]: time="2024-07-1
3T22:30:04.829942463+01:00" level=warning msg="error
locating sandbox id 80da8bf682b0e27f82de0a9304004ec1e
c1db8c3c26a00db5b551809628a2364: sandbox 80da8bf682b0
e27f82de0a9304004ec1ec1db8c3c26a00db5b551809628a2364
not found"
Jul 13 22:30:04 DietPi dockerd[2294]: time="2024-07-1
3T22:30:04.830796171+01:00" level=warning msg="error
locating sandbox id e877027340efaad5e95f55235b7a19513
125c47b5fcce51c81232c556ab8b8ce: sandbox e877027340ef
aad5e95f55235b7a19513125c47b5fcce51c81232c556ab8b8ce
not found"
Jul 13 22:30:05 DietPi kernel: br-8614d213917a: port
1(vethc152e5d) entered blocking state
Jul 13 22:30:05 DietPi kernel: br-8614d213917a: port
1(vethc152e5d) entered disabled state
Jul 13 22:30:05 DietPi kernel: vethc152e5d: entereda
llmulticast mode
Jul 13 22:30:05 DietPi kernel: vethc152e5d: enteredp
romiscuous mode
Jul 13 22:30:05 DietPi kernel: docker0: port 1(veth46
eba98) entered blocking state
Jul 13 22:30:05 DietPi kernel: docker0: port 1(veth46
eba98) entered disabled state
Jul 13 22:30:05 DietPi kernel: veth46eba98: entereda
llmulticast mode
Jul 13 22:30:05 DietPi kernel: veth46eba98: enteredp
romiscuous mode
Jul 13 22:30:05 DietPi kernel: docker0: port 1(veth46
eba98) entered blocking state
Jul 13 22:30:05 DietPi kernel: docker0: port 1(veth46
eba98) entered forwarding state
Jul 13 22:30:05 DietPi kernel: docker0: port 1(veth46
eba98) entered disabled state
Jul 13 22:30:05 DietPi kernel: br-8614d213917a: port
1(vethc152e5d) entered blocking state
Jul 13 22:30:05 DietPi kernel: br-8614d213917a: port
1(vethc152e5d) entered forwarding state
Jul 13 22:30:05 DietPi kernel: br-8614d213917a: port
1(vethc152e5d) entered disabled state
Jul 13 22:30:05 DietPi kernel: br-4bf64785e6d8: port
1(vethc9c8a33) entered blocking state
Jul 13 22:30:05 DietPi kernel: br-4bf64785e6d8: port
1(vethc9c8a33) entered disabled state
Jul 13 22:30:05 DietPi kernel: vethc9c8a33: entereda
llmulticast mode
Jul 13 22:30:05 DietPi kernel: vethc9c8a33: enteredp
romiscuous mode
Jul 13 22:30:05 DietPi kernel: br-4bf64785e6d8: port
1(vethc9c8a33) entered blocking state
[... 905 lines cut ...]
Jul 13 22:43:08 DietPi sshd[6142]: Accepted password
for root from 192.168.1.200 port 59834 ssh2
Jul 13 22:43:08 DietPi sshd[6142]: pam_unix(sshd:sess
ion): session opened for user root(uid=0) by (uid=0)
Jul 13 22:43:09 DietPi systemd[1]: Starting Cleanupo
f Temporary Directories...
Jul 13 22:43:09 DietPi systemd[1]: systemd-tmpfiles-c
lean.service: Succeeded.
Jul 13 22:43:09 DietPi systemd[1]: Finished Cleanupo
f Temporary Directories.
Jul 13 22:43:13 DietPi systemd[1]: Starting NetworkT
ime Synchronization...
Jul 13 22:43:14 DietPi systemd[1]: Started Network Ti
me Synchronization.
Jul 13 22:43:14 DietPi systemd-timesyncd[6201]: Initi
al synchronization to time server 141.98.136.83:123(
0.debian.pool.ntp.org).
Jul 13 22:43:15 DietPi systemd[1]: Stopping NetworkT
ime Synchronization...
Jul 13 22:43:15 DietPi systemd[1]: systemd-timesyncd.
service: Succeeded.
Jul 13 22:43:15 DietPi systemd[1]: Stopped Network Ti
me Synchronization.
root@DietPi:~#
````

MichaIng, 2024-07-13:
Nearly everything is the Docker engine and container startup, which should not be related, then I see the SSH logins, the tmpfiles cleanup (which we tested already) and the network time sync we just tested. The only other thing I see is this script executed from cron:
```sh
cat /opt/checkgw.sh
```
Though this runs some minutes after your first SSH login. There is another cron "session closed" log entry earlier, from which we do not see the start entry. So additionally, can you paste the cron logs again?
```sh
journalctl -u cron
```

hcr2020, 2024-07-13:
````console
@DietPi:~# journalctl | grep DietPi-PostBoot                                  Jul 13 22:28:01 DietPi systemd[1]: Started DietPi-PostBoot.
root@DietPi:~#
root@DietPi:~# journalctl -e --since "2024-07-13 22:28:01" --until=now > /mnt/sdb1/log.txt
````
Please find log.txt 
[log.txt](https://github.com/user-attachments/files/16203838/log.txt)

hcr2020, 2024-07-13:
````console 
root@DietPi:~# cat /opt/checkgw.sh
#!/bin/bash                  
maxPloss=10 #Maximum percent packet loss
before a restart             
# Initialize to a value that would force
a restart                    
# (just in case ping gives an error and p
loss doesn't get set)        
#modprobe ip6_tables         
ploss=101                    
ploss=$(ping -q -w10 dietpi.com | grep -o
 "[0-9]*%" | tr -d %) > /dev/null 2>&1
if [ "$ploss" -gt "$maxPloss" ]; then
        logger "internet unstable"
        echo "$(date "+%Y-%m-%d %H:%M:%S"
) : internet unstable" | tee -a /opt/netw
ork_check
x=`ping -c1 ifconfig.me 2>&1 | gr
ep unknown`
if [ ! "$x" = "" ]; then
        logger  "Internet's down!! Attemp
ting to ping gateway."
echo "$(date "+%Y-%m-%d %H:%M:%S"
) : Internet's down" | tee -a /opt/networ
k_check
        ping -c3 192.168.1.1 &> /dev/null
if [ $? -gt 0 ]then
#Make a note in s
yslog
echo "$(date "+%Y
-%m-%d %H:%M:%S") : gateway is unreacheab
le down restarting H96Max+" | tee -a /opt
/network_check
        logger "network_c
heck: Network connection is down, restart
ing H96Max+ ..."
reboot
fi
fi
fi

root@DietPi:~#
`````

`````console
t@DietPi:~# journalctl -u cron        -- Journal begins at Sat 2024-07-13 22:27:50 BST, ends at Sat 2024-07-13 23:55:01 BST. --
Jul 13 22:27:52 DietPi systemd[1]: Started Regular background program processing daemon.
Jul 13 22:27:52 DietPi cron[354]: (CRON) INFO (pidfile fd = 3)
Jul 13 22:27:52 DietPi cron[354]: (CRON) INFO (Running @reboot jobs)
Jul 13 22:30:01 DietPi CRON[2813]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:30:01 DietPi CRON[2816]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:30:11 DietPi CRON[2813]: pam_unix(cron:session): session closed for user root
Jul 13 22:35:01 DietPi CRON[5198]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:35:01 DietPi CRON[5199]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:35:11 DietPi CRON[5198]: pam_unix(cron:session): session closed for user root
Jul 13 22:40:01 DietPi CRON[5770]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:40:01 DietPi CRON[5771]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:40:11 DietPi CRON[5770]: pam_unix(cron:session): session closed for user root
Jul 13 22:45:01 DietPi CRON[6423]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:45:01 DietPi CRON[6424]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:45:11 DietPi CRON[6423]: pam_unix(cron:session): session closed for user root
Jul 13 22:50:01 DietPi CRON[6995]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:50:01 DietPi CRON[6996]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:50:11 DietPi CRON[6995]: pam_unix(cron:session): session closed for user root
Jul 13 22:55:01 DietPi CRON[7583]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 22:55:01 DietPi CRON[7584]: (root) CMD (/opt/checkgw.sh)
Jul 13 22:55:11 DietPi CRON[7583]: pam_unix(cron:session): session closed for user root
Jul 13 23:00:02 DietPi CRON[8166]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:00:02 DietPi CRON[8167]: (root) CMD (/opt/checkgw.sh)
[... 13 lines cut ...]
Jul 13 23:20:01 DietPi CRON[10492]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:20:01 DietPi CRON[10493]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:20:11 DietPi CRON[10492]: pam_unix(cron:session): session closed for user root
Jul 13 23:25:01 DietPi CRON[11045]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:25:01 DietPi CRON[11046]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:25:11 DietPi CRON[11045]: pam_unix(cron:session): session closed for user root
Jul 13 23:30:01 DietPi CRON[11618]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:30:01 DietPi CRON[11619]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:30:11 DietPi CRON[11618]: pam_unix(cron:session): session closed for user root
Jul 13 23:35:01 DietPi CRON[12223]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:35:01 DietPi CRON[12224]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:35:11 DietPi CRON[12223]: pam_unix(cron:session): session closed for user root
Jul 13 23:40:01 DietPi CRON[12806]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:40:01 DietPi CRON[12807]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:40:11 DietPi CRON[12806]: pam_unix(cron:session): session closed for user root
Jul 13 23:45:01 DietPi CRON[13374]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:45:01 DietPi CRON[13375]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:45:11 DietPi CRON[13374]: pam_unix(cron:session): session closed for user root
Jul 13 23:50:01 DietPi CRON[13935]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:50:01 DietPi CRON[13936]: (root) CMD (/opt/checkgw.sh)
Jul 13 23:50:11 DietPi CRON[13935]: pam_unix(cron:session): session closed for user root
Jul 13 23:55:01 DietPi CRON[14494]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 13 23:55:01 DietPi CRON[14495]: (root) CMD (/opt/checkgw.sh)
root@DietPi:~#
````

MichaIng, 2024-07-13:
So indeed the first cron entry was also from this script. Can you check whether it is the culprit:
```sh
rm /dev/null
mknod -m 0666 /dev/null c 1 3
/opt/checkgw.sh
ls -l /dev/null
```
It seems to have at least one syntax error, but it could be also caused by copy&paste.

hcr2020, 2024-07-14:
````console
root@DietPi:~# rm /dev/null                   root@DietPi:~# mknod -m 0666 /dev/null c 1 3
root@DietPi:~# /opt/checkgw.sh

root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 14 06:49 /dev/null
root@DietPi:~# ls -l /dev/null
crw-rw-rw- 1 root root 1, 3 Jul 14 06:49 /dev/null
````

hcr2020, 2024-07-14:
I  commented the cron entry calling checkgw.sh using crontab -e
I executed this
````console 
rm /dev/null
mknod -m 0666 /dev/null c 1 3

root@DietPi:~# ls -l /dev/null 
crw-rw-rw- 1 root root 1, 3 Jul 14 21:15 /dev/null 

root@DietPi:~# reboot

root@DietPi:~# ls -l /dev/null
-rw-r--r-- 1 root root 0 Jul 14 21:19 /dev/null
root@DietPi:~# journalctl -u cron
-- Journal begins at Sun 2024-07-14 21:14:40 BST, ends atSun2024-07
-14 21:19:39 BST. --
Jul 14 21:14:42 DietPi systemd[1]: Started Regular backgroundprogram
 processing daemon.
Jul 14 21:14:42 DietPi cron[353]: (CRON) INFO (pidfile fd=3)
Jul 14 21:14:42 DietPi cron[353]: (CRON) INFO (Running @rebootjobs)
Jul 14 21:17:01 DietPi CRON[4013]: pam_unix(cron:session):session opened for user root(uid=0) by (uid=0)
Jul 14 21:17:01 DietPi CRON[4020]: (root) CMD (cd / && run-parts--re
port /etc/cron.hourly)
Jul 14 21:17:03 DietPi CRON[4013]: pam_unix(cron:session):session closed for user root
root@DietPi:~#
`````

hcr2020, 2024-07-14:
Can you update dietpi-update to include this
rm /dev/null
 mknod -m 0666 /dev/null c 1 3

MichaIng, 2024-07-16:
This is not a solution. We need to find out what exactly is removing/replacing the node, and fix it.

I btw found the reason for the issue on my servers: https://github.com/MichaIng/hacks/blob/main/rootfs/etc/mysql/mariadb.conf.d/99-micha.cnf#L3
`pid_file=/dev/null` in MariaDB now leads to the removal of the node, when the service is stopped. I tried to find a way to disable the unnecessary PID file with a drop-in config, and this worked, but obviously broke with the latest MariaDB update on Debian Trixie. However this is not the reason in your case: You do not use MariaDB, service stops/restarts do not trigger the issue, reboots do (in my case not), and this config was never built into DietPi, but just my personal one.

This however means it really is a individual problem, not a general one, and it was a coincidence that we both have it, with a very different culprit.

I have one idea how to probably find out what is causing it:
```sh
cat << '_EOF_' > /var/lib/dietpi/postboot.d/devnull.sh
#!/bin/bash
while [[ $(stat -c '%a' /dev/null) == '666' ]]; do :; done
lsof /dev/null > /root/devnull.log
_EOF_
```
I hope postboot scripts run early enough and `lsof` runs fast enough to catch it. This is causing a constant CPU load of ~15% (on two of my systems) since the mode of the node is checked very frequently, but I currently have no better idea. Check the log after reboot:
```sh
cat /root/devnull.log
```

MichaIng, 2025-06-07:
Yes, above I suggested another way to find the culprit:
```sh
cat << '_EOF_' > /var/lib/dietpi/postboot.d/devnull.sh
#!/bin/bash
while [[ $(stat -c '%a' /dev/null) == '666' ]]; do :; done
lsof /dev/null > /root/devnull.log
_EOF_
reboot
```
And after reboot:
```sh
cat /root/devnull.log
```