MichaIng/DietPi #7163: Jre-17-headless incompatible with PaperMC 1.21

osakaruhub, 2024-07-17:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version |
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | 'bookworm'
- Kernel version | Linux dietpi 6.6.34-current-rockchip64 #2 SMP PREEMPT Sun Jun 16 11:47:49 UTC 2024 aarch64 GNU/Linux
- SBC model | ROCK64 (aarch64)
- Power supply used | 5V 1A standard barrel
- SD card used |  SanDisk ultra 32G

#### Additional Information (if applicable)
- Software title | PaperMC
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
yes.

#### Steps to reproduce

1. sudo dietpi-software install 181
2. systemctl start papermc.service

logs:
journalctl -u papermc

#### Expected behaviour
PaperMC is running as expected after starting

#### Actual behaviour
PaperMC daemon stopped for UnsupportedClassVersionError

#### Extra details
from #4505 it already mentioned that the new upcoming releases will be needing newer versions of java. the installation of PaperMC includes the 1.21 version needing java 21, while dietpi installs the latest stable debian version (17) as mentioned here: `curl -sSfL https://papermc.io/api/v2/projects/paper/versions/1.21/builds/81/downloads/paper-1.21-81.jar -o paper-1.21-81.jar`

The Options could be
- include a way to install different java versions from oracle (temporarely from the tar.gz releases),
- downgrade PaperMC to the latest java 17-compatible version, which would be 1.16.5

MichaIng, 2024-07-18:
Thanks for the hint. I have no time to take care of support for different Java versions, and also would not do it for a single software title only. For now, easier is to install the lastest compatible version of PaperMC. Paper v1.17 requires Java 17, which is supported on Bullseye and Bookworm already.

Paper v1.20.y still supports Java 17, right? I mix up Paper versions and Minecraft versions 😄.

MichaIng, 2025-01-07:
Have Paper versions and Minecraft version been aligned in the meantime? Forgetting again about the difference, I implemented the install of a Paper version, matching the latest Minecraft which supports Java 17 (1.20.4), and it works: https://github.com/MichaIng/DietPi/commit/e0430a7

... wait, it has become much easier ...

MichaIng, 2025-01-07:
Fixed for now, but I actually aim to implement the Adoptium repo until release.

closed 2025-02-09 by MichaIng