MichaIng/DietPi #7164: Unable to upgrade DietPi from v9.5.1 to v9.6.1

Drajcoshi, 2024-07-18:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
-- Date           | Thu Jul 18 09:41:00 BST 2024
- DietPi version | v9.5.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux pi4B-Dietpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Get:2 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 https://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Hit:5 https://dietpi.com/apt bookworm InRelease
Get:6 https://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:7 https://deb.debian.org/debian bookworm-backports InRelease [56.6 kB]
Hit:8 https://dietpi.com/apt all InRelease
Get:3 https://packages.sury.org/php buster InRelease
Err:3 https://packages.sury.org/php buster InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Ign:9 https://download.webmin.com/download/newkey/repository stable InRelease
Hit:10 https://download.webmin.com/download/newkey/repository stable Release
Get:11 https://archive.raspberrypi.com/debian bookworm/main arm64 Packages [424 kB]
Get:12 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:13 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-07-18-0207.35-F-2024-07-17-1417.00.pdiff [21.3 kB]
Get:13 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-07-18-0207.35-F-2024-07-17-1417.00.pdiff [21.3 kB]
Reading package lists...
E: Failed to fetch https://packages.sury.org/php/dists/buster/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
E: The repository 'https://packages.sury.org/php buster InRelease' is not signed.

MichaIng, 2024-07-18:
You are using an incompatible version of Ondrej's PHP repository, which we do not use anymore since Buster. Do you use it intentionally to run a more recent PHP version? If so, change its suite from `buster` to `bookworm`, and probably update its public key as well. Else remove it.

Drajcoshi, 2024-07-18:
Hi Michalng,

I have not install it intentionally. Only have used stock application from the dietpi list of software.

Joulinar, 2024-07-18:
In this case you might remove the source configuration

MichaIng, 2024-07-18:
Probably you did not follow our Buster upgrade instructions then: https://dietpi.com/docs/usage/#how-to-upgrade-to-buster
There are the 3 relevant steps:
```sh
rm -f /etc/apt/sources.list.d/dietpi-php.list
rm -f /etc/apt/trusted.gpg.d/dietpi-php.gpg
rm -f /etc/apt/preferences.d/dietpi-{php,openssl,xrdp}
```

Drajcoshi, 2024-07-19:
> In this case you might remove the source configuration

Hi Joulinar,

How would I do that? I had repeated the Buster upgrade instruction by Michalng. Also made sure I followed the 3 relevant steps. rebooted it and tried again with same error

MichaIng, 2024-07-19:
Please show all your lists:
```sh
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

Drajcoshi, 2024-07-19:
Hi Michalng,

Please see below.

/etc/apt/sources.list:
deb https://deb.debian.org/debian/ bookworm main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-updates main contrib non-free non-free-firmware
deb https://deb.debian.org/debian-security/ bookworm-security main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-backports main contrib non-free non-free-firmware
/etc/apt/sources.list.d/dietpi.list:
deb https://dietpi.com/apt bookworm main rpi4
deb https://dietpi.com/apt all rpi
/etc/apt/sources.list.d/php.list:
deb https://packages.sury.org/php/ buster main
/etc/apt/sources.list.d/raspi.list:
deb https://archive.raspberrypi.com/debian/ bookworm main
/etc/apt/sources.list.d/webmin.list:
deb https://download.webmin.com/download/newkey/repository stable contrib

Drajcoshi, 2024-07-19:
would this be the culprit?

deb https://packages.sury.org/php/ buster main
/etc/apt/sources.list.d/raspi.list:

MichaIng, 2024-07-19:
Yes, remove it:
```sh
rm /etc/apt/sources.list.d/php.list
```
And also check for the related APT key:
```sh
ls -l /etc/apt/trusted.gpg.d
```

Drajcoshi, 2024-07-19:
removed it /etc/apt/sources.list.d/php.list

APT key output

-rw-r--r-- 1 root root  1769 Feb 16  2021 apt.gpg~
-rw-r--r-- 1 root root 11861 Jul 30  2023 debian-archive-bookworm-automatic.asc
-rw-r--r-- 1 root root 11873 Jul 30  2023 debian-archive-bookworm-security-automatic.asc
-rw-r--r-- 1 root root   461 Jul 30  2023 debian-archive-bookworm-stable.asc
-rw-r--r-- 1 root root 11861 Jul 30  2023 debian-archive-bullseye-automatic.asc
-rw-r--r-- 1 root root 11873 Jul 30  2023 debian-archive-bullseye-security-automatic.asc
-rw-r--r-- 1 root root  3403 Jul 30  2023 debian-archive-bullseye-stable.asc
-rw-r--r-- 1 root root 11093 Jul 30  2023 debian-archive-buster-automatic.asc
-rw-r--r-- 1 root root 11105 Jul 30  2023 debian-archive-buster-security-automatic.asc
-rw-r--r-- 1 root root  1704 Jul 30  2023 debian-archive-buster-stable.asc
-rw-r--r-- 1 root root  1769 Jul 19 11:54 dietpi-php.gpg~
-rw-r--r-- 1 root root  2261 Jan 26 16:59 dietpi-webmin.gpg
-rw-r--r-- 1 root root  4694 Apr 16 11:27 dietpi.asc
-rw-r--r-- 1 root root 17948 Jan 11  2018 libffi6_3.2.1-8_amd64.deb
-rw-r--r-- 1 root root     0 Sep 15  2023 mc3man_ubuntu_xerus-media.gpg
-rw-r--r-- 1 root root     0 Sep 10  2023 ondrej_ubuntu_php.gpg
-rw-r--r-- 1 root root  1769 Feb  5 10:26 php.gpg
-rw-r--r-- 1 root root  1183 Jan 13  2022 raspberrypi-archive-stable.gpg
-rw-r--r-- 1 root root     0 Sep 26  2023 ubuntudde-dev_ubuntu_stable.gpg

Drajcoshi, 2024-07-19:
i did install nextcloud for pi from the software list. Had issues with the php version 7.4 as it was outdated. So I had removed it, I wonder if this lingered on. But the purge command should have removed it all though

MichaIng, 2024-07-19:
Debian Bullseye (PHP 7.4) supports Nextcloud 25 only. But you are on Bookworm (PHP 7.2) already, so latest Nextcloud is supported without installing newer PHP from Ondrej's repo. From Buster on, we never installed this repository, just installed the latest supported Nextcloud. But probably you did so manually.

Generally, if you are not somewhat experiences with APT repositories, potential package conflicts (and solving them manually), keeping the list and key updated etc, I recommend to ignore Nextcloud admin panel warnings about your PHP version. If those show up, there is usually a newer Debian version (with newer PHP version) available and we offer a script to upgrade it, including all possibly needed migration steps. This is the better way, IMO.

There are a lot of funny APT keys. Those can be removed:
```sh
rm /etc/apt/trusted.gpg.d/{apt.gpg~,dietpi-php.gpg~,libffi6_3.2.1-8_amd64.deb,mc3man_ubuntu_xerus-media.gpg,ondrej_ubuntu_php.gpg,php.gpg,ubuntudde-dev_ubuntu_stable.gpg}
apt update # failsafe test that all needed keys are still there
```

Drajcoshi, 2024-07-19:
Michalng,

Thanks for explanation and will stick to your method. I have removed the APT keys, and managed to update my dietpi.

thanks a million

Joulinar, 2024-07-19:
Going to close the issue, feel free to reopen if needed

closed 2024-07-19 by Joulinar