MichaIng/DietPi #7168: Odroid N2 Plus: Boot loop when attached USB storage NOT ready

Tin-Bui, 2024-07-22:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm
- Kernel version | `uname -a`
Linux n254 6.6.37-current-meson64 #1 SMP PREEMPT Fri Jul  5 07:34:07 UTC 2024 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid N2 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
12V 2A
- SD card used | (EG: SanDisk ultra)
Sandisk High Endurance 128GB
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
N/A
- Was the software title installed freshly or updated/migrated?
N/A
- Can this issue be replicated on a fresh installation of DietPi?
Yes
- Bug report ID | `echo $G_HW_UUID`
792a1110-3713-4144-ae3d-41ff25f28819
#### Steps to reproduce
1. Flash dietpi image: DietPi_OdroidN2-ARMv8-Bookworm.img
2. Power on the Odroid N2 board with Western Digital External HardDrive via USB port.
3. Observe the u-boot process scanning for USB storage device, found them NOT ready and then resetting CPU, and repeat the cycle.

#### Expected behaviour
u-boot proceed to boot from mmc after detecting USB storage not ready.

#### Actual behaviour
Boot loop when detecting USB storage NOT ready

#### Extra details
I manually got around the issue by:
1 .Hit any key to interrup the boot process
2. Change the boot_target environment variable: setenv boot_targets 'mmc0 mmc1 mmc2 romusb usb0 pxe dhcp'
3. => boot : succesful.
But this require interruption on every boot.
Is there anyway to change and persist the boot_target environment variable?

See output from the boot process:

````
G12B:BL:6e7c85:2a3b91;FEAT:E0F83180:402000;POC:F;RCY:0;EMMC:800;NAND:81;SD?:0;SD:0;READ:0;0.▒!,K▒▒х▒▒}▒▒▒с0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

no sdio debug board detected
L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 153057

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 5
[... 426 lines cut ...]
## Loading init Ramdisk from Legacy Image at 13000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    13650025 Bytes = 13 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 04080000
   Booting using the fdt blob at 0x4080000
   Loading Ramdisk to 3f2fb000, end 3ffff869 ... OK
   Loading Device Tree to 000000003f2e4000, end 000000003f2fa8ef ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.37-current-meson64 (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Fri Jul  5 07:34:07 UTC 2024
[    0.000000] Machine model: Hardkernel ODROID-N2Plus
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x00000000e4c00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x00000000e4c00000..0x00000000f4bfffff (262144 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000005000000..0x00000000052fffff (3072 KiB) nomap non-reusable secmon@5000000
[    0.000000] OF: reserved mem: 0x0000000005300000..0x00000000072fffff (32768 KiB) nomap non-reusable secmon@5300000

````

MichaIng, 2024-07-22:
Thanks for reporting.

You you boot via SPI bootloader or MMC?

MichaIng, 2024-07-22:
In my case, I just tested it with an old little USB 2.0 stick. When using the MMC bootloader, all works fine. When using the SPI bootloader (I flashed the latest one from `linux-u-boot-current-odroidn2`), I get an error on first attempt, but it works on 2nd:
```console
G12B:BL:6e7c85:2a3b91;FEAT:E0F83180:402000;POC:F;RCY:0;EMMC:0;READ:0;0.▒!,K▒▒х▒▒}▒▒▒с0x01
bl2_stage_init 0x81
hw id: 0x0000 - pwm id 0x01
bl2_stage_init 0xc1
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:0000c067
L3:14000020
B2:00402000
B1:e0f83180

TE: 84167

BL2 Built : 06:17:13, Jun 28 2019. g12b gf0505d7-dirty - qi.duan@droid13

Board ID = 5
Set A53 clk to 24M
Set A73 clk to 24M
Set clk81 to 24M
A53 clk: 1200 MHz
A73 clk: 1200 MHz
CLK81: 166.6M
smccc: 000190a2
eMMC boot @ 0
sw8 s
DDR driver_vesion: LPDDR4_PHY_V_0_1_14 build time: Jun 28 2019 06:17:09
board id: 5
Load FIP HDR from eMMC, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from eMMC, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from eMMC, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
00000000
emmc switch 1 ok
00000000
emmc switch 2 ok
fastboot data verify
verify result: 255
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
DDR4 probe
ddr clk to 1320MHz
Load ddrfw from eMMC, src: 0x00014200, des: 0xfffd0000, size: 0x0000c000, part: 0
00000000
emmc switch 0 ok
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of Write leveling coarse delay
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from eMMC, src: 0x00020200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

R0_RxClkDly_Margin==94 ps 8
R0_TxDqDly_Margi==118 ps 10
[... 262 lines cut ...]
Unexpected XHCI event TRB, skipping... (f0f719a0 00000000 13000000 04008400)
BUG at drivers/usb/host/xhci-ring.c:530/abort_td()!
BUG!
resetting ...
```
On next attempt, it works:
```console
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... usb_new_device: Cannot read configuration, skipping device 058f:6387
4 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
Card did not respond to voltage select! : -110
switch to partitions #0, OK
mmc1(part 0) is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
```
On another reboot, it looped 4 times before it succeeded 🤔. Weird, either there is a bug in the driver, or not. Though probably one at the stage where the USB device is not fully ready yet, being some kind of race condition. I have another USB 3.0 device, where it works well on every attempt, so it seems to depend on the particular USB stick, how fast it initialises or such.

Did you try other USB ports?

MichaIng, 2024-07-22:
I just pushed a U-Boot update to our APT server:
```sh
apt update
apt upgrade
```
And then `dietpi-config` > Advanced Options > Update MMC bootloader.

Tin-Bui, 2024-07-23:
I think that broke something else.

```
sudo apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://download.docker.com/linux/debian bookworm InRelease
Get:3 https://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Get:5 https://deb.debian.org/debian bookworm-backports InRelease [56.6 kB]
Hit:6 https://dietpi.com/apt bookworm InRelease
Hit:7 https://dietpi.com/apt all InRelease
Get:8 https://deb.debian.org/debian bookworm-backports/main arm64 Packages.diff/Index [63.3 kB]
Get:9 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-07-23-0206.05-F-2024-07-23-0206.05.pdiff [696 B]
Get:9 https://deb.debian.org/debian bookworm-backports/main arm64 Packages T-2024-07-23-0206.05-F-2024-07-23-0206.05.pdiff [696 B]
Fetched 176 kB in 3s (60.9 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
dietpi@***:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  linux-u-boot-odroidn2-current
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 761 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://dietpi.com/apt all/odroidn2 arm64 linux-u-boot-odroidn2-current arm64 24.8.0-trunk-dietpi2 [761 kB]
Fetched 761 kB in 1s (1,093 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 28023 files and directories currently installed.)
Preparing to unpack .../linux-u-boot-odroidn2-current_24.8.0-trunk-dietpi2_arm64.deb ...
Unpacking linux-u-boot-odroidn2-current (24.8.0-trunk-dietpi2) over (24.8.0-trunk-dietpi1) ...
dpkg: unrecoverable fatal error, aborting:
 unable to fsync updated status of 'linux-u-boot-odroidn2-current': Input/output errorE: Sub-process /usr/bin/dpkg returned an error code (2)
dietpi@***:~$ sudo apt update
bash: /usr/bin/sudo: Input/output error
dietpi@***:~$ sudo apt upgrade
bash: /usr/bin/sudo: Input/output error
```

a reboot didn't make it better:
```
dietpi@***:~$ sudo apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://download.docker.com/linux/debian bookworm InRelease
Hit:5 https://dietpi.com/apt bookworm InRelease
Hit:6 https://dietpi.com/apt all InRelease
Hit:7 https://deb.debian.org/debian bookworm-backports InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
dietpi@***:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/761 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
dpkg: error processing package linux-u-boot-odroidn2-current (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Errors were encountered while processing:
 linux-u-boot-odroidn2-current
E: Sub-process /usr/bin/dpkg returned an error code (1)
dietpi@***:~$ sudo dpkg --configure -a
dpkg: error processing package linux-u-boot-odroidn2-current (--configure):
 package is in a very bad inconsistent state; you should
 reinstall it before attempting configuration
Errors were encountered while processing:
 linux-u-boot-odroidn2-current

```

I'll flash a fresh SD card to see if this still happen.

Tin-Bui, 2024-07-23:
Yep, that worked for me on a fresh SD card.
Thank you.