MichaIng/DietPi #7169: changing from RamLog to FullLog using dietpi-software should give notice that reboot is needed

te-anhydros, 2024-07-22:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- After using `dietpi-software` to reconfigure logging from **ramLog** to **fullLog**, running `df /var/log/.` showed that path to still be on a tmpfs. 
- After switching back and forth between ramLog and fullLog several times, it occurred to me that a reboot might be required. And so it was. 
- This is particularly annoying because the change from fullLog to ramLog does **not** require a reboot for the files system change.

#### Describe the solution you'd like:

- When running `dietpi-software` manually, please add a prominent message to the module for managing `/var/log` so that after a change from `/var/log` being a tmpfs to being part of a regular filesystem telling the operator that the filesystem will change after the next reboot.
- When `dietpi-software` is running as part of the initial unattended install/update and the operator selects the fullLog option via `/boot/dietpi.txt`, the process should add one more reboot to complete the process. That is what bit me today!

#### Describe alternatives you've considered:

- I looked in the documentation and bug-reports and issues for problems with `dietpi-software` having problems with this; found nothing.

#### Additional context

- Reference code: d2f560e4-1fd5-47a4-ae41-52fae7d8c10f

MichaIng, 2024-07-22:
The reason is that on `dietpi-software` installs, services are stopped, so that the `/var/log` tmpfs can be usually mounted without interrupting anything or being interrupted. On uninstalls, services are however not stopped, and some services keep their log files R/W open, so that trying to unmount the `/var/log` tmpfs results in a "device is busy" error, or may crash the service.

However, I though already that we should just try it:
1. Sync `/var/log` content to disk
2. Attempt to unmount `/var/log` tmpfs
3. If it works, sync content back in
4. If it fails, install the boot service to have it disable on next reboot, but show a prompt that a reboot is required

te-anhydros, 2024-07-23:
Hi @MichaIng, your counter-proposal is a more robust approach. I am all for it.

Do you have people on deck to implement this? If not, please direct me to the relevant files and functions and I'll try.

Thanks!

MichaIng, 2024-10-18:
Sorry for the late reply. It is all in the `dietpi-software` script: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software
Watch out for `To_Uninstall 103`.