MichaIng/DietPi #7170: Orange Pi Zero 2W ethernet speed < 1Mbps

nixielectra, 2024-07-23:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm`
- Kernel version | `Linux OPi02W 6.6.31-current-sunxi64 #1 SMP Fri May 17 10:02:40 UTC 2024 aarch64 GNU/Linux`
- SBC model | `Orange Pi Zero 2W (aarch64)`
- Power supply used | `5V2A`
- SD card used | `SanDisk ultra 64GB`

#### Steps to reproduce

1. Flash DietPi_OrangePiZero2W-ARMv8-Bookworm.img in the SD card
2. Boot the Pi

#### Expected behaviour

- Download speed should be ~90Mbps

#### Actual behaviour

- Download speed drop below 1Mbps on usb ethernet but upload speed as expected.
- I've ran this more than 10 times and all show the same result.
[image]

#### Extra details

- I have USB C to Ethernet connected to the Pi. Once it booted for the first time it took around ~30 mins to completed. When I install docker using `curl -sSL https://get.docker.com | sh` it took forever to complete.
- I installed speedtest cli `https://www.speedtest.net/apps/cli` just to check the speed.
- I have 600Mbps download and 500Mbps upload plan but with that USB C to Ethernet adapter using USB 2.0 it should cap at ~95Mbps for both down and up. At least I have tested this with my Galaxy Tab connected using that adapter.

nixielectra, 2024-07-25:
After switching to Samsung 25W power adapter, able to achieve 95Mbps on download.
Meaning the 5V2A as advertise in their website was not enough I guess if use with USB device but no issue if use WiFi.

closed 2024-07-25 by nixielectra

MichaIng, 2024-07-25:
That is good to know. A lot of issues regarding WiFi and Ethernet have been reported with this SBC. Probably more of them are solved with just a better PSU. Though I don't know exactly how it can have an effect, if it otherwise does boot fine and not crash.

Were there any kernel errors with the 2A PSU?

nixielectra, 2024-07-26:
At least from what I saw in the log, nothing to point out. Unfortunately I didn't save the log, if you want for your reference, and already thrown away the adapter out of anger for wasted three days of my life because of this. 😂

MichaIng, 2024-07-27:
Okay, at least something to keep in mind when others report similar problems 🙂 .