MichaIng/DietPi #7171: Raspberry: No Audio Device after installation

mschlenstedt, 2024-07-24:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm 0

- Kernel version | `uname -a`

Linux loxberrypoolboy 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

#### Steps to reproduce
Fresh Image Installation: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bookworm.img.xz
Same beaviour on an old Pi2 PCB V1.1 (https://dietpi.com/downloads/images/DietPi_RPi-ARMv7-Bookworm.img.xz)

#### Expected behaviour
The default Audio device (rpi-bcm2835-auto: Onboard HDMI if plugged, else 3.5mm) should be activated during installation. In older DietPi versions this was done during installation (IIRC).

#### Actual behaviour
No Aduio device will be activated during installation.

MichaIng, 2024-07-29:
No audio device is enabled OOTB as long as you do not select ALSA or any audio software for installation. Was this the case?

If so, I think there is still some issue with automatic switch between HDMI and 3.5mm audio, depending on whether an (audio capable) HDMI device is attached or not. With KMS enabled, the KMS-internal HDMI ALSA PCM is always present, hence HDMI audio is enforced, and for 3.5mm one then must select it explicitly. I need to review the code whether we already addressed this, and how. But in any case, an audio device should be enabled automatically whenever any local/jukebox audio software is installed.

mschlenstedt, 2024-07-31:
We are using the minimal installation with LoxBerry. So I think there's no audio software selected during installation. We do this "from the very first beginning" - and I was pretty sure that the audio device was activated in older DietPi versions with this minimal installation. Maybe I was wrong...

So no big thing, we can do this by our own during LoxBerry installation.

mschlenstedt, 2024-07-31:
I just checked the code of dietpi-config - we definetly had "aplay" installed, because I didn't get any warning to install alsa:

```if ! command -v aplay > /dev/null; then

    G_WHIP_MENU_ARRAY=('Enable' ': Install ALSA to enable audio capabilities')

·else
```

MichaIng, 2024-08-01:
Installing `alsa-utils` manually at least would not enable the audio devices firmware-wise. It is either `dtparam=audio=on` needed (`/boot/config.txt`) to enable the Broadcom audio codec, and `/etc/modprobe.d/dietpi-disable_rpi_audio.conf` removed to have the driver loaded at boot. Or KMS enabled without `noaudio` overlay parameter, for the KMS HDMI audio driver, or both, as the Broadcom codec is still needed for analogue 3.5mm output.

This is however all done when installing ALSA from `dietpi-software`, e.g. via `AUTO_SETUP_INSTALL_SOFTWARE_ID=5` in `dietpi.txt` before first boot.

mschlenstedt, 2024-08-02:
Thanks, we will do it via dietpi-software!

closed 2024-08-02 by mschlenstedt