MichaIng/DietPi #7173: DietPi on ODroid N2 fails to restart if USB drives are plugged in

beesyrupfactory, 2024-07-25:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues: Similar to #5729 .

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
# cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
# echo $G_DISTRO_NAME $G_RASPBIAN
bookworm
```

- Kernel version | `uname -a`
```
# uname -a
Linux DietPi 6.6.37-current-meson64 #1 SMP PREEMPT Fri Jul  5 07:34:07 UTC 2024 aarch64 GNU/Linux
```

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
# echo $G_HW_MODEL_NAME
Odroid N2 (aarch64)
```

- Power supply used | (EG: 5V 1A RAVpower)
12V/2A power supply from [HARDKERNEL](https://www.hardkernel.com/shop/12v-2a-power-supply-us-plug/) (Odroid's parent company)

- SD card used | (EG: SanDisk ultra)
Using an eMMC module: [32GB eMMC Module N2 Linux](https://www.hardkernel.com/shop/32gb-emmc-module-n2-linux/)

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated?
I recently had to recreate my DietPi installation due to a problem where a kernel installation failed due to the the `/boot` partition being full. I tried to restart, but I could not figure it out, so I reflashed a fresh DietPi image for the ODroid N2/N2+

- Can this issue be replicated on a fresh installation of DietPi?
Yes. Prior to the reflash/reinstallation, I did not have this problem. 

- Bug report ID | `echo $G_HW_UUID`
`60c6967f-490b-4d62-b7ae-a9081e311e92`

#### Steps to reproduce
1. Download DietPi_OdroidN2-ARMv8-Bookworm.img.xz, and extract the file
2. Connect eMMC module via USB adaptor
3. Using balenaEtcher, flash the IMG file to the eMMC module
4. Ensure SPI - eMMC switch is set to eMMC
5. Install the eMMC card
6. Start up the device, and go through the initial setup processes 
7. After everything has completed, mount existing USB drives (1 SSD, 3 SATA) formatted ext4
8. Set up drives using dietpi-drive_manager setting up mount points.
9. Reboot the system
10. Observe bootloop (sorry, I don't have the errors, but I remember UBoot showing up and it constantly restarting after a failure was observed).
11. Pull the plug on the system, then unplug all USB drives
12. Insert the plug and restart the system successfully
13. Plug in the USB drives and observed that they are recognized

#### Expected behaviour
- The system will know to boot from the eMMC even if USB drives are connected.

#### Actual behaviour
- The system bootloops until I power off, and unplug the USB drives.

#### Extra details
-  I don't know what happened between my earlier setup and my current setup, but on the previous install, I may have upgraded from an earlier version of DietPi that included a Debian Bullseye to Bookworm upgrade. The kernel that was in use was 4.9.312 prior to the reflash.

```# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME         FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT PARTUUID                             UUID
sda                       119.2G  0 disk
`-sda1       ext4         119.2G  0 part /mnt/ssd0  223c8ef1-80f0-4ab9-b6a8-3ea19a89ba9d 9cdbb1be-4eb8-4a22-bb9c-bfb26e6bbe7e
sdb                         5.5T  0 disk
`-sdb1       ext4           5.5T  0 part /mnt/hdd0  14c5e6e6-43c1-4b2e-a248-468bf4e5fcf7 b4fe66c2-1e7f-4a0c-9507-18797071ac29
sdc                        10.9T  0 disk
`-sdc1       ext4          10.9T  0 part /mnt/hdd1  0ece33a9-5abf-41c2-a438-9881473b9ad1 df3536df-c062-4f19-b65d-61c2b73398df
sdd                         9.1T  0 disk
`-sdd1       ext4           9.1T  0 part /mnt/hdd2  44b3b917-ea81-45ec-9fa9-706a4b4b4676 6b1df207-07d3-4e3e-ad99-a80996ea529b
mmcblk1                    29.1G  0 disk
`-mmcblk1p1  ext4          29.1G  0 part /          c196e89d-01                          92b4b865-897e-47a6-afb4-c25d1bb51d57
mmcblk1boot0                  4M  1 disk
mmcblk1boot1                  4M  1 disk
```
If there are any other details that you need, please let me know and I will be happy to aid with the investigation.

Cheers!

MichaIng, 2024-07-25:
It has been reported and solved/worked around already: #7168
Armbian changed the default boot oder so that USB drives are checked first. However, in some cases, scanning the USB drives first crashes the bootloader, resulting in reboot and boot loop. We removed the patch which changes the boot order, so that again eMMC and SD card are checked first. The new bootloader package has been pushed to our APT server already:
```sh
apt update
apt upgrade
/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```

beesyrupfactory, 2024-07-26:
Thanks @MichaIng ! I'm going to run a backup, and test this out.
Cheers!

beesyrupfactory, 2024-07-26:
I realize now what this means now from #7168:
> And then dietpi-config > Advanced Options > Update MMC bootloader.

I take it that this is what `/boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc` does. Is this correct?

Thanks again for your support!

beesyrupfactory, 2024-07-26:
Legend. Thanks @MichaIng. It works perfectly. As this is a dupe of #7168 , and is on the [milestone 9.7](https://github.com/MichaIng/DietPi/milestone/143), this can be closed.

Thanks!

closed 2024-07-26 by beesyrupfactory

MichaIng, 2024-07-27:
Great, yes the `dietpi-set_software` command does the same, looks like I forgot about it the moment I was answering you, implemented it not too long ago 😅.