MichaIng/DietPi #7175: Can't find wifi adapter - Intel AX101

fra87, 2024-07-29:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=6
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm`
- Kernel version | `Linux ServerCasa 6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)` - Chuwi Herobox N5100
- Power supply used | Proprietary power supply
- SD card used | N/A (uses NVME SSD)

#### Steps to reproduce

1. Freshly install Dietpi on the board
2. Internal wifi does not work.

#### Expected behaviour

- Internal wifi works

#### Actual behaviour

- It does not work ;)

#### Extra details

- When looking at dietpi-config, the wifi state is `WiFi     : Not Found | [On] | Disconnected`
- When  trying to use dietpi-wifidb, it fails with `DietPi-WiFiDB | An interface of type "wlan" does not exist.`
- Module is found by lspci:
```
00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201 160MHz (rev 01)
        Subsystem: Intel Corporation Wi-Fi 6 AX201 160MHz
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
```
- dmesg output related to iwlwifi is 
[attached here](https://github.com/user-attachments/files/16413924/iwlwifi.txt)

MichaIng, 2024-07-29:
Hmm, it has problems loading the firmware. Is this a very new Intel WiFi chip? It finds and loads the correct kernel module, which again finds matching firmware to load, but probably it is too old.

You could try the following, either with a backup or on a newly flashed/installed system, for testing:

1. Install newer kernel from Debian backports:

```sh
apt install -t bookworm-backports linux-image-amd64
reboot
```

2. Install newer firmware from Debian Trixie. It has no dependency, so it can be installed:

```sh
cd /tmp
wget 'https://deb.debian.org/debian/pool/non-free-firmware/f/firmware-nonfree/firmware-iwlwifi_20240610-1_all.deb'
dpkg -i firmware-iwlwifi_20240610-1_all.deb
rm firmware-iwlwifi_20240610-1_all.deb
modprobe -r iwlwifi
modprobe iwlwifi
sleep 1
ip l
```
The last command is to show detected network devices. If all work well, the WiFi adapter should show up as `wlan0`.

fra87, 2024-07-29:
According to Intel's website it's new but not so much new. The chipset is dated Q3'22, while bookworm is from June 23, so it was already almost 1 year old when bookworm came out.

In any case after step 1 (reboot after installing new kernel) the WLAN0 comes out :)

Do you think step 2 (installing trixie firmware) is needed?

The dmesg output now reads

```
[    3.404798] Intel(R) Wireless WiFi driver for Linux
[    3.404866] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    3.433320] iwlwifi 0000:00:14.3: Detected crf-id 0x3617, cnv-id 0x20000302 wfpm id 0x80000000
[    3.433345] iwlwifi 0000:00:14.3: PCI dev 4df0/0244, rev=0x351, rfid=0x10c000
[    3.433469] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-77.ucode (-2)
[    3.433484] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-77.ucode (-2)
[    3.433486] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-77.ucode failed with error -2
[    3.433504] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-76.ucode (-2)
[    3.433513] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-76.ucode (-2)
[    3.433515] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-76.ucode failed with error -2
[    3.433529] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-75.ucode (-2)
[    3.433541] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-75.ucode (-2)
[    3.433542] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-75.ucode failed with error -2
[    3.433556] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-74.ucode (-2)
[    3.433566] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-74.ucode (-2)
[    3.433567] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-74.ucode failed with error -2
[    3.433582] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-73.ucode (-2)
[    3.433594] iwlwifi 0000:00:14.3: firmware: failed to load iwlwifi-QuZ-a0-hr-b0-73.ucode (-2)
[    3.433595] iwlwifi 0000:00:14.3: Direct firmware load for iwlwifi-QuZ-a0-hr-b0-73.ucode failed with error -2
[    3.439249] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    3.439808] iwlwifi 0000:00:14.3: firmware: failed to load iwl-debug-yoyo.bin (-2)
[    3.439819] iwlwifi 0000:00:14.3: firmware: failed to load iwl-debug-yoyo.bin (-2)
[    3.439823] iwlwifi 0000:00:14.3: loaded firmware version 72.daa05125.0 QuZ-a0-hr-b0-72.ucode op_mode iwlmvm
```

MichaIng, 2024-07-29:
So far so good. It finally loads the same firmware blob `QuZ-a0-hr-b0-72.ucode` like before, but this time successfully. Interesting is that it tires to load some over ones before, failing to do so. Looking at the long you posted before, it seems these other attempts were all done previously already, but the lines first lines were capped (kmsg ring buffer full)?

I checked the firmware package versions, and indeed the Bookworm one misses the firmware blobs the driver is looking for at first: https://packages.debian.org/bookworm/all/firmware-iwlwifi/filelist
The Trixie version contains them: https://packages.debian.org/trixie/all/firmware-iwlwifi/filelist

So I would do the 2nd step as well, and install the newer firmware package. Even when the currently loaded older firmware blobs seems still compatible (at least according to the driver's meta data), the one it looks for first is likely working better with this chip.

fra87, 2024-07-30:
And here is the dmesg at the end

```
[    3.480572] iwlwifi 0000:00:14.3: enabling device (0000 -> 0002)
[    3.482026] iwlwifi 0000:00:14.3: Detected crf-id 0x3617, cnv-id 0x20000302 wfpm id 0x80000000
[    3.482046] iwlwifi 0000:00:14.3: PCI dev 4df0/0244, rev=0x351, rfid=0x10c000
[    3.489735] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    3.490283] iwlwifi 0000:00:14.3: firmware: failed to load iwl-debug-yoyo.bin (-2)
[    3.490297] iwlwifi 0000:00:14.3: firmware: failed to load iwl-debug-yoyo.bin (-2)
[    3.490301] iwlwifi 0000:00:14.3: loaded firmware version 77.a20fb07d.0 QuZ-a0-hr-b0-77.ucode op_mode iwlmvm
[    3.724952] iwlwifi 0000:00:14.3: Detected Intel(R) Wi-Fi 6 AX101, REV=0x351
[    4.191748] iwlwifi 0000:00:14.3: Detected RF HR1 B3, rfid=0x10c000
```

Seems perfect 😁 tested WiFi and it works. Thank you

will I need to do something in the future to "align to mainline" when these modifications get pushed in the kernel?

Just a note: I had to unload also iwlmvm (command `modprobe -r iwlmvm iwlwifi`) otherwise owlwifi could not be unloaded (it was in use)

MichaIng, 2024-08-04:
Okay great. Not a solution we can ship with DietPi images, but it is good to know that this option exists for those who need to use this WiFi adapter or other newer ones, which are not natively supported by Linux 6.1 yet.

I'll mark this issue as closed.

closed 2024-08-04 by MichaIng