MichaIng/DietPi #7177: Support CM3588 NAS

Wuppie007, 2024-07-30:
### Creating an image request

#### Formal device information
- Device name | CM3588
- Official product URL | https://www.friendlyelec.com/index.php?route=product/product&path=69&product_id=292

#### Is the SBC officially supported by the Debian installer?
- ...

#### If not, is a reliable 3rd party Debian image available for this SBC?
- ...

#### If not, are there install instructions for Debian available?
armbian has a config for it: https://github.com/armbian/build/blob/468342d3f75f6696f99d388b5b76f1b35c5e44d3/config/boards/nanopc-cm3588-nas.csc#L4

Wuppie007, 2024-11-04:
I tried running with the image for NanoPC T6, but only one nvme socket is working (there are 4 on it). Who can push me in the right direction?

andreagdipaolo, 2024-11-16:
I recently got the same board and it would be nice to keep DietPi. I've been a happy user for years on my ROCK 4B and it's time to upgrade to something a little more powerful and with a more convenient form factor.

MichaIng, 2024-11-21:
@Wuppie007 
There is no `fdtfile=` defined in `/boot/dietpiEnv.txt`, right? Does the bootloader the correct one OOTB? Check whether this returns the correct SBC/CM name:
```sh
cat /proc/device-tree/model
```
If not, check in `/boot/dtb/rockchip` whether you can find the correct device tree, and then add a line to `/boot/dietpiEnv.txt` like this (adjust the file name accordingly):
```sh
fdtfile=rockchip/rk3588-nanopc-cm3588.dtb
```