MichaIng/DietPi #7178: dietpi-dashboard installs older version then whats available

mews-se, 2024-07-31:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9`
- Distro version | `bookworm 0`
- Kernel version | `Linux xxx 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)` or (EG: RPi3)

#### Additional Information (if applicable)
- Software title | dietpi-dashboard
- Was the software title installed freshly or updated/migrated? | freshly
- Can this issue be replicated on a fresh installation of DietPi? | yes

#### Steps to reproduce

1. install nightly version of dietpi dashboard. installed version is [v0.6.1-527-g9ea6873](https://github.com/ravenclaw900/DietPi-Dashboard/releases/tag/v0.6.1-527-g9ea6873)
2. version 0.6.2 is available but unable to update

#### Expected behaviour

- update to newest version

#### Actual behaviour

- same older version is installed

Joulinar, 2024-08-01:
Using stable branch should install `v0.6.2`

```
 DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing DietPi-Dashboard: Official lightweight DietPi web interface (Rust)

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://github.com/ravenclaw900/DietPi-Dashboard/releases/download/v0.6.2/dietpi-dashboard-x86_64 -o dietpi-dashboard-x86_64
[  OK  ] DietPi-Software | mkdir -p /opt/dietpi-dashboard
[  OK  ] DietPi-Software | mv dietpi-dashboard-x86_64 /opt/dietpi-dashboard/dietpi-dashboard
[  OK  ] DietPi-Software | chmod +x /opt/dietpi-dashboard/dietpi-dashboard
[  OK  ] DietPi-Software | curl -sSfL https://raw.githubusercontent.com/ravenclaw900/DietPi-Dashboard/v0.6.2/config.toml -o /opt/dietpi-dashboard/config.toml
```

MichaIng, 2024-08-01:
Yeah, I am also not sure why, but the nightly version is tagged v0.6.1. But it is newer than the v0.6.2 release.

ravenclaw900, 2024-08-02:
It's because 0.6.2 was a hotfix version, so it wasn't built from the main branch, but its own branch as a slightly modified version of 0.6.1. The nightly version names are derived from the releases on the main branch only. The nightly version is still the latest, despite the slightly confusing naming.

closed 2025-05-14 by MichaIng