MichaIng/DietPi #7179: Ошибка DNS в системе DietPi

Ruslan-Musatov, 2024-08-05:
[image]
Ошибка на изображении показывает, что система DietPi не смогла выполнить ping до адреса 8.8.8.8 (DNS сервер Google), что привело к 100% потере пакетов. Это указывает на проблему с сетевым подключением. Давайте подробнее разберем возможные причины и действия, которые можно предпринять для устранения этой проблемы.
При этом не пингуется ни один DNS .
В чем может проблема?

Joulinar, 2024-08-05:
Please use English as non of us speaks Russian. Therefore I translate your post using DeepL

To your problem. This has nothing to do with DNS. It's simple ping check to verify basic internet connection. Ensure you are able to access the internet. Depending on your Geo location, Google DNS server IP might not be reachable.

Ruslan-Musatov, 2024-08-05:
I use a Rock 4 Model C+ microcomputer, it's the same as the Raspberry Pi 4 model B. These are the settings I made in the file dietpi.txt:
AUTO_SETUP_NET_USESTATIC=1
AUTO_SETUP_NET_STATIC_IP=192.168.137.100
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.137.1
AUTO_SETUP_NET_STATIC_DNS=9.9.9.9 149.112.112.112
I use isernet to transfer the network to a micro computer. I am transmitting it from my PC using an adapter. These are the settings I have for network transmission.
[image]

Ruslan-Musatov, 2024-08-05:
> I use a Rock 4 Model C+ microcomputer, it's the same as the Raspberry Pi 4 model B. These are the settings I made in the file dietpi.txt: AUTO_SETUP_NET_USESTATIC=1 AUTO_SETUP_NET_STATIC_IP=192.168.137.100 AUTO_SETUP_NET_STATIC_MASK=255.255.255.0 AUTO_SETUP_NET_STATIC_GATEWAY=192.168.137.1 AUTO_SETUP_NET_STATIC_DNS=9.9.9.9 149.112.112.112 I use isernet to transfer the network to a micro computer. I am transmitting it from my PC using an adapter. These are the settings I have for network transmission. ![MKnoiXoVBhA](https://private-user-images.githubusercontent.com/166721463/355022430-227c9a7c-94d6-43b0-b2c9-09098c78de3f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4NDgyMzMsIm5iZiI6MTcyMjg0NzkzMywicGF0aCI6Ii8xNjY3MjE0NjMvMzU1MDIyNDMwLTIyN2M5YTdjLTk0ZDYtNDNiMC1iMmM5LTA5MDk4Yzc4ZGUzZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwNVQwODUyMTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MWQ3YzAwYTY4ZWEwODMzZWFjNjVkOTljNTUyMTZmODc1MWY3NTllZTg3MzZiZTFjM2IwODlmZGE3ZmEzODBlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.PI7pw2mu07Zia8m-CFW16ZA3L8fWqiCarLcQamzJWAE)

[image]
the izernet 2 network is connected to Rock 4, it says without an internet connection, can this affect my error?

closed 2024-08-05 by Ruslan-Musatov

MichaIng, 2024-08-06:
You solved it? If Ethernet 2 is connected to the SBC, then it is normal that it shows "no Internet", as this is only shown on those adapters, where the PC itself has Internet access through. Important is the other one, and that it is a shared Internet correction.